Unity-Technologies/Unity-Robotics-Hub #217: Socket error: "Connection reset by peer"

m-talha, 2021-05-06:
I am trying to subscribe to a `std_msgs/Float64MultiArray` topic in Unity which is being published by a MoveIt library in ROS. 

The subscription is not working properly. The socket connection repeatedly (every 30s or so) gives the error: `"Exception [Errno 104] Connection reset by peer"`. On the Unity side, I receive messages on the topic approximately every 30 secs also. Both events seem to coincide.

Any ideas on how to go about debugging this? Thanks

at669, 2021-05-06:
Hi @m-talha, you can try checking out our [ROS-Unity Integration Networking guide](https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/dev/tutorials/ros_unity_integration/network.md), which documents the different components of the communication alongside a troubleshooting section. Please take a look and let us know if any of those steps resolve your issue.

If those steps don't resolve the problem, could you describe the steps to reproduce the issue, full console logs/stack traces, and environment you're using (e.g. Unity version, machine OS, ROS machine OS, ROS version, Docker/WSL/etc, and Connector and Endpoint package versions)?

m-talha, 2021-05-07:
So I have been debugging this with a minimal example and it seems like the issue is with repeated messages with the same data being sent from the TCP server to Unity.

To reproduce:

1. In Unity, set up an empty game object with the following subscriber script:

```
using UnityEngine;
using Unity.Robotics.ROSTCPConnector;
using RosF64MultiArray = RosMessageTypes.Std.Float64MultiArray;

public class FloatArrayMsgSubTest : MonoBehaviour
{
    public string topicName = "msg";

    void Start()
    {
        ROSConnection.instance.Subscribe<RosF64MultiArray>(topicName, MsgResponse);
    }
    void MsgResponse(RosF64MultiArray jointsMessage)
    {
        Debug.Log($"Message received.");  
    }
}
```

2. In ROS, start a `server_endpoint.py` script:
```
#!/usr/bin/env python3

import rospy

from ros_tcp_endpoint import TcpServer, RosPublisher, RosSubscriber, RosService
from std_msgs.msg import Float64MultiArray, Int16

def main():
    ros_node_name = rospy.get_param("/TCP_NODE_NAME", 'TCPServer')
    tcp_server = TcpServer(ros_node_name)
    rospy.init_node(ros_node_name, anonymous=True)

    # Start the Server Endpoint with a ROS communications object dictionary for routing messages
    tcp_server.start({
        'joint_cmds' : RosSubscriber('msg', Float64MultiArray, tcp_server)
    })
    # Blocks until shutdown
    rospy.spin()

if __name__ == "__main__":
    main()
```

3. In another terminal, run the message publisher:

```
rostopic pub -r 10 /msg std_msgs/Float64MultiArray "layout:
  dim:
  - label: ''
    size: 0
    stride: 0
  data_offset: 0
data: [0, 5, 2, 1]" 
```

4. Run the scene in Unity. 

Expected Output: "Message received" messages at 10Hz in the Unity console

Actual Output:
"Message received" messages are printed at about 8 secs apart
At the same time, there is a error message in the ROS output where the server_endpoint.py is running:
`[INFO] [1620407697.294539]: Exception [Errno 104] Connection reset by peer`

Unity:
[image]

ROS:
[image]

My setup:
Unity 2020.1.3f1 (also tried with 2021.6.1f1) on Win10 Home 64-bit
ROS: Noetic in Docker on Ubuntu 18.04
Connector package version: 0.2.0

ROS IP Settings:
[image]

at669, 2021-05-07:
Hi @m-talha, I was able to recreate a similar issue with the ROS-TCP-Connector package version`v0.2.0`, but these issues are resolved on my end using the updated `v0.3.0` version (i.e. `https://github.com/Unity-Technologies/ROS-TCP-Connector.git?path=/com.unity.robotics.ros-tcp-connector#v0.3.0`)--could you try updating and seeing if this resolves your issue? 

Please note that the message classes have updated in the package and you will have to prepend references to the class names with `M`, e.g.:

```csharp
using UnityEngine;
using Unity.Robotics.ROSTCPConnector;
using RosMessageTypes.Std;

public class FloatArrayMsgSubTest : MonoBehaviour
{
    public string topicName = "msg";

    void Start()
    {
        ROSConnection.instance.Subscribe<MFloat64MultiArray>(topicName, MsgResponse);
    }
    void MsgResponse(MFloat64MultiArray jointsMessage)
    {
        Debug.Log($"Message received");  
    }
}
```

I've attached a brief video clip of what I believe to be the desired behavior. Please let me know if the package update helps!

https://user-images.githubusercontent.com/31416491/117507684-9eb17900-af44-11eb-870c-cecc9ae4704d.mov

m-talha, 2021-05-07:
Hi Amanda,

Thanks for the update.

This is exactly what I was looking for and now it is working on my end also. 

Should mention the likely cause of this is the outdated documentation in the [Quick Start](https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/main/tutorials/quick_setup.md). Could really do with updating the doc to use v.0.3.0.

Thanks again for the update.

closed 2021-05-07 by m-talha