Unity-Technologies/Unity-Robotics-Hub #223: Subscribed message isn't being received in Unity.

bryceikeda, 2021-05-15:
**Describe the bug**
I have a simple Odometry subscriber on the Unity side that was working with v0.2.0. However, with v0.3.0, once I create the new message type of MOdometry and subscribe to the topic my odometry is being published to, no messages are being received. However, I can see the TCP handshake occurring on both sides. Oddly, I can recreate the Pick and Place tutorial and send and receive messages just fine. Is there something wrong with how I have written my subscriber?

**To Reproduce**
Steps to reproduce the behavior:
Odometry Subscriber Script
`using System.Collections;
using System.Collections.Generic;
using UnityEngine;
using RosMessageTypes.Nav;
using Unity.Robotics.ROSTCPConnector;
using Unity.Robotics.ROSTCPConnector.ROSGeometry;

public class OdomSubscriber : MonoBehaviour
{
    private ROSConnection ros;

    public Transform PublishedTransform;
    public Vector3 position;
    public Quaternion rotation;
    public bool messageReceived;

    // Start is called before the first frame update
    void Start()
    {
        ROSConnection.instance.Subscribe<MOdometry>("estimated_odom", Write);
    }

    // Update is called once per frame
    void Update()
    {
        if (messageReceived)
        {
            PublishedTransform.position = position;
            PublishedTransform.rotation = rotation;
            messageReceived = false; 
        }
    }

    public void Write(MOdometry odometry)
    {
        position = odometry.pose.pose.position.From<FLU>();
        rotation = odometry.pose.pose.orientation.From<FLU>();
        messageReceived = true;
    }
}
`

server_endpoint.py script

`import rospy

from ros_tcp_endpoint import TcpServer, RosPublisher, RosSubscriber, RosService
from nav_msgs.msg import Odometry

def main():
    ros_node_name = rospy.get_param("/TCP_NODE_NAME", 'TCPServer')
    buffer_size = rospy.get_param("/TCP_BUFFER_SIZE", 1024)
    connections = rospy.get_param("/TCP_CONNECTIONS", 10)
    tcp_server = TcpServer(ros_node_name, buffer_size, connections)
    rospy.init_node(ros_node_name, anonymous=True)

    tcp_server.start({
         'map' : RosSubscriber('map', OccupancyGrid, tcp_server) })

    rospy.spin()

if __name__ == "__main__":
    main()
`

MOdometry msg

`//Do not edit! This file was generated by Unity-ROS MessageGeneration.
using System;
using System.Linq;
using System.Collections.Generic;
using System.Text;
using Unity.Robotics.ROSTCPConnector.MessageGeneration;
using RosMessageTypes.Std;

namespace RosMessageTypes.Nav
{
    public class MOdometry : Message
    {
        public const string RosMessageName = "nav_msgs/Odometry";

        //  This represents an estimate of a position and velocity in free space.  
        //  The pose in this message should be specified in the coordinate frame given by header.frame_id.
        //  The twist in this message should be specified in the coordinate frame given by the child_frame_id
        public MHeader header;
        public string child_frame_id;
        public Geometry.MPoseWithCovariance pose;
        public Geometry.MTwistWithCovariance twist;

        public MOdometry()
        {
            this.header = new MHeader();
            this.child_frame_id = "";
            this.pose = new Geometry.MPoseWithCovariance();
            this.twist = new Geometry.MTwistWithCovariance();
        }

        public MOdometry(MHeader header, string child_frame_id, Geometry.MPoseWithCovariance pose, Geometry.MTwistWithCovariance twist)
        {
            this.header = header;
            this.child_frame_id = child_frame_id;
            this.pose = pose;
            this.twist = twist;
        }
        public override List<byte[]> SerializationStatements()
        {
            var listOfSerializations = new List<byte[]>();
            listOfSerializations.AddRange(header.SerializationStatements());
            listOfSerializations.Add(SerializeString(this.child_frame_id));
            listOfSerializations.AddRange(pose.SerializationStatements());
            listOfSerializations.AddRange(twist.SerializationStatements());

            return listOfSerializations;
        }

        public override int Deserialize(byte[] data, int offset)
        {
            offset = this.header.Deserialize(data, offset);
            var child_frame_idStringBytesLength = DeserializeLength(data, offset);
            offset += 4;
            this.child_frame_id = DeserializeString(data, offset, child_frame_idStringBytesLength);
            offset += child_frame_idStringBytesLength;
            offset = this.pose.Deserialize(data, offset);
            offset = this.twist.Deserialize(data, offset);

            return offset;
        }

        public override string ToString()
        {
            return "MOdometry: " +
            "\nheader: " + header.ToString() +
            "\nchild_frame_id: " + child_frame_id.ToString() +
            "\npose: " + pose.ToString() +
            "\ntwist: " + twist.ToString();
        }
    }
}
`

**Expected behavior**
Odometry values sent from ROS and received by Unity.

**Environment (please complete the following information, where applicable):**
- Unity Version: [e.g. Unity 2020.3.8f1]
- Unity machine OS + version: Windows 10
- ROS machine OS + version: Ubuntu 18.04, ROS Melodic
- ROS–Unity communication: Manual
- Branch or version: v0.3.0

rahidraghav, 2021-05-17:
i have the same issue. Is this fixed ?

hyounesy, 2021-05-17:
Hi, Thanks for reporting the issue. We will investigate this ASAP and get back to you.
Internal ticket # AIRO-686

rahidraghav, 2021-05-18:
Solved mine.. had to start the subscriber node before starting the unity env and this solves my problem.

bryceikeda, 2021-05-18:
> Solved mine.. had to start the subscriber node before starting the unity env and this solves my problem.

@rahidraghav Do you mean you didn't start the server_endpoint.py script before running Unity?

bryceikeda, 2021-05-19:
The problem was even though I allowed Unity through my firewall, it wouldn't allow data to be passed through. I had to go through the advanced settings to disable an inbound rule that blocked the incoming data. However, I was still able to use service requests which was weird. Maybe because it requires a response to an outbound message so it allows that through? Not sure.

closed 2021-05-19 by bryceikeda