Unity-Technologies/Unity-Robotics-Hub #224: : Starting server on 192.168.178.58:10000 ROS-Unity Handshake received, will connect to 192.168.178.54:5005

florianspy, 2021-05-17:
**Describe the bug**
Following the pick-and-place tutorial till the step roslaunch niryo_moveit part_2.launch
Instead of  geting the heard message in terminal i only get the ROS-Unity Handshake received and but then nothing when the publish button is pushed
This happens with the docker image and ros melodic running in a VM

hyounesy, 2021-05-17:
Thanks for reporting the issue. We will investigate ASAP and get back to you.
Internal ticket #  AIRO-687

hyounesy, 2021-05-22:
Hi @florianspy, Would you please check the following:

- do you see any error message on the Console on the UnityEditor after pressing play?
- on the info box top left corner of the Game panel in the window, toggle the [View Service Status] checkbox. Do you see "handshake Service Requested" and "handshake Service Response"?
- With the "Publisher" GameObject selected in the Hierarchy pane, check that the properties for "Source Destination Publisher" in the Inspector pane are all set.

florianspy, 2021-05-25:
@hyounesy 
1. No Errors in the console
[image]
2. Handshake service response is there 
[image]
3. Publisher object setup as explained in tutorial
[image]

hyounesy, 2021-05-25:
Thanks for the info @florianspy , Based on your 2nd screenshot, looks like the UNITY IP may not be setup. Please try the following:
- In Unity Editor click on the main menu item: Robotics -> ROS Settings
- set the "Override Unity IP Address" field to the local IP of the machine running the Unity Editor. 
- close the "ROS Setting" window
- In Unity Editor click on the main menu item: File -> Save Project

florianspy, 2021-05-26:
@hyounesy 
I adopted the settings 
[image]
unfortunately still no information on ros side
[image]

hyounesy, 2021-05-28:
Hi @florianspy , Based on the screenshot above, look like the ROS IP address is incorrectly set to the local Unity IP address. Please try changing it to 127.0.0.1 in the ROS Settings in your Unity Editor.

florianspy, 2021-05-31:
Dear @hyounesy  with this setting i instantly get the following error message:
[image]
Note that my ros runs in a vm.

florianspy, 2021-05-31:
Dear @hyounesy  I figured now out where there problem is with a friend of mine. So basically the EventSystem was missing that why buttons never worked, so it has nothing to due with the ip. So please add to the instruction to add the EventSystem.

hyounesy, 2021-05-31:
Hi @florianspy , Thanks for letting us know. I am happy that you managed to resolve the issue. It is strange that you didn't have the EventSystem object as it is automatically added to the scene as soon as a UI object like a button is added to the scene. Was it possibly deleted by mistake? We will update the troubleshooting document as you suggested.
Not having the EventSystem can in fact be reason for the  UI events not to trigger. However it would not cause the "Socket Exception: Connection Refused" issue you mentioned above. Are you still experiencing the connection issue?

hyounesy, 2021-05-31:
internal ticket#: AIRO-749

florianspy, 2021-06-01:
The only additional thing I did which is probably not necessary as you don't communicate via Ros port 11311 but via 10000 is in bashrc file the variable link ROS_HOSTNAME and Ros_master_uri to the IP address of ros. 
"The socket refused to connect" only happened when i gave in Unity localhost as the adresse of ROS and not the ip of the system running ROS which makes sense as ROS was running in the VM and not directly on localhost

sdiao, 2021-06-02:
Hi @florianspy, thanks for the update. We just released several ROSConnection improvements in [ROS-TCP-Connector v0.4.0 ](https://github.com/Unity-Technologies/ROS-TCP-Connector/tree/v0.4.0). We have simplified the ROS Setting window in Unity Editor. I would recommend check out the latest features with the pick and place tutorial. Please let us know if there is anything else we can help with. Thank you.

florianspy, 2021-06-02:
Dear @sdiao  I upgraded to the newest version it works. 
However before I could get it working i had to change the Demo.cs file to the latest release avoid errors with rosConnection missing. 
Then i had to download the latest version of the ROSpackage otherwise i got the error "Can't send a message, no defined unity IP!".
Additionally Topic Name of the SourceDestinationPublisher currently seems to be not used anymore, because  even if i give "SourceDestination_input" in unity still "SourceDestination" is used.
[image]
[image]

sdiao, 2021-06-02:
Hi @florianspy , I am glad the v0.4.0 works for you. Yes, we would recommend upgrade to v0.4.0 in Unity Robotics Hub and ROS-TCP-Endpoint as well. For the topic name, the TopicName in SourceDestinationPublisher is used. The confusion is because in ROS, we registered the topic as below in [server_endpoint](https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/v0.4.0/tutorials/pick_and_place/ROS/src/niryo_moveit/scripts/server_endpoint.py#L18):
`'SourceDestination_input': RosPublisher('SourceDestination', NiryoMoveitJoints, queue_size=10)`

Hence, `rostopic list` returns `SourceDestination` while the validation is using `SourceDestination_input`.  Hope it helps. Let me know if you have further questions. Thanks.

closed 2021-06-08 by sdiao