Unity-Technologies/Unity-Robotics-Hub #225: Part 3 Tutorial, MoveIT failed to fetch robot state

lberridge1992, 2021-05-19:
Hello,

I've been trying to configure the Part 3 tutorial and cannot get the service call from the trajectory planner to successfully return a trajectory. Part 2 works as expected and I get the ros topic printed in the terminal. The output in my ROS terminal for part 3 shows the following error:

[image]

The error sounds like moveit hasn't received any robot state information recently which from running through the tutorials and the project doesn't look like it is ever sent. I've tried implementing a publish of /joint_states from Unity with the correct timestamps to get this working which seems to work based on the output of a rostopic echo but still with no luck with the error in the output when requesting a trajectory via the publish button in Unity. Am I missing something really obvious with this?

Many thanks!

hyounesy, 2021-05-22:
Hello @lberridge1992,
The robot state information are sent as a `MMoverServiceRequest` the `TrajectoryPlanner.PublishJoints()`. Could you please check the following?
- Do you receive the error on the ROS after pressing the UI button with the OnClick() set to `TrajectoryPlanner.PublishJoints` ? 
- Do you see any errors in the Console window in the Unity Editor?
- With the "Publish" game object selected in the Hierarchy pane, are all the properties for the "Trajectory Planner" script set to the correct values in the inspector?

lberridge1992, 2021-05-24:
Hi @hyounesy
Thanks for getting back to me. The error is received after pressing the UI button and I don't get any errors in the Unity Editor console other than telling me that no trajectory was returned. I've definitely assigned all of the inspector components in the publish game object and have printed some statements out on the ros side to make sure the data makes it through and the state information comes through properly.

I downloaded the source code for the MoveIt package just so I could see what triggers the "Failed to fetch robot state" error. The python code that receives the MMoverServiceRequest attempts to set the start state of the move group which tracing through a python wrapper calls setStartState on line 1539 in the move_group_interface.cpp. This function tries to get the current state of the robot before using the state information passed in via MMoverServiceRequest and fails because getCurrentState() waits for a /joint_states topic with a recent timestamp to be published. Looking at rqt graph the only thing that was publishing  /joint_states was the niryo one driver which I don't think publishes anything unless a physical robot is connected (or certainly in my setup it isn't publishing anything!). Unless I've misunderstood the Moveit source code, I'm not sure how it would have worked before and it made me start to wonder if something has changed in a package or if i've missed something really obvious in my understanding of moveit, I've got the latest melodic release of moveit.

To try rectify it I've actually now successfully implemented publishing of /joint_states from Unity which seems to do the trick and I've got the pick and place to work, however I'm still very curious as to why I've had to do this so any thoughts would be appreciated.

hyounesy, 2021-05-24:
Thank you for sharing your findings @lberridge1992. That's strange indeed, and as you mentioned, the only publisher to the `/joint_states` is the `/niryo_one_driver` which may only be needed for part_4 tutorial:

```
> rostopic info /joint_states
Type: sensor_msgs/JointState

Publishers:
 * /niryo_one_driver (http://3a743fc236c1:42373/)

Subscribers:
 * /robot_state_publisher (http://3a743fc236c1:45225/)
 * /move_group (http://3a743fc236c1:35673/)
```

Is it possible that you may have local modifications in the launch file or python scripts, specifically`part_3.launch`, `mover.py`, or `demo.launch`?

Can you also compare the info for `server_endpoint` with the following (before the addition of the publisher to `/joint_states`?

```
> rosnode info /server_endpoint
--------------------------------------------------------------------------------
Node [/server_endpoint]
Publications:
 * /SourceDestination [niryo_moveit/NiryoMoveitJoints]
 * /rosout [rosgraph_msgs/Log]
 * /sim_real_pnp [niryo_moveit/MoverServiceRequest]

Subscriptions:
 * /NiryoTrajectory [unknown type]
 * /niryo_one/commander/robot_action/goal [unknown type]

Services:
 * /server_endpoint/get_loggers
 * /server_endpoint/set_logger_level

contacting node http://3a743fc236c1:33157/ ...
Pid: 344
Connections:
 * topic: /rosout
    * to: /rosout
    * direction: outbound (43441 - 172.17.0.2:57308) [8]
    * transport: TCPROS
```

sdiao, 2021-06-02:
Hi @lberridge1992 , we just released several ROSConnection improvements in [ROS-TCP-Connector v0.4.0](https://github.com/Unity-Technologies/ROS-TCP-Connector/tree/v0.4.0) . We have simplified the ROS Setting window in Unity Editor, removed "Override Unity IP", and removed the Handshake. I would recommend you to check out the latest features with the pick and place tutorial in[ Unity-Robotics-Hub](https://github.com/Unity-Technologies/Unity-Robotics-Hub/tree/v0.4.0). Please let us know if there is anything else we can help with. Thank you.

hyounesy, 2021-07-08:
I am closing this issue hoping that the new release has solved the issue. Please feel free to open a new issue if the problem continues, or you encounter a problem.

closed 2021-07-08 by hyounesy