Unity-Technologies/Unity-Robotics-Hub #226: Drift in articulation drive due to gravity when continuously assigning targets

m-talha, 2021-05-20:
I believe there is a problem with continuous assignment of targets to an articulation drive using the `xDrive.target` method:

```
for (int i = 0; i < _joints.Count; i++)
{
    var drive = _joints[i].xDrive;
    drive.target = result[i];
    _joints[i].xDrive = drive;
}
```

When target values for the joints are updated continuously through ROS, it results in the robot drifting towards the ground. I assume this is due to the drive being disabled during target assignment but gravity remaining active.

Is there a better way to do this?

hyounesy, 2021-05-20:
Hi @m-talha , we suspect this might be caused by the settings in the articulation body. What values are you using for your articulation body settings, specifically the damping and stiffness? In your case you may want to try using high stiffness and zero or low damping.

m-talha, 2021-05-20:
Thanks for your reply @hyounesy. 

My values for the joints are `stiffness = 10000` and `damping = 100`. 

I forgot to add, disabling `gravity` on the URDF robot script alleviates the issue so I'm quite sure it has something to do with gravity.

hyounesy, 2021-05-22:
internal ticket # AIRO-699

peifeng-unity, 2021-05-24:
Hi @m-talha, would you mind sharing your whole scripts or project so that we could have a better understanding of what does not look right to you? 

In Unity, the forward dynamics are computed after the assignment of the target of the articulation body, so the gravity will not affect the articulation body during the assignment if the above script is executed in the `FixedUpdate()` method. There might be some other factors relates to the issue. It would be very helpful for us to debug if you could share the project.

peifeng-unity, 2021-09-07:
I am closing this issue since it has been inactive for a few months. Please feel free to reopen it if you have more questions.

closed 2021-09-07 by peifeng-unity