Unity-Technologies/Unity-Robotics-Hub #230: Address is already in used

Albert950207, 2021-05-22:
Hello

In the part 2 and part3, I get an error reported by unity. The error is:
[image]
Address is already in used 
I run ROS in docker according to the part0 and my unity version is 2020.3.6f1 and I also try it in different versions but all get the same error.
Here is my IP address setting:
[image]

When I was at part 2 and got this error, I can still get the data from docker as the part shows if I turn off error pause even there is still the error "Address is already in used":
[image]
[image]

However, when I was at part 3, after I played the game and pressed the publisher, there was another error except for "Address is already in used":
[image]
and here is docker after pressed publish:
[image]

- Unity Version: 2020.3.6f1
- Unity machine OS + version: Ubuntu 20.04
- ROS–Unity communication: Docker

hyounesy, 2021-05-22:
Hi @Albert950207 , I think the issue might be due to the `Override Unity IP Address` in the Unity ROS Settings not being set to the IP Address of the unity machine. Can you please replace `127.0.0.1` with the actual IP of your unity machine and try again.

Albert950207, 2021-05-24:
Hi@hyounesy  Thank you for your solution. I tried to change the override unity IP but every time after I play the game, the IP will be reset to 127.0.0.1. What should I do ?

hyounesy, 2021-05-24:
Could you please try closing the ROS Settings window after you set the IP Address, and then in the UnityEditor window run File -> Save Project.

Albert950207, 2021-05-24:
@hyounesy  Thanks!  I replace the override unity IP address with  actual IP of my unity machine but it doesn't work. The errors in unity and dork are same as the previous override IP.

Albert950207, 2021-05-27:
Does anyone know what leads to this error?

hyounesy, 2021-05-28:
Apologies for the late response. Could you please verify the following:
- The Unity IP address you use is your "local" IP address (usually starts with 192.xxx.xxx.xxx) and not your external (public) IP assigned by the internet service provider.
- In the ROS window, the IP that is shown after "ROS-Unity Handshake received, will connect to 192.xxx.xxx.xxx:5005" matches your local Unity IP.
- In the Unity Editor, after you play the scene, click on the "View service status" checkbox and the IPs shown under MUnityHandshakeRequest and MUnityHandshakeResponse match your local Unity IP.

sdiao, 2021-06-02:
Hi @Albert950207 , We just released several ROSConnection improvements in [ROS-TCP-Connector v0.4.0](https://github.com/Unity-Technologies/ROS-TCP-Connector/tree/v0.4.0) . We have simplified the ROS Setting window in Unity Editor, removed "Override Unity IP", and removed the Handshake. I would recommend you to check out the latest features with the pick and place tutorial in[ Unity-Robotics-Hub](https://github.com/Unity-Technologies/Unity-Robotics-Hub/tree/v0.4.0). Please let us know if there is anything else we can help with. Thank you.

Albert950207, 2021-06-05:
Hi @hyounesy, thanks for your replying  
I have tried the way and here is my ROS setting and my local IP address 
1. [image]
2. In docker:
[image]

[image]

3. Service status after play
[image]

[image]

[image]

but still get error: address is already in used

Albert950207, 2021-06-05:
Hi@sdiao, thank you for letting me know there is a new version of ROS. I will try it !

Albert950207, 2021-06-14:
Hi @sdiao , I just checked out the new ROS-TCP connector. I update my order one in this project with the new connector by package manager in Unity. However, I got these errors.

[image]

Could you please help me ? Is there any instructions about how to use it in pick and place project ?

Thank you !

sdiao, 2021-06-22:
Hi @Albert950207 , the compile error was due to an interface change in `ROSConnection`. fields including `overrideUnityIP`, `unityPort`, `awaitDataMaxRetries`,  and `awaitDataSleepSeconds` are deprecated. 
We would recommend you get the latest code from the main branch in the Unity-Robotics-Hub, where we have updated the `Demo.cs` script to use the updated ROSConnection. Please see: https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/main/tutorials/pick_and_place/PickAndPlaceProject/Assets/DemoScripts/Demo.cs#L165-L172 which shows the method `CreateRosConnection` should be:
```private void CreateRosConnection()
    {
        // Create RosConnect
        GameObject rosConnect = new GameObject(rosConnectName);
        rosConnection = rosConnect.AddComponent<ROSConnection>();
        rosConnection.RosIPAddress = hostIP;
        rosConnection.RosPort = hostPort;
    }
    ```

Let us know if the tutorial works after checking out the latest code on main branch.

Albert950207, 2021-06-30:
Hi, @sdiao  I will the latest code on main branch. I didn't get the " address is already in used " error anymore but the arm didn't move and I still got the error "no trajectory returned from movie service"

Albert950207, 2021-06-30:
*tried

Albert950207, 2021-07-01:
This is error showed by my docker container
[image]

These are information provided by Unity
[image]
[image]
[image]

I can run part 2 and received the positions.

sdiao, 2021-07-01:
We are currently investigating the issue and will keep you posted. Just to confirm, is the error from the quick demo tutorial: https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/main/tutorials/pick_and_place/quick_demo.md?

Albert950207, 2021-07-01:
@sdiao  Yes, it is from the pick and place demo and I rebuild my docker image with latest code but I was having the old one when I rebuild. I am not sure if it will cover the old one.

sdiao, 2021-07-08:
@Albert950207 , you can use `docker container ls` to list the containers, and use `docker rm [OPTIONS] CONTAINER` to remove the pick and place container, then rebuild the docker image with `cd /PATH/TO/Unity-Robotics-Hub/tutorials/pick_and_place &&
git submodule update --init --recursive &&
docker build -t unity-robotics:pick-and-place -f docker/Dockerfile .`

It seems some part of the tutorial is out of sync with the latest main branch for your local environment. I would recommend that you delete the Unity-Robotics-Hub repo first and download it again, re-build docker images and set up the Unity project following the tutorial. 
Please make sure to use the git submodule when you clone the repo `git clone --recurse-submodules https://github.com/Unity-Technologies/Unity-Robotics-Hub.git`. 

Hope it helps. Let us know if removing&rebuilding docker container or starting the tutorial from scratch works. Thank you.

Albert950207, 2021-07-09:
@sdiao  Hi, thank you for your suggestions. The things are that when I exit the docker container, I can't see it anymore by using 'docker container ls' so I just download the latest repository into a new directory and rebuild the docker image according to the tutorials. However, it still doesn't work and same errors are returned.

[image]

[image]

sdiao, 2021-07-12:
Hi @Albert950207 , Let's set up a 30 minutes meeting to troubleshoot the issue. Could you send 3 of your availabilities to unity-robotics@unity3d.com? The Robotics team is in Pacific Daylight Timezone (PDT). Thanks!

Albert950207, 2021-07-15:
@sdiao Yes, sure! I have sent it just right now

sdiao, 2021-07-26:
@Albert950207 , I just wanted to follow up and see if you are able to get the tutorial part 2 and part3 working after the meeting. Let us know. Thanks.

sdiao, 2021-08-02:
I am going to close the issue now since we have been able to run the quick demo in the meeting. Please feel free to reach out if you have any questions in the future. Thanks!

closed 2021-08-02 by sdiao

NathanielRose, 2021-09-22:
Following up on this thread, It looks like I receive an Identical issue when I switch between scenes  with an established ROS connection. What were the outcomes of the meeting that mitigated this error @sdiao? I have an open issue #314  that seeks support on this.

sdiao, 2021-09-22:
Hi @NathanielRose , for the meeting, we tried the Pick and Place [quick demo ](https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/v0.4.0/tutorials/pick_and_place/quick_demo.md) , and verified that demo project works for the customer. 
We suspected the customer's local environment did not have a clean setup. Starting the tutorial from scratch solved the connection issue. 
Let's move the discussion to the new issue that you opened.