Unity-Technologies/Unity-Robotics-Hub #236: Arm wobble runing Unity 2020.3.9f1 (LTS)

iconnor, 2021-05-27:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Follow the (quick demo](https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/8af91808c5c19a390b7b9c9b66a73fa2ed7b9859/tutorials/pick_and_place/quick_demo.md)
2. Click Play
3. Watch robot arm move and wobble - I assume it should be still

**Console logs / stack traces**
```
root@f2b22363c7d5:/catkin_ws# roslaunch niryo_moveit part_3.launch
... logging to /root/.ros/log/b50d54b0-beaa-11eb-acc6-0242ac110002/roslaunch-f2b22363c7d5-37.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
xacro: in-order processing became default in ROS Melodic. You can drop the option.
started roslaunch server http://f2b22363c7d5:40113/

SUMMARY
========

PARAMETERS
 * /ROS_IP: 172.17.0.2
 * /hardware_control_loop/loop_hz: 25
 * /joint_state_controller/publish_rate: 50
 * /joint_state_controller/type: joint_state_contr...
 * /move_group/allow_trajectory_execution: True
 * /move_group/arm/enforce_joint_model_state_space: True
 * /move_group/arm/longest_valid_segment_fraction: 0.005
 * /move_group/arm/planner_configs: ['SBLkConfigDefau...
 * /move_group/arm/projection_evaluator: joints(joint_1,jo...
 * /move_group/controller_list: [{'default': True...
 * /move_group/jiggle_fraction: 0.05
 * /move_group/max_safe_path_cost: 1
 * /move_group/moveit_controller_manager: moveit_simple_con...
 * /move_group/moveit_manage_controllers: True
 * /move_group/planner_configs/BFMTkConfigDefault/balanced: 0
 * /move_group/planner_configs/BFMTkConfigDefault/cache_cc: 1
 * /move_group/planner_configs/BFMTkConfigDefault/extended_fmt: 1
 * /move_group/planner_configs/BFMTkConfigDefault/heuristics: 1
 * /move_group/planner_configs/BFMTkConfigDefault/nearest_k: 1
 * /move_group/planner_configs/BFMTkConfigDefault/num_samples: 1000
 * /move_group/planner_configs/BFMTkConfigDefault/optimality: 1
 * /move_group/planner_configs/BFMTkConfigDefault/radius_multiplier: 1.0
 * /move_group/planner_configs/BFMTkConfigDefault/type: geometric::BFMT
 * /move_group/planner_configs/BKPIECEkConfigDefault/border_fraction: 0.9
 * /move_group/planner_configs/BKPIECEkConfigDefault/failed_expansion_score_factor: 0.5
 * /move_group/planner_configs/BKPIECEkConfigDefault/min_valid_path_fraction: 0.5
 * /move_group/planner_configs/BKPIECEkConfigDefault/range: 0.0
 * /move_group/planner_configs/BKPIECEkConfigDefault/type: geometric::BKPIECE
 * /move_group/planner_configs/BiESTkConfigDefault/range: 0.0
 * /move_group/planner_configs/BiESTkConfigDefault/type: geometric::BiEST
 * /move_group/planner_configs/BiTRRTkConfigDefault/cost_threshold: 1e300
 * /move_group/planner_configs/BiTRRTkConfigDefault/frountier_node_ratio: 0.1
 * /move_group/planner_configs/BiTRRTkConfigDefault/frountier_threshold: 0.0
 * /move_group/planner_configs/BiTRRTkConfigDefault/init_temperature: 100
 * /move_group/planner_configs/BiTRRTkConfigDefault/range: 0.0
 * /move_group/planner_configs/BiTRRTkConfigDefault/temp_change_factor: 0.1
 * /move_group/planner_configs/BiTRRTkConfigDefault/type: geometric::BiTRRT
 * /move_group/planner_configs/ESTkConfigDefault/goal_bias: 0.05
 * /move_group/planner_configs/ESTkConfigDefault/range: 0.0
 * /move_group/planner_configs/ESTkConfigDefault/type: geometric::EST
 * /move_group/planner_configs/FMTkConfigDefault/cache_cc: 1
 * /move_group/planner_configs/FMTkConfigDefault/extended_fmt: 1
 * /move_group/planner_configs/FMTkConfigDefault/heuristics: 0
 * /move_group/planner_configs/FMTkConfigDefault/nearest_k: 1
 * /move_group/planner_configs/FMTkConfigDefault/num_samples: 1000
[... 423 lines cut ...]
System.Runtime.CompilerServices.AsyncVoidMethodBuilder:Start<Unity.Robotics.ROSTCPConnector.ROSConnection/<StartMessageServer>d__43> (Unity.Robotics.ROSTCPConnector.ROSConnection/<StartMessageServer>d__43&)
Unity.Robotics.ROSTCPConnector.ROSConnection:StartMessageServer (string,int)
Unity.Robotics.ROSTCPConnector.ROSConnection:Start () (at Library/PackageCache/com.unity.robotics.ros-tcp-connector@8ee9d5c391/Runtime/TcpConnector/ROSConnection.cs:239)

Trajectory returned.
UnityEngine.Debug:Log (object)
TrajectoryPlanner:TrajectoryResponse (RosMessageTypes.NiryoMoveit.MMoverServiceResponse)
Unity.Robotics.ROSTCPConnector.ROSConnection/<SendServiceMessage>d__24`1<RosMessageTypes.NiryoMoveit.MMoverServiceResponse>:MoveNext () (at Library/PackageCache/com.unity.robotics.ros-tcp-connector@8ee9d5c391/Runtime/TcpConnector/ROSConnection.cs:166)
System.Runtime.CompilerServices.AsyncTaskMethodBuilder`1<System.Tuple`2<string, byte[]>>:SetResult (System.Tuple`2<string, byte[]>)
Unity.Robotics.ROSTCPConnector.ROSConnection/<ReadMessageContents>d__40:MoveNext () (at Library/PackageCache/com.unity.robotics.ros-tcp-connector@8ee9d5c391/Runtime/TcpConnector/ROSConnection.cs:382)
UnityEngine.UnitySynchronizationContext:ExecuteTasks () (at /Users/bokken/buildslave/unity/build/Runtime/Export/Scripting/UnitySynchronizationContext.cs:107)

```

**Environment (please complete the following information, where applicable):**
- Unity Version: Unity 2020.3.9f1
- Unity machine OSX 11.4
- ROS machine ros:melodic-ros-base
- ROS–Unity communication: 
  - docker run -it --rm -p 10000:10000 -p 5005:5005 unity-robotics:pick-and-place /bin/bash
  - roslaunch niryo_moveit part_3.launch
- Branch or version: [e.g. v0.2.0]

**Additional context**
Was going to try 2020.2.0f1 next or Windows 10 as a test unless this is a known bug.

iconnor, 2021-05-27:
I tested on Windows 10 with Unity 2020.3.9f1, and it missed the block and wobbled also. I tried 2020.2.0f1, and it worked just fine. I would be keen to learn why it broke between builds.

peifeng-unity, 2021-05-27:
Hello @iconnor, thank you for bringing it up. This is a known issue and it is caused by some changes in the physics engine between the versions 2020.3.0f1 and 2020.3.1f1. We are actively working on exploring the root cause and will let you know once we have progress.

peifeng-unity, 2021-07-13:
Hello @iconnor, we have figured out the physics bug related to this issue. The fix has been deployed to the following Unity versions:
* 2021.2.0a16+
* 2021.1.8f1+
* 2020.3.10f1+

Please feel free to try the working versions and let us know if you have any further questions.

iconnor, 2021-07-27:
I just tested and it works for me with the 2020 LTS build - thanks!

closed 2021-07-27 by iconnor