Unity-Technologies/Unity-Robotics-Hub #248: Publishing images to ROS node

marcociara379, 2021-06-17:
Hello everyone, 
I want to publish the feed of the main camera in Unity, but it seems I cannot receive it properly in ROS: when I run the TCP server in ROS, with the image publisher running on Unity, I get the following trace:
```
[INFO] [1623947489.663173]: Disconnected from 172.30.224.1
Exception in thread Thread-11:
Traceback (most recent call last):
  File "/usr/lib/python3.8/threading.py", line 932, in _bootstrap_inner
    self.run()
  File "/home/marco/noetic_ws/src/ROS-TCP-Endpoint/src/ros_tcp_endpoint/client.py", line 216, in run
    response = ros_communicator.send(data)
  File "/home/marco/noetic_ws/src/ROS-TCP-Endpoint/src/ros_tcp_endpoint/subscriber.py", line 56, in send
    self.tcp_server.send_unity_message(self.topic, data)
  File "/home/marco/noetic_ws/src/ROS-TCP-Endpoint/src/ros_tcp_endpoint/server.py", line 96, in send_unity_message
    self.unity_tcp_sender.send_unity_message(topic, message)
  File "/home/marco/noetic_ws/src/ROS-TCP-Endpoint/src/ros_tcp_endpoint/tcp_sender.py", line 58, in send_unity_message
    serialized_message = ClientThread.serialize_message(topic, message)
  File "/home/marco/noetic_ws/src/ROS-TCP-Endpoint/src/ros_tcp_endpoint/client.py", line 134, in serialize_message
    message.serialize(serial_response)
AttributeError: 'bytes' object has no attribute 'serialize'
```

This is the file I use to publish the camera image:
```csharp
//using System.Collections;
//using System.Collections.Generic;
using UnityEngine;
using UnityEngine.Rendering;
//using RosMessageTypes.RoboticsDemo;
using Unity.Robotics.ROSTCPConnector;
using UnityEngine.Rendering.Universal;
using CompressedImage = RosMessageTypes.Sensor.MCompressedImage;

public class RosCameraPublisher : MonoBehaviour
{
    ROSConnection ros;
    public string topicName = "RGB_camera_image";
    public float publishMessageFrequency = 2.0f;
    private float timeElapsed; // Used to determine how much time has elapsed since the last message was published
    private Camera RGBcamera;
    Texture2D camTexture;

    void SendImage(string topic_name)
    {
        // Encode the texture as a PNG, send via ROS
        byte[] imageBytes = camTexture.EncodeToPNG();
        var message = new CompressedImage(new RosMessageTypes.Std.MHeader(), "png", imageBytes);
        ros.Send(topicName, message);
    }

    void SaveImageToDisk(Texture2D texture, string fullPath)
    {
        byte[] imageBytes = texture.EncodeToPNG();
        Debug.Log("Image saved." + texture.ToString());
        System.IO.File.WriteAllBytes(fullPath, imageBytes);
    }

    void OnEndFrameRendering(ScriptableRenderContext context, Camera[] cameras)
    {
        //var oldRenderTexture = RenderTexture.active;
        var oldRenderTexture = RenderTexture.GetTemporary(Screen.width, Screen.height);
        RGBcamera.targetTexture = oldRenderTexture;
        RenderTexture.active = RGBcamera.targetTexture;
        UniversalRenderPipeline.RenderSingleCamera(context, RGBcamera);
        //Create a new texture with the width and height of the screen
        camTexture = new Texture2D(RGBcamera.targetTexture.width, RGBcamera.targetTexture.height);
        //Read the pixels in the Rect starting at 0,0 and ending at the screen's width and height
        camTexture.ReadPixels(new Rect(0, 0, RGBcamera.targetTexture.width, RGBcamera.targetTexture.height), 0, 0, false);
        camTexture.Apply();
        RenderTexture.active = oldRenderTexture;
    }

    void OnDestroy()
    {
        RenderPipelineManager.endFrameRendering -= OnEndFrameRendering;
    }

    // Start is called before the first frame update
    void Start()
    {
        // start the ROS connection
        ros = ROSConnection.instance;
        RGBcamera = Camera.main;
        RenderPipelineManager.endFrameRendering += OnEndFrameRendering;
    }

    //Update is called once per frame
    void Update()
    {
        timeElapsed += Time.deltaTime;

        if (timeElapsed > publishMessageFrequency)
        {
            //SaveImageToDisk(camTexture, "c:\\users\\marco\\pictures\\UnityTrial.jpg");
            SendImage(topicName);
            timeElapsed = 0;
        }
    }
}
```

whereas this is the python script for the ROS TCP server:
```python
#!/usr/bin/env python

import rospy

from ros_tcp_endpoint import TcpServer, RosPublisher, RosSubscriber, RosService, UnityService
# from robotics_demo.msg import PosRot, UnityColor
# from robotics_demo.srv import PositionService, ObjectPoseService
from sensor_msgs.msg import CompressedImage

def main():
    ros_node_name = rospy.get_param("/TCP_NODE_NAME", 'TCPServer')
    buffer_size = rospy.get_param("/TCP_BUFFER_SIZE", 2048)
    connections = rospy.get_param("/TCP_CONNECTIONS", 10)
    tcp_server = TcpServer(ros_node_name, buffer_size, connections)
    rospy.init_node(ros_node_name, anonymous=True)
    
    tcp_server.start({
        # 'pos_rot': RosPublisher('pos_rot', PosRot, queue_size=10),
        # 'color': RosSubscriber('color', UnityColor, tcp_server),
        # 'pos_srv': RosService('pos_srv', PositionService),
        # 'obj_pose_srv': UnityService('obj_pose_srv', ObjectPoseService, tcp_server),
        'RGB_camera_image': RosSubscriber('RGB_camera_image', CompressedImage, tcp_server),
    })
    
    rospy.spin()

if __name__ == "__main__":
    main()
```

Do you have any idea to share?

Much appreciated.

Thank you,
Marco

Environment:
- Unity Version: 2020.2.7f1
- Machine: Unity Windows 10, ROS running on WSL2

LaurieCheers-unity, 2021-06-17:
Hmm, I see. We should handle this case better.
You're trying to publish a message from Unity, but in your endpoint you've declared Unity is only subscribing to that message. Try replacing your RosSubscriber line with:

    'RGB_camera_image': RosPublisher('RGB_camera_image', CompressedImage, queue_size=10),

marcociara379, 2021-06-21:
whoops, that was a silly one! 
Thank you a lot Laurie. 
I have been able to properly subscribe to the ROS server and get the images. 
I have another issue, but I will open another thread to keep this clean.

closed 2021-06-21 by marcociara379