Unity-Technologies/Unity-Robotics-Hub #252: Position Control Stiffness, Damping and Force Limit Values

bryceikeda, 2021-07-01:
**Describe the bug**
I am in the process of simulating a Fetch robot using the pick and place tutorial 4 with move action goals. Although I have its arm moving, I can't figure out the best stiffness, damping and force limit values for smooth motion. For force limit, I have it set to 450 because that is the highest force limit of one of all the joints I control. Looking at the [https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/main/tutorials/urdf_importer/urdf_appendix.md](write your own controller guide), I'm not sure how best to use the Force equation to figure out my stiffness and damping. 

I'm wondering if there is a good way to figuring out these values besides trial and error?

**Environment (please complete the following information, where applicable):**
- Unity Version: Unity 2020.3.8f1
- Unity machine OS + version: Windows 10
- ROS machine OS + version: Ubuntu 18.04, ROS Melodic
- ROS–Unity communication: Manual
- Branch or version: v0.4.0

vidurvij-Unity, 2021-07-01:
As described in the guide, the Articulation Bodies work with a logic of spring-damper system. With the stiffness delegating the stiffness of the spring attached to the joint, higher the stiffness, proportionally larger the restoring force. Similarly damping calculates the restoring force based on speed differential. You can think of stiffness and damping in terms of a P-D controller, where you can use the stiffness to set the proportional constant in the position controller in the xdrive of the articulation body and damping as the derivative constant.

https://www.youtube.com/watch?v=T9_C3jhnQXI

bryceikeda, 2021-07-02:
Ah ok it makes sense, thanks for the video reference. So basically I increment the stiffness until I get whatever the desired position will be. Then I tune the damping to reduce the overshoot/reduce the speed.

closed 2021-07-02 by bryceikeda