Unity-Technologies/Unity-Robotics-Hub #254: Can't build a the project from Unity for Universal Windows Platform

SMRazaRizvi96, 2021-07-12:
I created a Unity project and used the ROS-TCP-Connector package to communicate between Unity and ROS.

I followed the first three steps from here: https://github.com/Unity-Technologi...s-2/tutorials/ros_unity_integration/README.md

It works fine on Unity and messages are sent from Unity to ROS, but as soon as I build the project from Unity, to open it in Visual Studio, it gives me the following errors in Unity:

upload_2021-7-12_15-14-59.png
upload_2021-7-12_15-15-19.png
upload_2021-7-12_15-15-36.png

These are the build settings I am using:

upload_2021-7-12_15-14-23.png
[image]
[image]
[image]
[image]

hyounesy, 2021-07-12:
Hi, Thanks for bringing this to our attention. We will look into this error and get back to you with a potential solution.

Ticket#: AIRO-974

SMRazaRizvi96, 2021-07-13:
I solved the issue.

Solution: By default, the Universal Windows Platform is not checked in the 'Unity.Robotics.ROSTCPConnector.asmdef' file inside the ROS-TCP Connector Package, and the file is locked and cannot be edited from Unity. So, I opened the file locally on my laptop on the Notepad and added the line "WSA" under the supported platforms (Can be seen in the image attached below).

After adding this line, I saved the file, recompiled the Unity project, AND THE BUILD WORKED!!

[image]

peifeng-unity, 2021-07-13:
Thanks, @SMRazaRizvi96! It is so great to know that you figured it out. It is good for us to learn and we will update the supporting platforms.

sdiao, 2021-08-02:
Support for windows build is merged in https://github.com/Unity-Technologies/URDF-Importer/pull/124. Close this issue now.

closed 2021-08-02 by sdiao

SMRazaRizvi96, 2021-08-04:
Thank you so much for the support and quick actions.