Unity-Technologies/Unity-Robotics-Hub #262: URDF Robot Base Joint Falls or Falls Apart

Pterodactylus, 2021-07-18:
**Describe the bug**
When I import a generic URDF file the link above the base link falls through the simulation or falls off to the side. I've replicated this with more than one urdf file. Any help would be appreciated.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a new bland scene
2. Go to attached pi_robot.urdf file and import the robot.
3. Click the play button to start the scene

**Expected behavior**
Robot should hold together link in the scene window.
[image]

**Actual undesired behavior**
Robot falls through the base link like this.
[image]

**Environment (please complete the following information, where applicable):**
- Unity Version: Unity 2020.3.14f1 & 2021.2.0b4
- Unity machine OS + version: Windows 10 & Ubuntu 18.04
- ROS machine OS + version: N/A
- ROS–Unity communication: N/A
- Branch or version: N/A

**URDF File**
`<robot name="pi_robot">

	

 	<link name="base_link">
		<visual>
		    <origin xyz="0 0 0.0425" rpy="0 0 0"/>
			<geometry>
				<box size="0.32 0.26 0.085"/>
			</geometry>
			<material name="Cyan1">
	       		<color rgba="0 0.9 0.9 1.0"/>
	     	</material>
		</visual>	
	</link>
	
 	<link name="base_laser">
		<visual>
		    <origin xyz="0 0 0" rpy="0 0 0"/>
			<geometry>
				<cylinder radius="0.025" length="0.07"/>
			</geometry>
			<material name="Black1">
	       		<color rgba="0.2 0.2 0.2 1.0"/>
	     	</material>
		</visual>	
	</link>
	
	<link name="cpu_link">
		<visual>
		    <origin xyz="0 0 0.035" rpy="0 0 0"/>
			<geometry>
				<box size="0.19 0.19 0.07"/>
			</geometry>
			<material name="Cyan2">
	       		<color rgba="0 0.7 0.7 1.0"/>
	     	</material>
		</visual>	
	</link>

  	<link name="upper_base_link">
		<visual>
			<origin xyz="0 0 0.05" rpy="0 0 0"/>
			<geometry>
				<cylinder radius="0.085" length="0.10"/>
[... 424 lines cut ...]
       	<limit lower="-1.57" upper="1.57" effort="10" velocity="3"/>
        <axis xyz="1 0 0"/>
	</joint>
	
 	<joint name="right_wrist_joint" type="revolute">
    	<parent link="right_lower_arm_link"/>
    	<child link="right_wrist_link"/>
    	<origin xyz="0 0 -0.05" rpy="0 0 0"/>
       	<limit lower="-1.57" upper="1.57" effort="10" velocity="3"/>
        <axis xyz="1 0 0"/>
	</joint>
	
 	<joint name="left_hand_joint" type="fixed">
    	<parent link="left_wrist_link"/>
    	<child link="left_hand_link"/>
    	<origin xyz="0 0 -0.055" rpy="0 0 0"/>
	</joint>
	
 	<joint name="right_hand_joint" type="fixed">
    	<parent link="right_wrist_link"/>
    	<child link="right_hand_link"/>
    	<origin xyz="0 0 -0.055" rpy="0 0 0"/>
	</joint>
 
</robot>`

peifeng-unity, 2021-07-19:
Hello @Pterodactylus, I have created an internal ticket AIRO-1006 for this request. We will take a look and reply soon.

In the meantime, I wish to double-check if you have set the parameters such as the `stiffness` and `damping` in the controller component of the robot game object? (like Step 4 and 5 in [this tutorial section](https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/main/tutorials/pick_and_place/1_urdf.md#setting-up-the-robot))

Pterodactylus, 2021-07-20:
I have reviewed the tutorial and set the parameters as described in steps 4 & 5. 

For step 4 it makes absolutely no difference what the parameters are when I run it:
[image]

For step 5 it seems to keep the robot from falling apart which solves one problem. However, it creates another by locking the joint above the base joint so that it cannot move. Some urdf files have the joint next to the base as a motive joint.
[image]
[image]

at669, 2021-08-31:
Closing this issue due to inactivity. Please feel free to reopen if the problem persists!

ryangoh206, 2021-11-12:
Hi I have faced this problem as well, I'm trying to import the spot robot. But, when the scene is ran, all the links fall apart.

at669, 2021-11-12:
Hi @ryangoh206, can you describe your issue using the bug report template below so we can take a look into the issue? Thanks!

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Console logs / stack traces**
Please wrap in [triple backticks (```)](https://help.github.com/en/articles/creating-and-highlighting-code-blocks) to make it easier to read.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots or videos to help explain your problem.

**Environment (please complete the following information, where applicable):**
- Unity Version: [e.g. Unity 2020.2.0f1]
- Unity machine OS + version: [e.g. Windows 10]
- ROS machine OS + version: [e.g. Ubuntu 18.04, ROS Noetic]
- ROS–Unity communication: [e.g. Docker]
- Branch or version: [e.g. v0.2.0]

**Additional context**
Add any other context about the problem here.

ryangoh206, 2021-11-15:
**Description of Bug**
I have imported the spot and husky robot from their respective URDF files found in their repositories.
Spot: [Link](https://github.com/clearpathrobotics/spot_ros)
Husky: [Link](https://github.com/husky/husky)
However, when I run the scene, all the links fall apart as seen in the image below (attached image shows spot robot but similar behavior is observed for husky robot as well).
[image]

**To reproduce**
1. Create a new blank scene
2. Download the attached zip file (contains the meshes and urdf for spot)
3. Extract the file and place the contents in the project
4. Use the URDF Importer Tool to import the spot.urdf file as per normal
5. Play the scene and observe the behavior of the robot (links should fall apart)

**Console logs / stack traces**
None

**Expected Behavior**
The expected behavior of the robot is for all the links to not fall apart after the scene is played and hopefully to be able to use the default controller attached to control the moving of the different joints. See image below. _Note that the image is taken from before the scene is played but accurately represents the expected behavior._
[image]

**Environment**
- Unity Version: 2020.3.18f1
- Unity machine OS + version: Ubuntu 20.04.3 LTS
- ROS machine OS + version: Ubuntu 20.04.3 LTS, ROS Noetic (However, note that this problem exists even as ROS is not connected to Unity yet. Only the xacro package in ROS is used to convert the spot.urdf.xacro to spot.urdf)
- ROS-Unity communication: -
- Branch or version: Importer version 0.5.0

**Additional Context**
I am relatively new to ROS and integrating it with Unity. Hence, it may be due to a small issue that I do not know about.

Thank you in advance!

rshanor, 2021-12-02:
@ryangoh206 I am also having issues importing the spot URDF. I am able to import the robot, but when I hit play, all the limbs fall off the body. The legs themselves seems to be coupled correctly, but not correctly coupled to the body. I am able to make a bit of progress by setting all the hips to `Immovable` and editing the parameters in the controller script. Once I do this, I am able to articulate the two of the three leg joints via the provided script, but I am not able to move one of the hip degrees of freedom.

peifeng-unity, 2021-12-07:
Hello @ryangoh206 and @rshanor, thanks for reporting the issue. I am able to reproduce the issue on my laptop, and the reason for the problem is the hierarchy of links defined in the URDF file does not match what the URDF-Importer expected. 

The current hierarchy of the links are:
```
- spot
  - body
    - base_link
    - front_right_hip
    - front_left_hip
    ...
```
But the importer expects:
```
- spot
  - body
    - base_link
      - front_right_hip
      - front_left_hip
      ...
```

The mismatch of the hierarchy makes the `ArticulationBody` component missing on the `body` object so that the `hip` objects will be treated as the root links of the robot, which causes the issue of falling parts. 

In addition, the geometry meshes and visuals are assigned to `body` instead of `base_link`, which will also cause the mismatching problem because there is no `ArticulationBody` component assigned to the `body` object.

To make it work, there are two places that need to be modified in the URDF file:
1. Change joint parent from `body` to `base_link` for the `hip` joints and `rail` joints
2. Move the visual and collision geometry from the `body` link to the `base_link` link.

I have tested the attached URDF file and it should work. Please let us know if you have any questions.

[spot.urdf.zip](https://github.com/Unity-Technologies/Unity-Robotics-Hub/files/7671579/spot.urdf.zip)

ryangoh206, 2021-12-20:
Hi @peifeng-unity,
Thanks! I have tried it out and it works.

at669, 2022-01-06:
Closing this out, as the issue has been resolved.

closed 2022-01-06 by at669

bryceikeda, 2022-01-19:
I created a quick video tutorial on how to get the spot moving using Unity and ROS. 

https://www.youtube.com/watch?v=3MYeBxFVBHI 

Let me know if it helps and if you have any suggestions for improvement. In the description, I added an edit on how I get the spot movement smoother.

ryangoh206, 2022-01-19:
> 
Ok thanks @bryceikeda ! I will take a look at the video.