Unity-Technologies/Unity-Robotics-Hub #263: TrajectoryPlanner.cs isn't actually using current joint angles in Tutorial

sarika93, 2021-07-20:
The [CurrentJointConfig](https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/main/tutorials/pick_and_place/Scripts/TrajectoryPlanner.cs#L81) function in TrajectoryPlanner.cs doesn't actually get the current joint angles. It instead gets the current Target. 

This doesn't break the implementation because the intended use is for a user to only use this function when the robot is already at some target. However, having the current joint angles updated with `joints.joint_00 = jointArticulationBodies[0].jointPositions[0] * Mathf.Rad2Deg;` instead of `joints.joint_00 = jointArticulationBodies[0].xDrive.target;` would be less misleading to someone using this tutorial as a guide. The xDrive.target value and jointPositions[0] won't always return the same value.

Replacing `target` with `jointPositions[0]` gives basically the same performance. Should TrajectoryPlanner.cs be updated to use `jointPositions` instead for greater clarity? If this is intentional, would it be possible to get an explanation as to why (having some controller issues, so maybe I'm misunderstanding something)?

LaurieCheers-unity, 2021-07-21:
Hi, thanks for the question. I'll look into this and get back to you.

sdiao, 2021-07-29:
We are currently reviewing the pull request.

sdiao, 2021-07-30:
@sarika93 , thanks for contributing to the repo. I have merged your pull request. I am going to close this issue.

closed 2021-07-30 by sdiao