Unity-Technologies/Unity-Robotics-Hub #273: Pick and Place Part 2 - Unity doesn't send anything

ConeRX, 2021-07-24:
Hello. I went through the Pick and Place Tutorial and did everything as described. However, after completing the second part Unity should send the coordinates of the robot to ROS (running in a cloud), which it doesn't. It connects to ROS, as seen by the fact that the connection times out after some time and it gives error (no route to host) if a wrong IP is entered. 

**Screenshots**
[image]

**Environment:**
- Unity Version: Unity 2020.3.14f1
- Unity machine OS + version: Ubuntu 20.04
- ROS machine OS + version: Ubuntu 20.04, ROS Noetic

What am I doing wrong?

ja99, 2021-07-24:
Hey,
did you try entering the local IP of your Ros instance under YOURWORKSPACE/src/ROS-TCP-Endpoint/config/params.yaml?
You can find the IP by typing "ifconfig" and taking the one starting with "192."

Then also make sure to add the same IP under your ROS connection in unity.
 

If you start up the server endpoint then make sure to do it in this order:
roscore
rosparam load YOURWORKSPACE/src/ROS-TCP-Endpoint/config/params.yaml
and then rosrun the server_endpoint

Hope this helps✌️

ConeRX, 2021-07-26:
Unfortunately, it doesn't.
I did the additional steps you described, but it doesn't change anything. It ROS and Unity connect (there's a timeout after a few moments and an error if a different IP is entered in Unity) but it doesn't send anything.

sdiao, 2021-07-27:
Hi @ConeRX , we are currently looking into the issue. Based on the screenshot you shared, my understanding is that Unity was never able to connect to your ROS host. Correct? Was there any log on ROS side?
You mentioned that your ROS is running in the cloud. Was there any VPC or Firewall setting that may prevent Unity connecting to ROS?

ConeRX, 2021-08-10:
Sorry for the late reply.
I think Unity connects to the ROS host. At least I get a connection time out after a few second if I enter the correct IP and a no route to host error if a different, wrong IP is used. There are no logs on the ROS side.

mrpropellers, 2021-08-10:
It doesn't look like it's actually fully connected to ROS from the image you've posted. It's resolved the route to the IP address, but is failing to handshake with the endpoint. If your endpoint is running on a cloud host, it's likely the default security settings are not going to allow connection traffic through arbitrary ports.  @ConeRX , could you confirm that port `10000` is being appropriately forwarded to the host running the endpoint?  When Unity has fully connected with the TCP Endpoint, you should see the endpoint print some log messages to terminal acknowledging the connection and listing any publishers or subscribers you've declared in Unity.

DiosVan, 2021-08-14:
Me too, I successfully connected, but publish nothing happens, my ROS running in local...not any error log on ROS side & unity side, but I can successfully execute Pick and Place Part 3.....I don't think there is any problem with the connection between unity and ROS.
The difference lies in calling the send<T>() method on part2, Do I have a misunderstanding?@@

at669, 2021-08-16:
Hi @DiosVan, sounds like the issue you're encountering is separate from the one in this thread--would you mind opening up a new issue for a successful connection and execution but unsuccessful logs? Thank you!

DiosVan, 2021-08-17:
hi @at669  in pick and place part 2, my situation solved by change [server_endpoint.py](https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/main/tutorials/pick_and_place/ROS/src/niryo_moveit/scripts/server_endpoint.py) RosPublisher("SourceDestination_Input",....) to RosPublisher("SourceDestination",....), then everything works fine.
[image]

ConeRX, 2021-08-20:
It doesn't print anything. The terminal on the ROS side is completly silent after starting.

I tried again, now using docker on my local machine. It still doesn't send anything, but I get output from ROS:

[INFO] [1629469835.634767]: Starting server on 172.17.0.2:10000
[INFO] [1629469923.501837]: Connection from 172.17.0.1
[ERROR] [1629469945.544651]: Exception: No more data available
[INFO] [1629469945.548387]: Disconnected from 172.17.0.1

mrpropellers, 2021-08-24:
Hey @ConeRX - thanks for following up! That's not an error I'm familiar with, so this may require some internal investigation to see if we can reproduce the issue.  Can you confirm your endpoint and connector packages are on the same version? Mysterious communication errors can sometimes be solved by pulling down the latest updates on `main` for both packages.

sabeaussan, 2021-09-06:
Hi @ConeRX, had the same issue. Solved by modifying the topic name in trajectory_subscriber scripts from "SourceDestination" to "SourceDestination_input" to match the routing in the server_endpoint script.
[image]
[image]
[image]
Unity can connect to the TCP server correctly but since the topic name does not match in the routing dictionnary, the message sent is lost.

Hope this help

peifeng-unity, 2021-09-07:
Thanks, @sabeaussan, and @DiosVan. 

@ConeRX, we just made [a PR](https://github.com/Unity-Technologies/Unity-Robotics-Hub/pull/308) fix this issue and merged it to the dev branch. Please feel free to try it and let us know if it works for you.

uninterrupted7, 2021-09-08:
> hi @at669 in pick and place part 2, my situation solved by change [...]

it work on me. thanks :)

closed 2021-10-11 by mrpropellers