Unity-Technologies/Unity-Robotics-Hub #274: Messages still get generated mentioning MTime even though it is now called TimeMsg

ja99, 2021-07-24:
**Describe the bug**
The naming scheme was changed from MMessageName to MessageNameMsg. It seems like MTime though was forgotten for the msg generator.

**To Reproduce**
Steps to reproduce the behavior:
generate own message containing Time

**Console logs / stack traces**
-Use TimeMsg and the programm tells you it only takes MTime
-Try MTime and it tells you it isn't implemented

**Expected behavior**
-

**Screenshots**
-

**Environment (please complete the following information, where applicable):**
- Unity Version: [e.g. Unity 2021.1.15f1]
- Unity machine Ubuntu18.04LTS
- ROS machine Ubuntu 18.04, ROS Melodic
- ROS–Unity communication: Direct
- main: [v0.5.0]

**Additional context**
-

Thank you in advance!

sdiao, 2021-07-26:
Hi @ja99 , thanks for reporting the issue. We are looking into the issue and will make a fix.

mrpropellers, 2021-08-03:
Hi @ja99 - can you share some additional information about your issue?  Which lines of code, specifically, are producing this error?  Can you share the snippet of the log message which contains this error?

ja99, 2021-08-04:
hi @mrpropellers and @sdiao,

I'm sorry, I accidentally forgot to change the name from the old (and still present in the project) "MCarState" generated message to the newly generated "CarStateMsg" and this is why my code still referenced MTime...

That beeing said, I at least found something usefull to you guys, where the "MMessage" naming scheme is still used:
https://github.com/Unity-Technologies/ROS-TCP-Connector/blob/f7f669f212423ff03817a2094f489547158a3852/com.unity.robotics.ros-tcp-connector/Runtime/Messages/HandwrittenMessages/msg/HeaderMsg.cs#L84

Sorry for the mistake and thanks for your time!

closed 2021-08-04 by ja99