Unity-Technologies/Unity-Robotics-Hub #275: No respond on ROS when testing TCP connction

kisstherain8677, 2021-07-26:
Hello,thanks for you guys tutorials .I went through Pick-and-Place Tutorial: Part 2 and pressing publish button,there is nothing print on 
ROS terminal.

[image]

I run ROS in docker,and I have checked the ROS IP:
[image]

[image]

**Environment **
- Unity Version: [Unity 2020.3.13f1]
- Unity machine OS: Windows 10
- ROS machine OS : [Windows10 on Docker, ROS melodic]
- ROS–Unity communication: [Docker]

sdiao, 2021-07-26:
Hi @kisstherain8677 , thanks for reaching out. The warning that you posted is harmless if you are running from the Unity Editor. Was there any error or info from your Unity Editor's console output? Were you able to run the Pick and Place Quick Demo https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/main/tutorials/pick_and_place/quick_demo.md before doing the Part 2 tutorial? I would recommend to start with the Quick Demo if you haven't done so. 
Based on the screenshot you shared, the connection between Unity and ROS is good.

kisstherain8677, 2021-07-27:
Thacks for your reply! I did run the quick_demo,but the robot didn't move either. I guess it should have moved while function publisher.PublishJoints() executes in update()?

no error info print in unity consule or ROS terminal

here are my screen shots
**unity**
[image]

[image]

**ROS terminal**
[image]

kisstherain8677, 2021-07-27:
Hi! I checked the port 127.0.0.1:10000 and found that a service occupied this port...it's ok after I killing it ,sorry for taking your time and thank you!

sdiao, 2021-07-28:
No worries. I am glad you figured it out. Feel free to reach out if you have further questions.

closed 2021-07-28 by sdiao