Unity-Technologies/Unity-Robotics-Hub #276: Unity to ROS Publisher Frequency Drops

SeanLMH, 2021-07-26:
Hi, I'm trying to get Unity to publish data to a topic at a constant 100 Hz so I've placed the Send command inside FixedUpdate() (set at 0.01 tiimetstep). When I play from editor and use rostopic hz to monitor frequency, I'm getting about 90 Hz. Any thoughts on what I can do to fix this? I also tried using having the send occur in Update() by monitoring elapsed time but performance was even worse at 60 Hz. Thanks in advance for any help!

sdiao, 2021-07-26:
This issue seems related to the [forum thread](https://forum.unity.com/threads/unity-to-ros-publisher-frequency-drops-when-subscribed-to-a-topic.1145393/) . Could you share which Unity Editor, ROS-TCP-Connector package, and ROS1 or ROS2 that you are using? Thanks.

SeanLMH, 2021-07-27:
Hi, thank you for the reply. I posted here because the forum seemed a bit dead, but I see that I've received a reply on that end so I'll move this issue over to the forum. Thanks again!

closed 2021-07-27 by SeanLMH