Unity-Technologies/Unity-Robotics-Hub #277: Error during basic Unity Service tutorial

Gregory-Baker, 2021-07-27:
**Describe the bug**
Error calling UnityService, when following [this section](https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/main/tutorials/ros_unity_integration/unity_service.md) of the tutorial (using v0.5.0 release). I can see that the service has been successfully registered by ROS when Unity starts, but I get the following errors when I call ```rosservice call /obj_pose_srv Cube``` from the ROS PC. A similar error also occurs in the [subsequent tutorial](https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/main/tutorials/ros_unity_integration/service_call.md).

**Console logs / stack traces**
On Unity side: 
```
Can't publish a message to an unregistered topic 'obj_pose_srv'
UnityEngine.Debug:LogError (object)
Unity.Robotics.ROSTCPConnector.ROSConnection:Send<Unity.Robotics.ROSTCPConnector.MessageGeneration.Message> (string,Unity.Robotics.ROSTCPConnector.MessageGeneration.Message) (at Library/PackageCache/com.unity.robotics.ros-tcp-connector@3f5d167040/Runtime/TcpConnector/ROSConnection.cs:731)
Unity.Robotics.ROSTCPConnector.ROSConnection/<>c__DisplayClass91_0:<ReceiveSysCommand>b__0 (string,byte[]) (at Library/PackageCache/com.unity.robotics.ros-tcp-connector@3f5d167040/Runtime/TcpConnector/ROSConnection.cs:463)
Unity.Robotics.ROSTCPConnector.ROSConnection:Update () (at Library/PackageCache/com.unity.robotics.ros-tcp-connector@3f5d167040/Runtime/TcpConnector/ROSConnection.cs:391)
```

On Ros side: 
- Response to rosservice call command:
```
ERROR: service [/obj_pose_srv] responded with an error: error processing request: unpack requires a string argument of length 56
```

- in terminal running server_endpoint.py
```
rosout: Error processing request: unpack requires a string argument of length 56
['Traceback (most recent call last):\n', '  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 632, in _handle_request\n    response = convert_return_to_response(self.handler(request), self.response_class)\n', '  File "/home/pioneer/catkin_ws/src/ROS-TCP-Endpoint/src/ros_tcp_endpoint/unity_service.py", line 53, in send\n    return self.tcp_server.send_unity_service(self.topic, self.service_class, data)\n', '  File "/home/pioneer/catkin_ws/src/ROS-TCP-Endpoint/src/ros_tcp_endpoint/server.py", line 99, in send_unity_service\n    return self.unity_tcp_sender.send_unity_service_request(topic, service_class, request)\n', '  File "/home/pioneer/catkin_ws/src/ROS-TCP-Endpoint/src/ros_tcp_endpoint/tcp_sender.py", line 106, in send_unity_service_request\n    response = service_class._response_class().deserialize(thread_pauser.result)\n', '  File "/home/pioneer/catkin_ws/devel/lib/python2.7/dist-packages/unity_robotics_demo_msgs/srv/_ObjectPoseService.py", line 226, in deserialize\n    raise genpy.DeserializationError(e)  # most likely buffer underfill\n', 'DeserializationError: unpack requires a string argument of length 56\n']
```

**Expected behavior**
As per the tutorial I expect to receive the position of the cube as a response to the service call.

**Environment:**
- Unity Version: Unity 2020.3.14f1
- Unity machine OS + version: Windows 10
- ROS machine OS + version: Ubuntu 18.04 (Running on Jetson Nano), ROS Melodic
- ROS–Unity communication: ROS-TCP-Connector (Non-docker-based) - communicating over ethernet.
- Branch or version: v0.5.0

**Additional comment**
Thanks so much to the team for your amazing work on ROS-Unity Integration and thanks to anyone that helps me solve this issue. Sorry if it's just me being stupid.

sdiao, 2021-07-28:
Hi @Gregory-Baker , Thanks for reaching out. We are looking into the issue and will let you know once we have an update. Thanks.

sdiao, 2021-07-30:
Hi @Gregory-Baker , in your Unity Editor's package manager, could you delete the ROS-TCP-Connector package, and then re-add it using the github url: `https://github.com/Unity-Technologies/ROS-TCP-Connector.git?path=/com.unity.robotics.ros-tcp-connector#v0.5.0`?

Our current tutorial suggested using the dev branch of the Unity ROS-TCP-Connector in  https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/main/tutorials/ros_unity_integration/setup.md#-unity-setup . However, dev branch is not stable and we are currently looking into fixing the bug you reported. 

Let us know if the error goes away after you switch to use the `https://github.com/Unity-Technologies/ROS-TCP-Connector.git?path=/com.unity.robotics.ros-tcp-connector#v0.5.0` in package manager. Thanks.

sdiao, 2021-07-30:
We have also pushed a fix to the ROS-TCP-Connector dev branch, you should be able to run the tutorial with your current settings. Let us know if you are still running into issues with the tutorial. Thanks.

Gregory-Baker, 2021-08-02:
I've switched off the dev branch and onto main and it's working fine now. Thanks for responding and resolving this so quickly!

closed 2021-08-02 by Gregory-Baker