Unity-Technologies/Unity-Robotics-Hub #278: Pick and place doesn't work because Ubuntu doesn't return a response.

KazukiKuriyama, 2021-07-29:
**Describe the bug**
I tried "Pick-and-Place Demo Part 3: Pick-and-Place In Unity", but at the time of release, the trajectory was displayed on the console side of Docker, but no response was returned to Unity side.
Of course, the response callback doesn't work.

I also used WireShark to monitor packets on the cable between my Windows 10 PC and Ubuntu 18.04, but it seems that only ACK was returned from the Ubuntu side.

Is it possible to run the demo in my environment?

**To Reproduce**
Execute "Pick-and-Place Demo Part 3: Pick-and-Place In Unity" in the following environment.

**Environment (please complete the following information, where applicable):**
- Unity Version: [e.g. Unity 2020.3.14f1]
- Unity machine OS + version: [ Windows 10]
  Firewall has been set to disabled
- ROS machine OS + version: [ Ubuntu 18.04, ROS melodic]
  Firewall has been set to disabled
- ROS–Unity communication: [Docker]
  Execute ROS from Docker as described in the pick-and-place demo procedure manual.
- Branch or version: [Remove ROS2 Beta Mentions (#272)]
- The two PCs are directly connected by a LAN cable.

**Additional context**
When I install Unity on an Ubuntu PC, the response comes back without any problems and the robot works correctly, but I want to receive the response from windows 10.

sdiao, 2021-07-30:
Hi @KazukiKuriyama , thanks for reaching out. Short answer is that you should be able to run the tutorial with Unity running on a Windows machine and ROS running on Ubuntu.  We are looking into the issue and will let you know once we have an update. Was there any error or warning on either ROS or Unity side?

KazukiKuriyama, 2021-07-30:
Thank you for your support!

I don't know the relevance, but I get the following error on the Unity console:
> ArgumentException: JSON parse error: Invalid value.
UnityEngine.JsonUtility.FromJson (System.String json, System.Type type) (at <0b59f3b614a04547a8bf657de1e23f2c>:0)
UnityEngine.JsonUtility.FromJson[T] (System.String json) (at <0b59f3b614a04547a8bf657de1e23f2c>:0)
Unity.Robotics.ROSTCPConnector.ROSConnection.ReceiveSysCommand (System.String topic, System.String json) (at Library/PackageCache/com.unity.robotics.ros-tcp-connector@f7f669f212/Runtime/TcpConnector/ROSConnection.cs:433)
Unity.Robotics.ROSTCPConnector.ROSConnection.Update () (at Library/PackageCache/com.unity.robotics.ros-tcp-connector@f7f669f212/Runtime/TcpConnector/ROSConnection.cs:395)

The contents of Json at that time are as follows.

> "�\0\0\0Topic \'__request\' is not registered! Known topics are: [\'sim_real_pnp\', \'SourceDestination_input\', \'niryo_one/commander/robot_action/goal\', \'niryo_moveit\', \'NiryoTrajectory\'] "

KazukiKuriyama, 2021-07-30:
I added a block diagram and a Docker console to the first comment.

KazukiKuriyama, 2021-07-30:
I made a little progress.
Only on the Windows side where Unity is installed
After changing to v.0.4.0, I confirmed that the robot works without any problem.
When I cloned v.0.5.0 again in that environment, it didn't work in this version as before.

Isn't there a bug between v.0.4.0 and v.0.5.0?

There are other points I noticed.
-It did not occur in v.0.4.0, but in v.0.5.0, "RosMessageTypes.NiryoMoveit" is not recognized in the namespace unless the C # file permissions are changed.
-In v.0.5.0, "git submodule" is always required because the contents of "tutorials \ pick_and_place \ ROS \ src" are only slightly recognized by the ROS settings.

sdiao, 2021-07-30:
Thanks for the detailed notes. It sounds like there might be a bug introduced between v0.4.0 and v0.5.0. We are investigating.

sdiao, 2021-07-30:
The version of ROS-TCP-Connector has to be the same as ROS-TCP-Endpoint. Based on "After changing to v.0.4.0, I confirmed that the robot works without any problem", it seems that your ROS-TCP-Endpoint is on v0.4.0 in Docker, even though the Unity project switched to v0.5.0. 
Please use git submodule when you update the Unity-Robotics-Hub, and rebuild your docker container. 

You can check the version in Unity-Robotics-Hub/tutorials/pick_and_place/ROS/src/ros_tcp_endpoint/package.xml
Something like:
`<?xml version="1.0"?>
<package format="2">
  <name>ros_tcp_endpoint</name>
  <version>0.5.0</version>`

Please let us know if it helps. Thanks.

KazukiKuriyama, 2021-08-02:
Thank you for your reply.
When I checked the xml file, "0.4.0" was displayed, so I think that was the cause.
When I tried it with "0.5.0" Docker, I was able to confirm that the communication was successful!

closed 2021-08-02 by KazukiKuriyama