Unity-Technologies/Unity-Robotics-Hub #279: The Ross Setting screen of ROS – Unity Integration Networking is different from the current menu.

KazukiKuriyama, 2021-07-29:
**Describe the bug**
The Ross Setting screen of ROS – Unity Integration Networking is different from the current menu.
Currently, there is no text box to set Unity's IP address. Is it unnecessary?

https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/main/tutorials/ros_unity_integration/network.md

current Unity version
[image]

network.md
[image]

sdiao, 2021-07-30:
Hi @KazukiKuriyama , thanks for letting us know about the mismatch in the ROS Settings window. We will update the screenshot. Yes, there is no need to set the Unity IP in Unity Editor. Please let us know if you have further questions.

KazukiKuriyama, 2021-07-30:
Regarding "Explicitly set the UNITY_IP" described in "2. Can ROS send messages to Unity?", Where should I clarify it?
Is this also unnecessary now?

sdiao, 2021-07-30:
Apologies for the outdated doc. We will update the doc. You should not need to set the UNITY_IP. When you start your Unity scene, it should open the connection between Unity and ROS, and ROS can send messages to Unity. I would recommend check out the Subscriber tutorial.

sdiao, 2021-08-02:
I have pushed the document update to dev branch. Close the issue now. Please feel free to reach out if you have any question in the future.

closed 2021-08-02 by sdiao