Unity-Technologies/Unity-Robotics-Hub #280: Intermittend Connection Issue: ROS: [Errno 104] Connection reset by peer <-->Unity: System.IO.IOException: Unable to write data to the transport connection: An existing connection was forcibly closed by the remote host

Axelius89, 2021-07-30:
Hello,

we have a setup for which we are using ROS mostly to write long continuous data into ROSbag. Most of it is handled through regular ROS nodes in Python and C++ all running on a single workstation.

I developed an additional Unity application which uses ROS-TCP-Endpoint v0.5.0 and ROS-TCP-Connector v0.5.0 which sends sporadic strings from Unity to ROS which are used to annotate the ROSbags that we are writing, so that we can quickly find different sections in the continuous data. We chose Unity for this part of the project, as it gives us platform independence, as this way we can use a Windows Tablet or Android device to send these annotations.

The annotations are made by a user, taken from a list of buttons. They are relatively short strings (maybe 100 characters or so) and are usually send several minutes apart. So it really is not a high performance thing.

We use a Microsoft Surface Pro 2 running Windows 10 which runs an application built in Unity 2020.3.11f1 to send these messages to a Workstation running Ubuntu 18.04.5 LTS, which runs ROS melodic 1.14.10. The annotations are directly written to ROSbag with the rosbag command line tool and no other subscriber listening to the /annotation topic.

**Describe the bug**
At the beginning this setup works very nicely and annotation strings are received and written to the ROSbag. However after some time - sometimes hours - we get sporadic error messages and the annotations stop coming through for some time. After some time (maybe a few seconds or minutes) the annotations start getting through again without me restarting either the receiving or sending side. This is very difficult to debug, due to the sporadic nature of the problem. I hope someone here with more experience can spot some obvious flaw in my setup.

On the Unity side I'm getting these kind of errors:

    Connection to 192.168.123.1:10000 failed - System.IO.IOException: Unable to write data to the transport connection: An existing connection was forcibly closed by the remote host.
    Connection to 192.168.123.1:10000 failed - System.Net.Sockets.SocketException (0x80004005): A socket operation was attempted to an unreachable host.
    Connection to 192.168.123.1:10000 failed - System.Net.Sockets.SocketException (0x80004005): A socket operation was attempted to an unreachable host.
    Connection to 192.168.123.1:10000 failed - System.Net.Sockets.SocketException (0x80004005): A socket operation was attempted to an unreachable host.
    Connection to 192.168.123.1:10000 failed - System.Net.Sockets.SocketException (0x80004005): A socket operation was attempted to an unreachable host.
    ...

and on the ROS workstation I'm getting these errors:

    1627381093.275654315 ERROR /TCPServer_4557_1627365220527 [client.py:238(ClientThread.run)] [topics: /annotation, /rosout] Exception: [Errno 104] Connection reset by peer
    1627381093.280012369 INFO /TCPServer_4557_1627365220527 [client.py:242(ClientThread.run)] [topics: /annotation, /rosout] Disconnected from 192.168.20.101

Then, some time after it connects automatically again:

    1627381344.376739025 INFO /TCPServer_4557_1627365220527 [client.py:204(ClientThread.run)] [topics: /annotation, /rosout] Connection from 192.168.20.101

This is the ROS-TCP-Endpoint that I'm starting on the workstation:

    #!/usr/bin/env python
    
    import rospy
    
    from ros_tcp_endpoint import TcpServer, RosPublisher
    from std_msgs.msg import String
    
    
    def main():
        ros_node_name = rospy.get_param("/TCP_NODE_NAME", 'TCPServer')
        buffer_size = rospy.get_param("/TCP_BUFFER_SIZE", 1024)
        connections = rospy.get_param("/TCP_CONNECTIONS", 10)
        tcp_server = TcpServer(ros_node_name, buffer_size, connections)
        rospy.init_node(ros_node_name, anonymous=True)
        
        tcp_server.start({'annotation': RosPublisher('annotation', String, queue_size=10)
        })
        
        rospy.spin()
    
    
    if __name__ == "__main__":
        main()

and this is the Unity code to send messages (in a callback when the button is pressed. "ros" was initialized with as ros = ROSConnection.instance;)

    RosMessageTypes.Std.StringMsg rosmsg = new RosMessageTypes.Std.StringMsg(msg);
    ros.Send(topicName, rosmsg);

**Console logs / stack traces**
I post some logfiles of both Unity and ROS to shed some light on the nature of the problem.

ROS log (only the output of TCPServer):
```
1627365220.651114940 INFO /TCPServer_4557_1627365220527 [server.py:78(TcpServer.listen_loop)] [topics: /annotation, /rosout] Starting server on 192.168.123.1:10000
1627365221.124732255 INFO /TCPServer_4557_1627365220527 [client.py:204(ClientThread.run)] [topics: /annotation, /rosout] Connection from 192.168.20.101
1627365221.137033700 INFO /TCPServer_4557_1627365220527 [server.py:160(SysCommands.publish)] [topics: /annotation, /rosout] RegisterPublisher(annotation, <class 'std_msgs.msg._String.String'>) OK
[... 333 lines cut ...]
[1627397153664][16:45:53.665] Message
[1627397460624][16:51:00.624] Message
Connection to 192.168.123.1:10000 failed - System.IO.IOException: Unable to write data to the transport connection: An established connection was aborted by the software in your host machine.
Connection to 192.168.123.1:10000 failed - System.Net.Sockets.SocketException (0x80004005): No connection could be made because the target machine actively refused it.
Connection to 192.168.123.1:10000 failed - System.Net.Sockets.SocketException (0x80004005): No connection could be made because the target machine actively refused it.
Connection to 192.168.123.1:10000 failed - System.Net.Sockets.SocketException (0x80004005): No connection could be made because the target machine actively refused it.
...
Setting up 2 worker threads for Enlighten.
  Thread -> id: 2290 -> priority: 1 
  Thread -> id: 5d0 -> priority: 1 
```

**Expected behavior**
Continuously be able to send messages to ROS, that are reliably received.
No System.IO.IOException and System.Net.Sockets.SocketException in Unity
No [Errno 104] Connection reset by peer in ROS

**Environment**
- Unity Version: Unity 2020.3.11f1
- Unity machine OS + version: Windows 10
- ROS machine OS + version: Ubuntu 18.04.5 LTS, ROS melodic 1.14.10.
- Branch or version: v0.5.0

**Additional context**
Thank you so much for this amazing project!

sdiao, 2021-08-02:
Hi @Axelius89 , thanks for reaching out and providing the detailed notes. We are looking into the issue and will let you know if there is any update.

mrpropellers, 2021-08-03:
This feels a bit like some resource mismanagement is happening somewhere in our code. Potentially a memory leak or a good old-fashioned buffer overflow maybe, but weird that you're not seeing more error text from the endpoint side... @Axelius89 just to confirm I'm reading this correctly: The Unity logs are taken over ~12 hours of operation?  And then the Endpoint side is about ~30,000 seconds, or ~8 hours?

We haven't stress-tested our code in this way before, so I'll need to see about setting up a similar test internally. In the meantime, if you end up doing more of these types of sessions, could you potentially monitor your memory/cpu/network usage on the machines involved?  I'd be curious to know if there are spikes in any of the resource usage around these failure points.

Axelius89, 2021-08-03:
Thanks for the reply!

The length of the recording sounds about right. The Unity App was running longer than the ROS workstation (at least as long as the logs I posted are concerned. Some time in the first few hours we needed to restart the workstation), so 12 hours and 8 hours can be correct.

In fact we're recording another session today (and more in the upcoming weeks), I'll see if there will be error messages. Keeping a look at memory, cpu and network sounds good, I'll see if I can set up something. I have the feeling that the errors do not show up when I just leave the system running, so some correlation to other things happening on the workstation or network might be a thing.

mrpropellers, 2021-08-04:
If the errors only show up when _in use_, and not when everything is open but idle, then that further smells like a buffer overflow to me... Could be there is a ROS publisher or subscriber somewhere with an infinite `queue_size` that is filling up all the available memory. Keep us posted!

P3TE, 2021-08-19:
Not saying it's related, but I had a drop out issue with the ROS-TCP-Endpoint with a much earlier version (one I cloned late 2020). The problem was, I would stop sending messages for a while from the Unity side (I think I was pausing the simulation) and then the connection would drop out.

My hacky fix was to add: `settimeout(9999999)` to server.py
For example:
```
tcp_server.bind((self.tcp_ip, self.tcp_port))
tcp_server.settimeout(9999999)
tcp_server.listen(self.connections)
```
and
```
(conn, (ip, port)) = tcp_server.accept()
conn.settimeout(9999999)
```

I will note that I haven't experienced this issue with later versions of the Endpoint / Connector which I think is because there's a new keep alive message being sent periodically from Unity.

Anyhow, I think it wouldn't hurt to add `settimeout(9999999)` into the ROS-TCP-Endpoint

closed 2021-10-17 by github-actions[bot]