Unity-Technologies/Unity-Robotics-Hub #284: ROS Unity unable to connect with no connection information

keli95566, 2021-08-08:
I am trying to run the unity-ros-integration tutorials following every step. Was able to build the catkin project, set environment, and get the robo_demo.launch file to run. However, Ping from both devices, which are connected to same wifi, and the pings were successful. 
But if I run the pos publisher ```rostopic echo pos_rot``` from windows unity and then a subscriber from Linux ROS, the terminal hangs with the following message:

```bash
WARNING: topic [/pos_rot] does not appear to be published yet
```
ROS Master is running, and no message about connection is available when I start or stop Unity editor play.

```
setting /run_id to 5c1a867c-f87d-11eb-b01b-336712ef486f
process[rosout-1]: started with pid [174801]
started core service [/rosout]
process[position_service_and_endpoint/position_service-2]: started with pid [174808]
process[position_service_and_endpoint/server_endpoint-3]: started with pid [174809]
```

Just each time when I stop Unity editor play, on Linux ROS master output the following exception:

```
[ERROR] [1628450321.770259]: Exception: No more data available
[ERROR] [1628450346.369471]: Exception: No more data available
[ERROR] [1628450567.044703]: Exception: No more data available
```

No errors given from the player log,

It seems that there is some connection between ROS and Unity judging from the screenshot. As you can see from the blue error on the upper left corner:
[image]

**Environment (please complete the following information, where applicable):**
- Unity Version: [e.g. Unity 2020.3.13f1]
- Unity machine OS + version: [Windows 10]
- ROS machine OS + version: [Ubuntu 20.04LTS, ROS Noetic]
- ROS–Unity communication: [Wifi]
- Branch or version: [master]

Please, can anyone help.

at669, 2021-08-09:
Hi @keli95566, thanks for reaching out! It looks like your Unity configuration is trying to publish to the pos_rot topic, but your server endpoint isn't set up to receive them, or something else is blocking the communication--you should receive confirmation that it connects with Unity when it enters and exits Play mode. 

Since you are able to successfully ping each device and are using the demo launch file, can you verify that the proper ROS IP is being set from the launch file? You can verify both the launch file (e.g. that `<env name="ROS_IP" value="..."/>` matches your IP), and that `rosparam get ROS_IP` returns the correct value? Additionally, can you verify that there are no typos in the Unity side ROS Settings IP address? Additionally, please let us know if you're getting any errors in either Unity or ROS console before the exception.

Since you are running Windows, it may be possible the communication is getting blocked by the firewall. You can check out this [Network Troubleshooting Guide](https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/main/tutorials/ros_unity_integration/network.md#troubleshooting) to try opening the port for the firewall.

Let us know if this helps resolve your issue!

keli95566, 2021-08-09:
Hi @at669, thanks a lot for quickly getting back to me!

1. Indeed that it seems something is blocking the communication, but I could not figure out what.
2. I think all the IP addresses, checking the env variable in the launch file, and the env variable globally while running ros master are correct. 
3. The devices are connected via local network, therefore I have disabled the firewall for local network on Windows, would it be sufficient?
4. Do I need to do anything else in terms of communication on Linux side?
5. Unfortuanetly no errors were given from both the ros console and the Unity console.  There is only a warning from the Unity console. 
6. I have a Photon PUN2 plugin running in the project, but no photon view setup in this specific scene that I am running, could this be the problem?

```
SendMessage cannot be called during Awake, CheckConsistency, or OnValidate
UnityEngine.GameObject:AddComponent<Unity.Robotics.ROSTCPConnector.HUDPanel> ()
Unity.Robotics.ROSTCPConnector.ROSConnection:InitializeHUD () (at Library/PackageCache/com.unity.robotics.ros-tcp-connector@f7f669f212/Runtime/TcpConnector/ROSConnection.cs:371)
Unity.Robotics.ROSTCPConnector.ROSConnection:OnValidate () (at Library/PackageCache/com.unity.robotics.ros-tcp-connector@f7f669f212/Runtime/TcpConnector/ROSConnection.cs:361)

```
Additionally, I will attach the log from ros launch:

```
... logging to /home/keli/.ros/log/cf4af948-f930-11eb-b01b-336712ef486f/roslaunch-keli-Latitude-7400-268901.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.44.73:35199/

SUMMARY
========

PARAMETERS
 * /ROS_IP: 192.168.44.73
 * /ROS_TCP_PORT: 10000
 * /TCP_BUFFER_SIZE: 1024
 * /TCP_CONNECTIONS: 10
 * /TCP_NODE_NAME: TCPServer
 * /UNITY_IP: 192.168.44.72
 * /UNITY_SERVER_PORT: 5005
 * /rosdistro: noetic
 * /rosversion: 1.15.11

NODES
  /position_service_and_endpoint/
    position_service (unity_robotics_demo/position_service.py)
    server_endpoint (ros_tcp_endpoint/default_server_endpoint.py)

auto-starting new master
process[master]: started with pid [268909]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to cf4af948-f930-11eb-b01b-336712ef486f
process[rosout-1]: started with pid [268919]
started core service [/rosout]
process[position_service_and_endpoint/position_service-2]: started with pid [268926]
process[position_service_and_endpoint/server_endpoint-3]: started with pid [268927]

```
Also my launch file:
```
<launch>

    <env name="ROS_IP" value="192.168.44.73"/>
    <env name="ROS_HOSTNAME" value="$(env ROS_IP)"/>
    
    <param name="ROS_IP" type="str" value="$(env ROS_IP)" />
    <param name="ROS_TCP_PORT" type="int" value="10000" />
    <param name="TCP_NODE_NAME" type="str" value="TCPServer" />
    <param name="TCP_BUFFER_SIZE" type="int" value="1024" />
    <param name="TCP_CONNECTIONS" type="int" value="10" />

    <param name="UNITY_IP" type="str" value="192.168.44.72" />
    <param name="UNITY_SERVER_PORT" type="int" value="5005" />

    <group ns="position_service_and_endpoint">
           <node pkg="unity_robotics_demo" name="position_service" type="position_service.py"/>
           <node pkg="ros_tcp_endpoint" name="server_endpoint" type="default_server_endpoint.py"/>
    </group>
</launch>
```

And the default tcp endpoint.py that I was running:

```python3
#!/usr/bin/python3

import rospy

from ros_tcp_endpoint import TcpServer

def main():
    ros_node_name = rospy.get_param("/TCP_NODE_NAME", "TCPServer")
    tcp_server = TcpServer(ros_node_name)

    # Start the Server Endpoint
    rospy.init_node(ros_node_name, anonymous=True)
    tcp_server.start()
    rospy.spin()

if __name__ == "__main__":
    main()
```

Thank you so much for helping!

at669, 2021-08-09:
The output from the launch file is interesting--you should see something like `[INFO] [1628537615.986693]: Starting server on 192.168.44.73` after launching the endpoint. Were you able to successfully run the [Setup](https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/main/tutorials/ros_unity_integration/setup.md) and launch the default server endpoint in this step?

> The devices are connected via local network, therefore I have disabled the firewall for local network on Windows, would it be sufficient?

I'd suggest trying to set up the firewall rule to open the port specifically in this case. 

> Do I need to do anything else in terms of communication on Linux side?

Aside from setting up the valid ros params, building/sourcing your workspace, and launching the server endpoint, I think you should be fine! You can verify the [minimum requirements](https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/main/tutorials/ros_unity_integration/network.md#communication-minimum-requirements) for communication.

> I have a Photon PUN2 plugin running in the project, but no photon view setup in this specific scene that I am running, could this be the problem?

if it's not actively running in your test scene, I think we can rule it out for now, unless it is throwing errors.

Additionally, could you let us know which version of ROS-TCP-Connector and Endpoint you are using (e.g. v0.5.0)? You may need to update to the latest matching versions if they are not all up-to-date.

keli95566, 2021-08-09:
Thanks a lot for the hint! I am currently using v0.5.0. 
All the demo scripts work, if I don't run the demo launch file, but run each node separately on different  terminal.
I have the ```roscore``` running.
Then I have ```rosrun ros_tcp_endpoint default_server_endpoint.py ```, with the console log (I masked my ip address):
```
INFO] [1628541643.204472]: Starting server on ***.***.**.**:10000
[INFO] [1628541757.910116]: Connection from ***.***.**.**
[INFO] [1628541757.942574]: RegisterSubscriber(color, <class 'unity_robotics_demo_msgs.msg._UnityColor.UnityColor'>) OK
[INFO] [1628541757.961074]: RegisterUnityService(obj_pose_srv, <class 'unity_robotics_demo_msgs.srv._ObjectPoseService.ObjectPoseService'>) OK
[INFO] [1628541757.970334]: RegisterPublisher(pos_rot, <class 'unity_robotics_demo_msgs.msg._PosRot.PosRot'>) OK
```
And ```rostopic echo pos_rot``` was able to update correctly. 
It seems that the network communication is normal, but there is something with the demo.launch file, I am just into ROS for a few days and couldn't quite understand why.

at669, 2021-08-10:
Glad to hear the other demos worked! After further investigation, you're absolutely correct, the `robo_demo.launch` file is outdated. The paths in the launchfile point to the wrong locations--I've created an internal ticket for investigation.

Until the tutorial is fully updated, as a workaround, you should be able to replace your launch file with something like this:

```xml
<launch>
    <param name="ROS_IP" value="192.168.44.73" />
    <node name="position_service" pkg="unity_robotics_demo" type="position_service.py" args="--wait" output="screen" respawn="true" />
    <node name="server_endpoint" pkg="ros_tcp_endpoint" type="default_server_endpoint.py" args="--wait" output="screen" respawn="true" />
</launch>
```

Ticket#: AIRO-1118

at669, 2021-08-13:
Hi @keli95566, the fix has been merged to the `dev` branch and will eventually make its way to `main` in the next release. Feel free to try out the changes on your end! I'll close this issue for now, but go ahead and reopen it if the problem persists, or open a new issue if a new problem crops up. Thanks for your contribution!

closed 2021-08-13 by at669