Unity-Technologies/Unity-Robotics-Hub #285: Robot loses mesh colliders after first run if two links are using the same dae files

ZhuoyunZhong, 2021-08-10:
**Describe the bug**
If two links are using the same DAE file, one of the robot links will lose its mesh colliders after the first run in Unity.

**To Reproduce**
Steps to reproduce the behavior:
1. Pick & Place demo, use an empty scene and import the robot (Y-axis, YHACD)
2. Click the root of the robot gameObject, and all colliders exist and are visible
3. Click Run in Unity, then stop
4. Click the root of the robot gameObject again, navigate to control_rod_right and control_rod_left, which are using the same DAE file. Click their collision sub-gameObject, you will see one of them has colliders and the other doesn't anymore. And all collider mesh slots become None.
5. By the way, if you do not import the robot with URDF-importer, but use Part1DoneScene directly, this problem does not exist.

**Console logs / stack traces**
None

**Environment (please complete the following information, where applicable):**
- Unity Version: Unity 2020.3.12f1
- Unity machine OS + version: Ubuntu 18.04
- ROS machine OS + version: Ubuntu 18.04, ROS Melodic
- ROS–Unity communication: Machine
- Branch or version: v0.5.0

**Additional context**
This problem does not have a big difference in this demo robot. But in my case, I am building a humanoid robot with two robot arms. Also, the two grippers of each arm share the same DAE file. After importing my robot, one of the arms and three of the four grippers will not have colliders attached to them.

Thank you very much for your help!

at669, 2021-08-10:
Hi, thanks for reporting this! I've created a ticket internally--we'll take a look into this and get back to you with a potential solution.

As a temporary workaround, so long as you are using the VHACD decomposer, you may be able to modify [UrdfGeometryCollision](https://github.com/Unity-Technologies/URDF-Importer/blob/e9859c4becda48194e153356afc0a802516b1488/com.unity.robotics.urdf-importer/Runtime/Extensions/UrdfGeometryCollision.cs#L199-L206) to only create a new mesh asset when one doesn't already exist (i.e. so as to not overwrite the previously imported mesh). 

```csharp
if (!RuntimeUrdf.IsRuntimeMode())
{
    meshIndex++;
    string name = $"{filePath}/{templateFileName}_{meshIndex}.asset";
    // Only create new asset if one doesn't exist
    if (!RuntimeUrdf.AssetExists(name))
    {
        Debug.Log($"Creating new mesh file: {name}");
        RuntimeUrdf.AssetDatabase_CreateAsset(collider, name);
        RuntimeUrdf.AssetDatabase_SaveAssets();       
    }
}
```

Ticket#: AIRO-1117

at669, 2021-09-27:
The PR has been merged into `dev` and will be in the next release of the URDF-Importer. Please feel free to try it out on `dev`. I'll be closing this issue, but feel free to reopen it if the problem persists, or open a new ticket if you encounter other issues.

closed 2021-09-27 by at669