Unity-Technologies/Unity-Robotics-Hub #287: Driver for Robot with Wheels?

keli95566, 2021-08-11:
Will there be similar wheel drivers such as skid steer driver for Gazebo for quick control for common robots with four or two wheels? Currently I am writing my own scripts to manage the robot movement, but it would be nice to have a driver plugin from the package to steer the robot for users who are not too deep into wheel mechanics. 
In addition, I am wondering if it would be problematic to use the wheel collider from Unity to manage the wheel movements of robot?

at669, 2021-08-11:
Hi, thanks for the question--for wheel colliders, you can use the wheel colliders for movement, but there are two main things you will need to keep in mind:
1. The mesh of the wheels will not rotate visually, and you would need to transform the mesh of the wheels manually.
2. Wheel colliders don't use physics necessarily--instead, they use analytical formulas to calculate the movement of the wheel and vehicle.

Additionally, we will soon be releasing our AMR demo, which will showcase using wheeled robot control in Unity! I will post an update in this thread when that goes live.

at669, 2021-08-13:
Hey @keli95566, the AMR demo is live now! You can learn more about it in our blog post [here](https://blog.unity.com/manufacturing/advance-your-robot-autonomy-with-ros-2-and-unity).

ZhuoyunZhong, 2021-08-14:
Hi, I have been developing a project which simulates mobile robots in Unity, and recently created a [mobile robot demo](https://github.com/ZhuoyunZhong/Mobile-Robot-Demo-with-Unity-ROS) to show how to do it. I was originally thinking I may contribute to the community by doing so. But I guess I am a bit late 😂, I didn't know @at669 you guys are releasing an AMR demo recently.

I take a quick look at the AMR demo, it looks like my [mobile robot demo](https://github.com/ZhuoyunZhong/Mobile-Robot-Demo-with-Unity-ROS) still has something different that might be useful. My project includes ImagePublisher to publish camera images to ROS, a TwistSubscriber to receive control command (cmd_vel) from ROS, and JointStatePublisher to send joints info. 

Anyway, I post here just in case anyone may need more examples. And thank you so much for giving the AMR demo!

keli95566, 2021-08-14:
> Hey @keli95566, the AMR demo is live now! You can learn more about it in our blog post [here](https://blog.unity.com/manufacturing/advance-your-robot-autonomy-with-ros-2-and-unity).

Thanks a lot!! :)

at669, 2021-08-16:
Hi @ZhuoyunZhong, your demo looks great! If you're comfortable doing so, you can post your project in the [Robotics Forum](https://forum.unity.com/forums/robotics.623/) with the `Showcase` prefix! 

I'll be closing out this issue, thanks for the contributions everyone!

closed 2021-08-16 by at669