Unity-Technologies/Unity-Robotics-Hub #292: Outdated ROS Settings Instruction and Window in Pick-and-Place Tutorial

A-Ivan, 2021-08-13:
**Describe the bug**
Hi Unity Robotics Team.

There is an outdated ROS Settings Instructions and Window in the Pick-and-Place Tutorial - [Part 2](https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/main/tutorials/pick_and_place/2_ros_tcp.md#the-unity-side)

The Unity Side - Item 9 - The ROS settings setup instructions and window image are out of date (still shows/mentions Override Unity IP Address)

[image]

at669, 2021-08-13:
Hey @A-Ivan, thanks for letting us know! We'll make sure this instruction gets updated.

[Ticket#: AIRO-1133]

at669, 2021-08-13:
@A-Ivan, the fix has been merged to the `dev` branch and will make its way to `main` in the next release. I'll close out this issue now--thanks for bringing it to our attention!

closed 2021-08-13 by at669