Unity-Technologies/Unity-Robotics-Hub #293: Create docker image failed

DiosVan, 2021-08-14:
**Describe the bug**
construct image error in ros_unity_integration/ros_docker,  I can construct ros2_docker image and pick_and_place docker image, but ros_docker not, Does anyone know why?

[image]

- Unity machine OS + version:  Windows 10 
- ROS machine OS + version: Ubuntu 20.04.2, ROS melodic
- ROS–Unity communication: Docker
- Branch or version:  v0.5.0

at669, 2021-08-16:
Hi @DiosVan, thanks for reporting this! I've filed an internal ticket to investigate this issue and we will get back to you soon.

[Ticket#: AIRO-1135]

peifeng-unity, 2021-09-07:
Hi @DiosVan, sorry for the late reply. It seems that your step [4/8] used a cached one for copying the `/setup.sh` file and the step [5/8] cannot find the copied file. Could you try to clear the cache using `docker builder prune -a -f` and try to build it again?

cdh4u, 2021-09-23:
I get the same problem when trying to build the image on Windows (I have not tried to build the image on Linux)

When comparing to the pick_and_place Dockerfile (which works), I see that the ros_unity_integration lacks 'dos2unix' commands. I try to add those, but it doesn't help. It is kind of strange, because the Dockerfile become almost identical, with the main difference being which scripts etc are copied into the image.

DavidbVEC, 2022-03-29:
I managed to fix this by replacing the line with this 
RUN dos2unix /setup.sh && chmod +x /setup.sh && /setup.sh && rm /setup.sh

and adding:
RUN sudo apt-get update && sudo apt-get install -y dos2unix

After setting the ENV