Unity-Technologies/Unity-Robotics-Hub #295: ROS TCP handshake fails in Part 2 and No menu tab names Override Unity IP Address

JounMinPark, 2021-08-19:
I have similar error like the link below, It get connects to the ROS but dosen't show the hearing message on the ROS
[image]

https://github.com/Unity-Technologies/Unity-Robotics-Hub/issues/170
Title: ROS TCP handshake fails in Part 2, even after double checking the ROS_IP settings #170

but the difference is that I could not change the "Override Unity IP Address" since there in no tab in my ROS settings

[image]

I am currently using Window10/Unity 2020.3.11f1 / ROS on Docker

DiosVan, 2021-08-20:
HI, @JounMinPark Override Unity IP Address has been removed in the latest version of RosSettings, 
your situation is very similar to mine.
Although I still have some doubts about this topic, but maybe you can refer to my solution [this](https://github.com/Unity-Technologies/Unity-Robotics-Hub/issues/273#issuecomment-899948274), remember to rebuild docker image after modification.

JounMinPark, 2021-08-20:
Thank you for giving me help,

after rebuilding the docker image and modifying the script it worked  successfully!!!

[image]

[image]

DiosVan, 2021-08-20:
Good job, we also look forward to the official response to this issue.:)

vidurvij-Unity, 2021-08-24:
I am glad this solution worked out for you. I am investigating wether this is a potential bug that needs to be fixed. I will get back to you with a proper solution.

peifeng-unity, 2021-09-07:
You are right that this is related to issue #273. [A fix](https://github.com/Unity-Technologies/Unity-Robotics-Hub/pull/308) has been merged to the `dev` branch. Please let us know if it would work for you on the most recent dev branch.

closed 2021-10-09 by vidurvij-Unity