Unity-Technologies/Unity-Robotics-Hub #296: Basic Error found while running ROS node to subscribe to topic /pos_rot published by Unity (Following Unity Publisher tutorial)

Akashleena, 2021-08-20:
**Describe the bug**
I followed the [Publisher tutorial](https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/main/tutorials/ros_unity_integration/publisher.md)) to publish the pose of the robot on/pos_rot topic. On the ROS side I have written a C++ node which can read thode messages and display on the console. Later I plan to publish those messages to Gazebo.

I am getting a couple of errors in the src/unity_subscribe.cpp as also the CMakelists.txt
[image]
[image]
[image]
[image]

Errors I am getting:
[image]
To remove the above error, I commented the line in CMakeLists.txt to remove the error.. (But since in the cpp code I am trying to subscribe to the topics sent by Unity and reading those messages and dispalying it on the console, I think I need Line 30 add_dependencies(unity_subscribe robo_simulation_generate_messages_cpp) in the CMakeLists.txt

Error 2: After removing Line 30 add_dependencies line from CMakeLists.txt
[image]

I have checked the synatx to include a custom message header file in  [ROS WIKI](http://wiki.ros.org/ROS/Tutorials/DefiningCustomMessages)
and I feel my synatx is correct. I have also placed the unity_robotics_demo_msgs package inside the src folder of my workspace.
My directory structure is as follows
    

To Reproduce the errors
Steps to reproduce the behavior:
To reproduce the error just clone this [package](https://github.com/Akashleena/unity_gazebo_ros) and place it in the src/ folder of your catkin workspace 
(~/catkin_ws/src/unity_gazebo_ros) and do a catkin_make

**Console logs / stack traces**
Please wrap in [triple backticks (```)](https://help.github.com/en/articles/creating-and-highlighting-code-blocks) to make it easier to read.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots or videos to help explain your problem.

**Environment (please complete the following information, where applicable):**
- Unity Version: [e.g. Unity 2020.2.0f1]
- Unity machine OS + version: [e.g. Windows 10]
- ROS machine OS + version: [e.g. Ubuntu 18.04, ROS Noetic]
- ROS–Unity communication: [e.g. Docker]
- Branch or version: [e.g. v0.2.0]

**Additional context**
Add any other context about the problem here.

vidurvij-Unity, 2021-08-23:
I checked out the project, it seems that your folder hierarchy for ROS packages is incorrect. All the packages should be in src level of your working directory but it seems you have moved "unity_robotics_demo_msgs" inside your Robo_simulation folder. I moved it to unity_gazebo_ros/src folder and completed the build. 
I hope this helped and feel free to ask any other questions you have. !

Akashleena, 2021-08-24:
Thanks that works.
I headed on and copied and pasted the ROS_TCP_ENDPOINT package in my catkin_ws
A catkin_make builds the package successfully.

However on doing $rosrun ros_tcp_endpoint default_server_endpoint.py 

I get this error:
Traceback (most recent call last):
  File "/home/akashleena/gazebo_unity_ros_ws/src/ROS-TCP-Endpoint/src/ros_tcp_endpoint/default_server_endpoint.py", line 3, in <module>
    import rospy
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/__init__.py", line 47, in <module>
    from std_msgs.msg import Header
  File "/opt/ros/noetic/lib/python3/dist-packages/std_msgs/msg/__init__.py", line 1, in <module>
    from ._Bool import *
  File "/opt/ros/noetic/lib/python3/dist-packages/std_msgs/msg/_Bool.py", line 6, in <module>
    import genpy
  File "/opt/ros/noetic/lib/python3/dist-packages/genpy/__init__.py", line 34, in <module>
    from . message import Message, SerializationError, DeserializationError, MessageException, struct_I
  File "/opt/ros/noetic/lib/python3/dist-packages/genpy/message.py", line 48, in <module>
    import yaml
ImportError: No module named yaml

Akashleena, 2021-08-25:
Is there a way to align Unity and Gazebo co-ordinate system?

vidurvij-Unity, 2021-08-25:
ROS TCP Connector provides APIs for converting coordinate systems.
https://github.com/Unity-Technologies/ROS-TCP-Connector/blob/main/ROSGeometry.md
If you have any further questions, feel free to open a new issue.
Vidur

Akashleena, 2021-08-25:
Yup, thanks a tonne, the original issue was resolved

closed 2021-08-25 by vidurvij-Unity