Unity-Technologies/Unity-Robotics-Hub #297: Cannot get the ROS-Unity Integration to work!

Miller4603, 2021-08-20:
I am trying to do the ros-unity integration tutorials. I am trying to do it on 2 computers, so ROS runs on my Linux and Unity runs on my Mac. 

My error occurs when I press play in Unity. The connection is red and not BLUE like it is suppose to be, meaning there is no connection between the 2 computers. 

I am using the endpoint launch file and when I run it the server_endpoint node does start and it says "starting server on THE CORRECT ROS_IP".

Here is a couple side notes:
1. Both computers are on the same wireless network.
2. This is probably the issue but unsure how to resolve it. I am told to install unity_robotics_demo and unity_robotics_demo_msgs. Then I am suppose to catkin_make and source my workspace. I did not do this though. INSTEAD, I installed those packages on my Mac. The reason is because I need to use them in Unity shown here "https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/main/tutorials/ros_unity_integration/setup.md" at bottom of page. 

- Unity Version: Unity 2020.3.6f1
- Unity machine OS + version: Mac OS
- ROS machine OS + version: Ubuntu 18.04, ROS Melodic
- ROS–Unity communication: Trying to do it between two computers
- Branch or version: [e.g. v0.2.0]

The reason there is no log files of my terminal is because I am struggling to log in to GitHub on my Linux. If you NEED them, I can try and figure it out. 

NO ERRORS are thrown on either computer.

Thank you in advance! I know I did not provide a ton of information so if you need more, just ask!

vidurvij-Unity, 2021-08-24:
1. I would first ask you to check if you are able to communicate with the other computer over your network. Try pinging your linux machine from your unity system.
2. Double check the IP address in your ROS system and see if it is correct and there are no firewall issues.
3.  While in play mode check if you get any error similar in Unity in the form of a Debug.Log message.
4. And we would recommend you install those packages in your Linux machine and run catkin_make as recommended in instructions.

vidurvij-Unity, 2021-09-23:
From the looks of it, you have network issues. I before going ahead with using ROS-TCP-Connector you should make sure that you two terminals are communicating.