Unity-Technologies/Unity-Robotics-Hub #298: URDF Tutorials Appendix Links not working

A-Ivan, 2021-08-23:
The table of contents links on the "URDF Tutorials Appendix" [document ](https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/main/tutorials/urdf_importer/urdf_appendix.md) are not working.

For instance:

File Hierarchy
GameObject Hierarchy
URDF Comparator
Articulation Body axis definition

The problem is that there is an extra "#" symbol in their links, making these links unclickable, I believe it should only contain one of these symbols, not two.

An example is File Hierarchy currently has the following link "https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/main/tutorials/urdf_importer/urdf_appendix.md**##**File-Hierarchy"

I believe it should be "https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/main/tutorials/urdf_importer/urdf_appendix.md**#**File-Hierarchy".

This change should fix all of these links.

vidurvij-Unity, 2021-08-23:
Thank you for pointing this out to us. I have created a pull requested to make the necessary changes.

A-Ivan, 2021-08-23:
You're welcome

closed 2021-08-23 by vidurvij-Unity