Unity-Technologies/Unity-Robotics-Hub #303: Using ROS 2 on UWP / Hololens

0tobsam0, 2021-08-31:
According to the [this comment](https://github.com/Unity-Technologies/ROS-TCP-Connector/issues/78#issuecomment-810161167) the ROS-TCP-Connector should be platform independent and support UWP. After following [these instructions](https://github.com/Unity-Technologies/Unity-Robotics-Hub/issues/215#issuecomment-832181800) to enable the package for UWP the build process was successful, however the console was filled with exceptions. We later found out, that when switching the build platform within Unity from PC to UWP the protocol in the "ROS Settings" menu returns to ROS1. However when trying to reenable ROS2 the message "(Recompiling, please wait...)" just stays indefinitely. When reopening the "ROS Settings" menu it defaults back to ROS1. That obviously explains any further connection issues and exceptions, because the app is using the wrong protocol.

Environment:
Unity 2020.3.17
ROS-TCP-Connector v0.5.0
Trying to develop for Hololens 2
using MRTK 2.5.1

How can we enable ROS2 on UWP?

hyounesy, 2021-08-31:
Thanks for reporting the issue. We will investigate and return back to you.
Ticket: AIRO-1187

hyounesy, 2021-09-02:
Hi @0tobsam0,
The issue is verified and has been resolved. The fix will be available in the next release of the [ROS-TCP-Connector](https://github.com/Unity-Technologies/ROS-TCP-Connector) package. However if you would like to access it earlier, you may update your Unity project to use the dev branch of the package by modifying the Project\Packages\manifest.json in a text editor and replacing the line:
`    "com.unity.robotics.ros-tcp-connector": "https://github.com/Unity-Technologies/ROS-TCP-Connector.git?path=/com.unity.robotics.ros-tcp-connector#v0.5.0",
`
with:
`    "com.unity.robotics.ros-tcp-connector": "https://github.com/Unity-Technologies/ROS-TCP-Connector.git?path=/com.unity.robotics.ros-tcp-connector#dev",
`

closed 2021-10-09 by vidurvij-Unity