Unity-Technologies/Unity-Robotics-Hub #307: Failed to run an image build from docker

duongtran11, 2021-09-03:
**Describe the bug**
I cloned repository to my Windows 10 computer then I want to run with docker. I built an image for ROS2 Foxy but then failed to run it. 

**To Reproduce**
Steps to reproduce the behavior:
Navigate to tutorials/ros_unity_integration in the repo and run the following commands:

docker build -t melodic -f ros_docker/Dockerfile .
docker run -it --rm -p 10000:10000 melodic /bin/bash

**Console logs / stack traces**
```Admin@Admin MINGW64 /d/PRATI/Autoware/Unity-Robotics-Hub/tutorials/ros_unity_integration (main)
$ docker build -t foxy -f ros2_docker/Dockerfile .
#1 [internal] load build definition from Dockerfile
#1 sha256:9ce4c5f1d712d04f7c0a336834998a3c4ab42f6af26887402033cc98c837ba1a
#1 transferring dockerfile: 32B 0.1s done
#1 DONE 1.0s

#2 [internal] load .dockerignore
#2 sha256:996cd8fdf54f00e9974870ed01185467a73a570ca9192abf9a13f66d8c63dc1b
#2 transferring context: 2B 0.1s done
#2 DONE 1.4s

#3 [internal] load metadata for docker.io/library/ros:foxy-ros-base
#3 sha256:71d378aca1a34aad9c08ea18034f441327bf1cb8f9633c9505a04fb150ac14bd
#3 DONE 6.8s

#4 [1/8] FROM docker.io/library/ros:foxy-ros-base@sha256:5d18bf5492103d479c760141ee88d33a2f4478367c418a56585c474da1036bc4
#4 sha256:c212d9975b1681cc665b34163d1fee7a4a0c1f8557a1f28353f37be8e7baeda1
#4 DONE 0.0s

#6 [internal] load build context
#6 sha256:74c55441fe1cc9e974ceb5ee7621c9e77e29a5258ee645c0e51de9ea728e067f
#6 transferring context:
#6 transferring context: 1.75kB 0.0s done
#6 DONE 0.4s

#5 [2/8] RUN mkdir -p /home/dev_ws/src
#5 sha256:a1cb9abd997befd01f4fdb5ca363270c54cc61389d774b4eb9688163418a0a34
#5 CACHED

#7 [3/8] COPY ./ros2_packages/ /home/dev_ws/src
#7 sha256:7b58e31c0ee656ce5cd3fbd61f9aff0e0dfc892a5fdafb5c357a7e1b1c38d767
#7 CACHED

#8 [4/8] RUN git clone https://github.com/Unity-Technologies/ROS-TCP-Endpoint /home/dev_ws/src/ros_tcp_endpoint -b ROS2v0.5.0
#8 sha256:bfd419f204f75c6d70872eeebc281076aa59f1041076d1a0bb110613d0abeb0d
#8 CACHED

#9 [5/8] COPY ./ros2_docker/source_ros.sh /home/dev_ws/source_ros.sh
#9 sha256:0479414ce619c42c20356d262c64e6bf9a69f96bd5988c500359cece36ae6ab7
#9 CACHED

#10 [6/8] RUN echo "cd home/dev_ws" >> ~/.bashrc
#10 sha256:1350760f145d311b1b9ec1706609e3ccbc2430cb0cdb5746b18f0e68f39b1df8
#10 CACHED

#11 [7/8] RUN cd home/dev_ws && . /opt/ros/foxy/setup.sh && colcon build
#11 sha256:3e5e51ad31810d02dba03ea5085880635504e6b826635ebeef723a7b07e07d4f
#11 CACHED

#12 [8/8] RUN echo ". install/local_setup.bash" >> ~/.bashrc
#12 sha256:e06691dfcafd2e6e6466929c9dcceff04e25947f171f2df8c948ba9c1e778b39
#12 CACHED

#13 exporting to image
#13 sha256:e8c613e07b0b7ff33893b694f7759a10d42e180f2b4dc349fb57dc6b71dcab00
#13 exporting layers done
#13 writing image sha256:f12d8f04cb163231fb49079cfa3023b97a08d1e445ed888c8a5487517781f961
#13 writing image sha256:f12d8f04cb163231fb49079cfa3023b97a08d1e445ed888c8a5487517781f961 0.1s done
#13 naming to docker.io/library/foxy done
#13 DONE 1.3s

Use 'docker scan' to run Snyk tests against images to find vulnerabilities and learn how to fix them

Admin@Admin MINGW64 /d/PRATI/Autoware/Unity-Robotics-Hub/tutorials/ros_unity_integration (main)
$ winpty docker run -it --rm -p 10000:10000 foxy /bin/bash
/ros_entrypoint.sh: line 6: /C:/Program Files/Git/usr/bin/bash.exe: No such file or directory```

**Expected behavior**
I expected a docker container would start

**Screenshots**
![2021-09-03_222034](https://user-images.githubusercontent.com/67900559/132032603-153d0bf9-04a4-4d20-9d9b-a4fdfb8cfc2f.png)

**Environment (please complete the following information, where applicable):**
- Unity Version: 
- Unity machine OS + version: Windows 10 Pro 21H1
- ROS machine OS + version: 
- ROS–Unity communication: Docker
- Branch or version: main

hyounesy, 2021-09-07:
Hi @duongtran11,
In your comment above, looks like you are trying to build the ROS1 melodic image. But the screenshot shows ROS2 foxy. 

1.  Could you please verify that you are actually building ROS2 image as specified in [the tutorial](https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/main/tutorials/ros_unity_integration/setup.md#-ros2-environment):
  ```
  docker build -t foxy -f ros2_docker/Dockerfile .
  docker run -it --rm -p 10000:10000 foxy /bin/bash
  ```
2. Based on the error `/C:/Program Files/Git/usr/bin/bash.exe: No such file or directory`. Could you please verify that your git bash is actually setup properly?

duongtran11, 2021-09-07:
hi @hyounesy 
1. I'm building ROS2 image 
2. I'm not sure about this, I'm gonna check it

I found a workaround by delete the /bin/bash in the end of the comman `docker run -it --rm -p 10000:10000 foxy /bin/bash`. Is it a good practice?
And I have one minor question that I think it quite basic so I don't want to open a new issue: I couldn't get the HUD panel show up when I hit play in Unity. This is my setup following tutorial: 
-I install ROS_TCP_Connector through Package Manager in Unity
-I run command `ros2 run ros_tcp_endpoint default_server_endpoint --ros-args -p ROS_IP:=0.0.0.0` inside docker 
-I set Unity side like this
[image]
Nothing was printed in Unity console, or in docker console

peifeng-unity, 2021-09-07:
Hi @duongtran11, it is good to know that it works for you by removing `/bin/bash` at the end of the command line. The `/bin/bash` is optional for docker run and docker will use the default command if it is not provided [[reference](https://docs.docker.com/engine/reference/run/#cmd-default-command-or-options)].

To show the `HUD`, you may need to instantiate the `ROSConnection`, such as [the instantiation](https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/main/tutorials/pick_and_place/Scripts/SourceDestinationPublisher.cs#L32) in the `SourceDestinationPublisher.cs`. The HUD should appear if you could follow the [tutorial Part-2](https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/main/tutorials/pick_and_place/2_ros_tcp.md). Please let us know if it works for you.

duongtran11, 2021-09-08:
Hi @peifeng-unity 
I've made it work! I forgot to create a GameObject and asign Ros Publisher script to it.
Thank you so much.

closed 2021-09-08 by duongtran11