Unity-Technologies/Unity-Robotics-Hub #309: Making Unity service callback asynchronous?

HainingLuo, 2021-09-14:
Hi, 

I have been trying to create a service from Unity side, which receives requests from ROS, executes a series of actions and returns the resulting states back to ROS as the response. The problem is that the actions need to be executed across a few updates. However I do not think that the callback can be declared as async.

My current sulotion is using a publisher and subscriber combination. The subscriber callback registers the action at Unity side, execute one at every update, then publish the states to the ROS side. The side effect is, by using `wait_for_message` from ROS, a delay for about 200ms is introduced.

I think this can be solved by moving part after `service.callback` to a separate async function in file `ROSConnection.cs`. However, I have not checked tried yet and I don't know how to change the `ImplementService` to accept an async callback.

Please let me know if I can be of any help. Many thanks.

peifeng-unity, 2021-09-14:
Thank you for bringing this up to us. I have created an internal ticket AIRO-1243. We will follow up and come back to you soon.

HainingLuo, 2021-09-14:
> Thank you for bringing this up to us. I have created an internal ticket AIRO-1243. We will follow up and come back to you soon.

Thanks! Looking forward to hearing from you soon.

sdiao, 2021-09-24:
Hi @HainingLuo , we are looking into supporting this feature request. We will let you know once we have a branch that you can try out the feature. Thanks.

closed 2021-10-09 by vidurvij-Unity