Unity-Technologies/Unity-Robotics-Hub #310: mesh disappear if a component is used more than once

tonywangziteng, 2021-09-17:
**Describe the bug**
I exported a urdf file from solidworks. The model is an assembly and some of the components are used more than once. The first time I imported it into Unity, it works perfectly. However, if I open it again, some meshes are missing.

**Screenshots**
first import
[image]

second time opening
[image]

**Environment (please complete the following information, where applicable):**
- Unity Version: Unity 2021.2.0b8
- Unity machine OS + version: Windows 10
- Branch or version: v0.5.0-preview

kosswald, 2021-09-17:
Hi, thank you for reporting this bug. We are aware of the issue and are currently addressing it in #https://github.com/Unity-Technologies/URDF-Importer/pull/136. It should be updated soon and if the problem persists please follow up with us.

at669, 2021-09-27:
The PR for this issue has been merged into `dev` and will be in the next release of the URDF-Importer. Please feel free to try it out on `dev`. I'll be closing this issue, but feel free to reopen it if the problem persists, or open a new ticket if you encounter other issues.

closed 2021-09-27 by at669