Unity-Technologies/Unity-Robotics-Hub #311: Is there a way to Unsubscribe from a specific topic without ending the connection?

jackmca, 2021-09-20:
**Is your feature request related to a problem? Please describe.**
Because I am dealing with many robots, I need certain topics to be subscribed to and unsubscribed to at certain times during runtime.

**Describe the solution you'd like**
A way to unsubscribe from a specific topic, that might function similarly to how subscribing works:
ROSConnection.instance.Unsubscribe<MessageType>("/etc", example);

**Describe alternatives you've considered**
The only reliable way I've seen to unsubscribe is to end the connection entirely, however this is not ideal.

**Additional context**
I may be missing something here, would appreciate any replies!

sdiao, 2021-09-21:
Thanks for reaching out. We are reviewing this feature request. Tracked in AIRO-1269.

sdiao, 2021-09-21:
Hi @jackmca , we have "Unsubscribe" method available in branch `laurie/VisualizationPackage`: https://github.com/Unity-Technologies/ROS-TCP-Connector/blob/laurie/VisualizationPackage/com.unity.robotics.ros-tcp-connector/Runtime/TcpConnector/ROSConnection.cs#L189 . 

In Unity Editor's Package Manager, you can use the following git url to add the ROS-TCP-Connector package `https://github.com/Unity-Technologies/ROS-TCP-Connector.git?path=/com.unity.robotics.ros-tcp-connector#laurie/VisualizationPackage`

If you would like to try it, you will need to use the `dev` branch for [ROS-TCP-Endpoint]( https://github.com/Unity-Technologies/ROS-TCP-Endpoint/tree/dev) on ROS.

jackmca, 2021-09-22:
Thanks for the reply! I will check this out.

closed 2021-09-22 by jackmca

ricmua, 2022-03-14:
I am also looking for an unsubscribe feature, but the  the VisualizatioPackage link is broken. Before I submit a bug report or feature requesting, I just want to check: Are there any updates to this? @sdiao

My Unity script is currently calling the [Unsubscribe](https://github.com/Unity-Technologies/ROS-TCP-Connector/blob/c27f00c6cf750d2d0564349b3039d19aa3925e7c/com.unity.robotics.ros-tcp-connector/Runtime/TcpConnector/ROSConnection.cs#L203) function, which seems to call [UnsubscribeAll](https://github.com/Unity-Technologies/ROS-TCP-Connector/blob/c27f00c6cf750d2d0564349b3039d19aa3925e7c/com.unity.robotics.ros-tcp-connector/Runtime/TcpConnector/RosTopicState.cs#L166). This calls [SendSubscriberUnregistration](https://github.com/Unity-Technologies/ROS-TCP-Connector/blob/c27f00c6cf750d2d0564349b3039d19aa3925e7c/com.unity.robotics.ros-tcp-connector/Runtime/TcpConnector/ROSConnection.cs#L396), which sends the [remove_subscriber](https://github.com/Unity-Technologies/ROS-TCP-Connector/blob/c27f00c6cf750d2d0564349b3039d19aa3925e7c/com.unity.robotics.ros-tcp-connector/Runtime/TcpConnector/SysCommand.cs#L21) command. That results in the following error:

```
Exception in thread Thread-2:
Traceback (most recent call last):
  File "/usr/lib/python3.8/threading.py", line 932, in _bootstrap_inner
    self.run()
  File "ros_tcp_endpoint/client.py", line 225, in run
    self.tcp_server.handle_syscommand(destination, data)
  File "ros_tcp_endpoint/lib/python3.8/site-packages/ros_tcp_endpoint/server.py", line 114, in handle_syscommand
    function = getattr(self.syscommands, topic[2:])
AttributeError: 'SysCommands' object has no attribute 'remove_subscriber'
```

jackmca, 2022-03-15:
Hello, from what I understand, there have been no updates to Unsubscribing, the Unity side of the ROS messaging has added support for Unsubscribing (As you mentioned), but the actual ROS Endpoint does not have a handle for when this is called. Looks like the Endpoint needs updating.

Cheers,
Jack

KevinYuki, 2022-09-08:
Hello. I'm facing the same problem as @ricmua here. Is there any updates about the unsubscribe method? @sdiao

ricmua, 2022-09-08:
None from me, @KevinYuki . I've largely just tried to work around the problem.