Unity-Technologies/Unity-Robotics-Hub #313: Robotic hand not grabbing the target

Rajawat23, 2021-09-20:
While following the tutorial, I was able to get to part 3 of the tutorial.
In part 3 while trying to run, the robotic arm doesn't seem to grab and pick the target.
There aren't any errors published on the console. The ros is running on the docker provided with the tutorial.

**Expected behaviour**
The robot arm grabs the target and moves it to the target position.

**Screenshots**
Attaching the recording with the issue.

**Environment (please complete the following information, where applicable):**
- Unity Version: 2020.3.12f1
- Unity machine OS + version: Macbook air, Big Sur

https://user-images.githubusercontent.com/5962751/134021422-1db5d9df-ef1b-46b0-863d-32248afc9fb3.mov

sdiao, 2021-09-21:
Thanks for reporting the issue. Do you see any error/warning on ROS side?

sdiao, 2021-09-21:
Thanks for sharing the log. ROS log seems good and the path planing was successful. 
In your Unity Editor, did you happen to disable the collider on the cube or the gripper?

Rajawat23, 2021-09-22:
Hi @sdiao ,
thanks for you reply, I am sorry I am new to the unity. 
How can I make sure they are enabled in both components?
All I can find https://docs.unity3d.com/ScriptReference/Collider-enabled.html 
and not sure where to check if this piece of code should be integrated.

sdiao, 2021-09-24:
Hi @Rajawat23 , you check check the collider by:
1. Go to the hierarchy, and select the game object, e.g. Target
2. Check the Inspector, which shows the Collider. We need to make sure the Collider is selected. For example, "Box Collider" is selected in the screenshot below.
[image]

However, if you haven't modify the collider in the Inspector, I doubt the collider would be the root cause here. 

Could you try the quick demo of the Pick and Place following instruction in https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/main/tutorials/pick_and_place/quick_demo.md. Does the arm pick the cube successfully for you?

Rajawat23, 2021-09-27:
Hi @sdiao,

I checked the collider was selected, however deleting and working on project from the scratch did make it work. 
Thanks for the help. 
I will go ahead and close the issue :)

closed 2021-09-27 by Rajawat23

sdiao, 2021-09-27:
@Rajawat23 , I am glad building the Pick and Place project from scratch works. Please feel free to reach out if you have any question in the future.