Unity-Technologies/Unity-Robotics-Hub #314: Determining a stable and successful ROS connection in unity

NathanielRose, 2021-09-22:
**Is your feature request related to a problem? Please describe.**
Is there a way to check an event with the ROSConnection to confirm that an established connection has been created?

I am looking for the event to account for errors where:
```
No data available on network stream after 10 attempts.
```
Or when
```
No connection could be made because the target machine actively refused it?
```
What should I look for as an event to confirm when a stable connection has been made?

**Describe the solution you'd like**

```
ros = ROSConnection.instance
while (ros.ConnectionEstablished)
{
    //Do Work
}

```

**Describe alternatives you've considered**
I am aware of `ros.isActiveAndEnabled` but this is true for the thrown errors mentioned above.

NathanielRose, 2021-09-22:
Additionally, as I move from scene to scene. Let's say from a start menu to my ROS arm implementation, my TCP connection seems to get overloaded as my scenes are both trying to establish a ROS connection on the same port/address?

```
Exception raised!! System.Net.Sockets.SocketException (0x80004005): Only one usage of each socket address (protocol/network address/port) is normally permitted.
```

sdiao, 2021-09-23:
Hi @NathanielRose , may I know which version of the ROS-TCP-Connector that you are using? and what is your OS and how you run ROS?

We have made a lot of improvement in the ROS-TCP-Connector. I would recommend getting the latest version v0.5.0 for both ROS-TCP-Connector and ROS-TCP-Endpoint. 

As @LaurieCheers-unity recommended in the forum thread https://forum.unity.com/threads/ros-connection-success-event.1172993/ , you can use the flag `HasConnectionError ` to detect if there is any connection. See code:
https://github.com/Unity-Technologies/ROS-TCP-Connector/blob/v0.5.0/com.unity.robotics.ros-tcp-connector/Runtime/TcpConnector/ROSConnection.cs#L76 

Since v0.4.0, you will only need to establish the connection between Unity and ROS once. You should not need to establish connection again even when the scene switches. 

Please try out the v0.5.0 of both [ROS-TCP-Connector](https://github.com/Unity-Technologies/ROS-TCP-Connector/tree/v0.5.0) and [ROS-TCP-Endpoint](https://github.com/Unity-Technologies/ROS-TCP-ENDPOINT/tree/v0.5.0). Let us know if it helps. Thanks.

NathanielRose, 2021-09-27:
Thanks @sdiao , I updated and was able to use this method to detect ROS connectivity!

closed 2021-09-27 by NathanielRose

sdiao, 2021-09-27:
Hi @NathanielRose , I am glad that you are able to use `HasConnectionError` to detect ROS connectivity. Feel free to reach out if you have any question and feedback in the future.