Unity-Technologies/Unity-Robotics-Hub #316: MQTT support for Unity-ROS communication

cdh4u, 2021-09-24:
**Is your feature request related to a problem? Please describe.**
Currently, when configuring the TCP connection to ROS, the IP address and port of the ROS endpoint is required. Sometimes the address may vary, which requires re-configuration.

**Describe the solution you'd like**
I suggest support of MQTT in the Unity- and ROS TCP handlers. Unity and ROS would register with a configured MQTT broker, and exchange the ROS messages using MQTT and a configured topic. That way, Unity does not need to know the IP address of the ROS endpoint, only the IP address of the MQTT broker.

**Describe alternatives you've considered**
N/A

**Additional context**
N/A

sdiao, 2021-09-24:
Hi @cdh4u , thanks for submitting the feature request. We are reviewing your request and will let you know once we have an update. Tracked in AIRO-1284.

sdiao, 2021-09-24:
I would recommend that you can turn off the "Connection on Startup" in the "Robotics -> ROS Settings" in Unity Editor, and write some code that calls `RosConnection.instance.Connect(ip, port) ` to establish the connection after fetching the IP address, whether it is from a config file, command line arg or over the network. 

Adding MQTT would introduce new complexity and you still need to register the MQRR broker's IP address if it changes.

The screenshot shows the   "ROS Settings" window where you can turn off "Connection on Startup". 
[image]

cdh4u, 2021-09-24:
Thank You for the reply! It is true that the IP address of the MQTT broker may change too, but normally they don't. For example, I have been using an MQTT broker, with the same IP address, for many years.

mrpropellers, 2021-09-30:
ROS is already designed to handle this type of decentralized network discovery, and the ROS-TCP-Endpoint is in itself a message broker.  Your ROS components do not necessarily need to live on the same device or in the same container as the ROS-TCP-Endpoint, so if you have a device with a relatively static IP that you want to use as the bridge between Unity and your ROS processes, you can launch the ROS-TCP-Endpoint on this device. Roscore (or ROS2 DDS's network discovery) should handle the discovery from there.

cdh4u, 2021-10-09:
> ROS is already designed to handle this type of decentralized network discovery, and the ROS-TCP-Endpoint is in itself a message broker. Your ROS components do not necessarily need to live on the [...]

I was thinking about that. However, at least in my case I believe that there are some issues with that:

First, you need to be able to establish a multi-port bi-directional connection between the ROS devices. Firewalls are most likely going to prevent that. Yes, you could perhaps establish a VPN between the ROS devices, but that would in practice require static IP addresses on both devices (it is possible to establish VPNs where only one side has a static address, but I think there are lots of restrictions etc associated with that). 

Second, the ROS devices would run in different IP address domains. As far as I understand, in order for that to work the IP addresses would have to be static.

Now, I am not a ROS expert, so I am happy to be told that the above assumptions are wrong :)