Unity-Technologies/Unity-Robotics-Hub #317: No response data from service 'niryo_moveit'!

lilmik, 2021-09-26:
[ INFO] [1632640407.705825997]: Ready to take commands for planning group arm.
[ INFO] [1632640407.724244324]: Planning request received for MoveGroup action. Forwarding to planning pipeline.
[ INFO] [1632640407.725349154]: Planner configuration 'arm' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[ INFO] [1632640407.725612667]: arm/arm: Starting planning with 1 states already in datastructure
[ INFO] [1632640407.747327314]: arm/arm: Created 5 states (2 start + 3 goal)
[ INFO] [1632640407.747400752]: Solution found in 0.021847 seconds
[ INFO] [1632640407.752699740]: SimpleSetup: Path simplification took 0.005003 seconds and changed from 4 to 2 states
**[ERROR] [1632640407.757843]: Error processing request: 'tuple' object has no attribute 'joint_trajectory'
['Traceback (most recent call last):\n', '  File "/home/ros/ros_catkin_ws/install_isolated/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 632, in _handle_request\n    response = convert_return_to_response(self.handler(request), self.response_class)\n', '  File "/home/ros/ros_unity_ws/Unity-Robotics-Hub/tutorials/pick_and_place/ROS/src/niryo_moveit/scripts/mover.py", line 92, in plan_pick_and_place\n    if not pre_grasp_pose.joint_trajectory.points:\n', "AttributeError: 'tuple' object has no attribute 'joint_trajectory'\n"]
Service Exception raised. Attempt: 3
[ERROR] [1632640407.761048]: No response data from service 'niryo_moveit'!**

NathanielRose, 2021-09-26:
I am getting the same output.

sdiao, 2021-09-27:
@lilmik @nathand96 , could you use the following template to add more details of the issues that you are running into?
https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/main/.github/ISSUE_TEMPLATE/bug_report.md 

For Pick and Place project, I would recommend start a new project from scratch following the quick demo before going into the full tutorial: https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/main/tutorials/pick_and_place/quick_demo.md

Menno00, 2022-05-19:
Has anyone fixed this error?