Unity-Technologies/Unity-Robotics-Hub #332: Teleoperating robot produces strange behavior that deviates from Gazebo simulation

karthikm-0, 2021-11-12:
**Describe the bug**
I'm trying to teleoperate a robot by having it subscribe to joint states of a Gazebo simulation of the robot. However, there are moments where my inputs lead to the robot behaving strangely, but this doesn't happen in Gazebo. This makes me think there is something funky going on inside Unity. I've disabled the collisions between all links and am using the VHACD for importing the urdf. Please see a clip of the problematic behavior below. This still seems like a collision issue but I can't seem to pinpoint a fix.

https://gfycat.com/waryevengerbil

Thanks!

LaurieCheers-unity, 2021-11-16:
Hi, judging by the way it flips around in the video, I'm not convinced it's a collision issue. This looks more like you're doing some kind of conversion on your joint angles, and within a certain range the results you get are off by 180 degrees.

Can you share your code that's actually controlling the articulation bodies?

karthikm-0, 2021-11-16:
Hi,

The code is below:
`        // Apply the new joint state-
        for (var j = 0; j < armJoints.Length; j++)
        {
            // Get the current joint position
            float angle;
            
            var jointXDrive = armJoints[j].xDrive;
            // If joint state message
            if (currentJointState != null)
            {
                angle = Mathf.Rad2Deg * Convert.ToSingle(currentJointState.position[j + startofArmJoints]);
                angle = NormalizeAngle(angle, jointXDrive.lowerLimit, jointXDrive.upperLimit);
                jointXDrive.target = angle;
                
            }
            
            // If trajectory point message
            else if (currentTrajPoint != null)
            {
                angle = Mathf.Rad2Deg * Convert.ToSingle(currentTrajPoint.positions[j + startofArmJoints]);
                angle = NormalizeAngle(angle, jointXDrive.lowerLimit, jointXDrive.upperLimit);
                jointXDrive.target = angle;
            }
            
            armJoints[j].xDrive = jointXDrive;
        }`

karthikm-0, 2021-11-16:
Hi,

Please see the code below:

    public static void MoveJoints(ArticulationBody[] armJoints, JointStateMsg currentJointState=null,
        JointTrajectoryPointMsg currentTrajPoint=null, int startofArmJoints=2)
    {
        // Apply the new joint state-
        for (var j = 0; j < armJoints.Length; j++)
        {
            // Get the current joint position
            float angle;
            
            var jointXDrive = armJoints[j].xDrive;
            // If joint state message
            if (currentJointState != null)
            {
                angle = Mathf.Rad2Deg * Convert.ToSingle(currentJointState.position[j + startofArmJoints]);
                angle = NormalizeAngle(angle, jointXDrive.lowerLimit, jointXDrive.upperLimit);
                jointXDrive.target = angle;
                
            }
            
            // If trajectory point message
            else if (currentTrajPoint != null)
            {
                angle = Mathf.Rad2Deg * Convert.ToSingle(currentTrajPoint.positions[j + startofArmJoints]);
                angle = NormalizeAngle(angle, jointXDrive.lowerLimit, jointXDrive.upperLimit);
                jointXDrive.target = angle;
            }
            
            armJoints[j].xDrive = jointXDrive;
        }
    }

LaurieCheers-unity, 2021-11-16:
I can't tell what your NormalizeAngle function is for, but I'm pretty sure it's causing your bug. (When angle is supposed to be max, it returns min instead?)

If you try removing the calls to NormalizeAngle, I suspect the problem will go away.

karthikm-0, 2021-11-19:
Ahhh, that worked! I think this function was something I tried to use back when the ArticulationBodies were causing instabilities with the robot. But that has now been resolved so the function doesn't serve a purpose. I completely forgot about its existence. Thank you so much for helping me pinpoint this

LaurieCheers-unity, 2021-11-19:
You're welcome!

closed 2021-11-19 by LaurieCheers-unity