Unity-Technologies/Unity-Robotics-Hub #333: Part4 : "Error : Failed to resolve message name: list index out of range"

zarumo, 2021-11-12:
Hello. 

I finished parts 0~3 of Pick-and-Place tutorials using Ubuntu18.04 PC.
Now I am trying to do Part 4 with my real Niryo One, but I am having trouble getting errors and don't know what to do.

I connected my PC to Niryo One via Ethernet on Ubuntu.
I launched Niryo One Studio, got the calibration done, and then executed the Unity scene.
Both simulated Niryo One and real Niryo One didn't move.
The errors are as below. 

[After I pressed Play in part 4 Unity scene]

- Unity side
[image]
```
SysCommand.subscribe - Unknown message class ''
UnityEngine.Debug:LogError (object)
Unity.Robotics.ROSTCPConnector.ROSConnection:ReceiveSysCommand (string,string) (at Library/PackageCache/com.unity.robotics.ros-tcp-connector@7f055108f4/Runtime/TcpConnector/ROSConnection.cs:633)
Unity.Robotics.ROSTCPConnector.ROSConnection:Update () (at Library/PackageCache/com.unity.robotics.ros-tcp-connector@7f055108f4/Runtime/TcpConnector/ROSConnection.cs:578)
```

- ROS side
[image]
`[ERROR] [1636725544.585528]: Failed to resolve message name: list index out of range`

[After I pressed "publish" button in playing the scene]

- ROS side
No change from above.

- Unity side
[image]
```
Exception: No registered publisher on topic sim_real_pnp for type niryo_moveit/MoverService!
Unity.Robotics.ROSTCPConnector.ROSConnection.Publish (System.String rosTopicName, Unity.Robotics.ROSTCPConnector.MessageGeneration.Message message) (at Library/PackageCache/com.unity.robotics.ros-tcp-connector@7f055108f4/Runtime/TcpConnector/ROSConnection.cs:956)
RealSimPickAndPlace.PublishJoints () (at Assets/RealSimPickAndPlace.cs:136)
UnityEngine.Events.InvokableCall.Invoke () (at /home/bokken/buildslave/unity/build/Runtime/Export/UnityEvent/UnityEvent.cs:180)
UnityEngine.Events.UnityEvent.Invoke () (at /home/bokken/buildslave/unity/build/Runtime/Export/UnityEvent/UnityEvent/UnityEvent_0.cs:58)
UnityEngine.UI.Button.Press () (at Library/PackageCache/com.unity.ugui@1.0.0/Runtime/UI/Core/Button.cs:70)
UnityEngine.UI.Button.OnPointerClick (UnityEngine.EventSystems.PointerEventData eventData) (at Library/PackageCache/com.unity.ugui@1.0.0/Runtime/UI/Core/Button.cs:114)
UnityEngine.EventSystems.ExecuteEvents.Execute (UnityEngine.EventSystems.IPointerClickHandler handler, UnityEngine.EventSystems.BaseEventData eventData) (at Library/PackageCache/com.unity.ugui@1.0.0/Runtime/EventSystem/ExecuteEvents.cs:57)
UnityEngine.EventSystems.ExecuteEvents.Execute[T] (UnityEngine.GameObject target, UnityEngine.EventSystems.BaseEventData eventData, UnityEngine.EventSystems.ExecuteEvents+EventFunction`1[T1] functor) (at Library/PackageCache/com.unity.ugui@1.0.0/Runtime/EventSystem/ExecuteEvents.cs:272)
UnityEngine.EventSystems.EventSystem:Update() (at Library/PackageCache/com.unity.ugui@1.0.0/Runtime/EventSystem/EventSystem.cs:501)
```

I commented out one line in "Real Sim Pick And Place" script as below, and the result changed; playing the unity scene itself doesn't cause an error (Pressing "Publish" Button still causes errors.)

[image]
commented out:
[image]

[Before pressing "Publish" button]
- Unity side & ROS side
No errors.

[After pressing "Publish" button]

- Unity side
[image]
```
Exception: No registered publisher on topic sim_real_pnp for type niryo_moveit/MoverService!
Unity.Robotics.ROSTCPConnector.ROSConnection.Publish (System.String rosTopicName, Unity.Robotics.ROSTCPConnector.MessageGeneration.Message message) (at Library/PackageCache/com.unity.robotics.ros-tcp-connector@7f055108f4/Runtime/TcpConnector/ROSConnection.cs:956)
RealSimPickAndPlace.PublishJoints () (at Assets/RealSimPickAndPlace.cs:136)
UnityEngine.Events.InvokableCall.Invoke () (at /home/bokken/buildslave/unity/build/Runtime/Export/UnityEvent/UnityEvent.cs:180)
UnityEngine.Events.UnityEvent.Invoke () (at /home/bokken/buildslave/unity/build/Runtime/Export/UnityEvent/UnityEvent/UnityEvent_0.cs:58)
UnityEngine.UI.Button.Press () (at Library/PackageCache/com.unity.ugui@1.0.0/Runtime/UI/Core/Button.cs:70)
UnityEngine.UI.Button.OnPointerClick (UnityEngine.EventSystems.PointerEventData eventData) (at Library/PackageCache/com.unity.ugui@1.0.0/Runtime/UI/Core/Button.cs:114)
UnityEngine.EventSystems.ExecuteEvents.Execute (UnityEngine.EventSystems.IPointerClickHandler handler, UnityEngine.EventSystems.BaseEventData eventData) (at Library/PackageCache/com.unity.ugui@1.0.0/Runtime/EventSystem/ExecuteEvents.cs:57)
UnityEngine.EventSystems.ExecuteEvents.Execute[T] (UnityEngine.GameObject target, UnityEngine.EventSystems.BaseEventData eventData, UnityEngine.EventSystems.ExecuteEvents+EventFunction`1[T1] functor) (at Library/PackageCache/com.unity.ugui@1.0.0/Runtime/EventSystem/ExecuteEvents.cs:272)
UnityEngine.EventSystems.EventSystem:Update() (at Library/PackageCache/com.unity.ugui@1.0.0/Runtime/EventSystem/EventSystem.cs:501)
```

- ROS side: No change
[image]

After that, I added "Debug.Log()" code in the "Start()" and "ExecuteRobotCommands()" functions as below.

[image]
...
[image]
[The result]
[image]

Log 1 appeared but Log 2 didn't.
It seems that the program went wrong before ExecuteRobotCommands() was called.

**Environment**

Linux PC with Unity (not virtual machine)
- Ubuntu 18.04.6 LTS
- Unity Hub 2.4.5 (for Linux)
- Unity 2021.1.26f1
- Niryo One Studio for Linux
- (ROS Melodic)
- (Docker : but not used in part4)

real Niryo One (Raspberry Pi 3)
- Ubuntu 16.04.7 LTS
- ROS kinetic
- niryo_moveit & ros_tcp_endpoint packages (to home/niryo/catkin_ws/src directory)
- Contents of the home/niryo/catkin_ws/src directory are as below
[image]

Actually, I also tried with Windows PC (& Unity for Windows) connected to Niryo One via wi-fi network, and the result was the same...
Do you have any idea what should I do ?

Thank you for reading.

sdiao, 2021-11-15:
Hi @zarumo , thanks for reaching out and sharing the detailed logs. We are currently investigating the issue and will keep you posted.

PsePaleBlue, 2021-11-23:
See https://forum.unity.com/threads/followjointtrajectory-action-does-not-generate-register-code.1201636/.
Looks like the same issue to me.

zarumo, 2021-12-01:
Thank you @PsePaleBlue.
I looked at your solution and added the same code as you to the RobotMoveActionGoal code, instead of FollowJointTrajectoryActionFeedback code.

As a result, the Unity's 
 _Unknown message class ''_  
error and the Ros's 
 _Failed to resolve message name: list index out of range_  
error disappeared.

However, pressing Publish button still cause the same error below to Unity side
 _No registered publisher on topic sim_real_pnp for type niryo_moveit/MoverService!_

No error appears on the ROS side.

I honestly don't know much about ROS and am not sure if I'm really following your solution.
I edited  RobotMoveActionGoal.cs as below.

```
using System.Collections.Generic;
using Unity.Robotics.ROSTCPConnector.MessageGeneration;
using RosMessageTypes.Std;
using RosMessageTypes.Actionlib;

namespace RosMessageTypes.NiryoOne
{

    public class RobotMoveActionGoal : ActionGoal<RobotMoveGoal>
    {

        public const string k_RosMessageName = "niryo_one_msgs/RobotMoveActionGoal";

        public override string RosMessageName => k_RosMessageName;

        public RobotMoveActionGoal() : base()
        {
            this.goal = new RobotMoveGoal();
        }

        public RobotMoveActionGoal(HeaderMsg header, GoalIDMsg goal_id, RobotMoveGoal goal) : base(header, goal_id)
        {
            this.goal = goal;
        }

        public static RobotMoveActionGoal Deserialize(MessageDeserializer deserializer) => new RobotMoveActionGoal(deserializer);

        RobotMoveActionGoal(MessageDeserializer deserializer) : base(deserializer)
        {
            this.goal = RobotMoveGoal.Deserialize(deserializer);
        }

        public override void SerializeTo(MessageSerializer serializer)
        {
            serializer.Write(this.header);
            serializer.Write(this.goal_id);
            serializer.Write(this.goal);
        }
//Added following @PsePaleBlue solution
#if UNITY_EDITOR
        [UnityEditor.InitializeOnLoadMethod]
#else
        [UnityEngine.RuntimeInitializeOnLoadMethod]
#endif
        public static void Register()
        {
            MessageRegistry.Register(k_RosMessageName, Deserialize, MessageSubtopic.Feedback);
        }
    }
}
```

at669, 2021-12-06:
Hi @zarumo, yes--@PsePaleBlue's link above is a relevant forum thread, as our support for ROS Actions is currently minimal, but we hope to support them better in future. The relevant actions-registering bug has been filed for internal tracking--I'll add this thread to the ticket for tracking as well.  We will let you know when there are updates to this thread!

zarumo, 2021-12-13:
Thank you, @sdiao @PsePaleBlue @at669 @bryceikeda ! 
I am currently using the contents of the tutorial part3 for my university research, and I am very grateful for this repository.
Eventually I hope to work with actual NiryoOne robot, so I'm looking forward to the updates! Thanks!

mrpropellers, 2022-01-13:
Re-opening. Fix should go in when AIRO-1571 is resolved

vidurvij-Unity, 2022-02-06:
AIRO-1571 is resolved

closed 2022-02-06 by vidurvij-Unity