Unity-Technologies/Unity-Robotics-Hub #337: ROSTCPConnector ArgumentOutOfRangeException in PickAndPlaceDemo

PetterVMC, 2021-12-17:
Hi,

When running either DemoScene or a completed part3 of the Pick-And-Place tutorials, occasionally the ROSTCPConnector throws an ArgumentOutOfRangeException upon receiving the trajectory (service response) from ROS.

It usually works fine, but after playing DemoScene many times, it happens about every 20th or 30th execution or so.

- Anyone else experiencing the same problem?

- Is there something we can do to prevent it or is it an issue with the actual ROSTCPConnector implementation?

- Is there a way to detect that the problem has occurred (from another MonoBehaviour)?

**Exception:**

```
ArgumentOutOfRangeException: Index and count must refer to a location within the buffer.
Parameter name: bytes
System.Text.UTF8Encoding.GetString (System.Byte[] bytes, System.Int32 index, System.Int32 count) (at <695d1cc93cca45069c528c15c9fdd749>:0)
Unity.Robotics.ROSTCPConnector.MessageGeneration.MessageDeserializer.Read (System.String& value) (at Library/PackageCache/com.unity.robotics.ros-tcp-connector@7f055108f4/Runtime/MessageGeneration/MessageDeserializer.cs:138)
Unity.Robotics.ROSTCPConnector.MessageGeneration.MessageDeserializer.Read (System.String[]& values, System.Int32 length) (at Library/PackageCache/com.unity.robotics.ros-tcp-connector@7f055108f4/Runtime/MessageGeneration/MessageDeserializer.cs:169)
RosMessageTypes.Trajectory.JointTrajectoryMsg..ctor (Unity.Robotics.ROSTCPConnector.MessageGeneration.MessageDeserializer deserializer) (at Library/PackageCache/com.unity.robotics.ros-tcp-connector@7f055108f4/Runtime/Messages/Trajectory/msg/JointTrajectoryMsg.cs:45)
RosMessageTypes.Trajectory.JointTrajectoryMsg.Deserialize (Unity.Robotics.ROSTCPConnector.MessageGeneration.MessageDeserializer deserializer) (at Library/PackageCache/com.unity.robotics.ros-tcp-connector@7f055108f4/Runtime/Messages/Trajectory/msg/JointTrajectoryMsg.cs:40)
RosMessageTypes.Moveit.RobotTrajectoryMsg..ctor (Unity.Robotics.ROSTCPConnector.MessageGeneration.MessageDeserializer deserializer) (at <b682a19673f04b5ebaddbcd775aa531b>:0)
RosMessageTypes.Moveit.RobotTrajectoryMsg.Deserialize (Unity.Robotics.ROSTCPConnector.MessageGeneration.MessageDeserializer deserializer) (at <b682a19673f04b5ebaddbcd775aa531b>:0)
Unity.Robotics.ROSTCPConnector.MessageGeneration.MessageDeserializer.Read[T] (T[]& values, System.Func`2[T,TResult] loader, System.Int32 length) (at Library/PackageCache/com.unity.robotics.ros-tcp-connector@7f055108f4/Runtime/MessageGeneration/MessageDeserializer.cs:160)
RosMessageTypes.NiryoMoveit.MoverServiceResponse..ctor (Unity.Robotics.ROSTCPConnector.MessageGeneration.MessageDeserializer deserializer) (at <b682a19673f04b5ebaddbcd775aa531b>:0)
RosMessageTypes.NiryoMoveit.MoverServiceResponse.Deserialize (Unity.Robotics.ROSTCPConnector.MessageGeneration.MessageDeserializer deserializer) (at <b682a19673f04b5ebaddbcd775aa531b>:0)
Unity.Robotics.ROSTCPConnector.MessageGeneration.MessageDeserializer.DeserializeMessage[T] (System.Byte[] data) (at Library/PackageCache/com.unity.robotics.ros-tcp-connector@7f055108f4/Runtime/MessageGeneration/MessageDeserializer.cs:25)
Unity.Robotics.ROSTCPConnector.ROSConnection+<SendServiceMessage>d__83`1[RESPONSE].MoveNext () (at Library/PackageCache/com.unity.robotics.ros-tcp-connector@7f055108f4/Runtime/TcpConnector/ROSConnection.cs:314)
--- End of stack trace from previous location where exception was thrown ---
System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw () (at <695d1cc93cca45069c528c15c9fdd749>:0)
System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess (System.Threading.Tasks.Task task) (at <695d1cc93cca45069c528c15c9fdd749>:0)
System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification (System.Threading.Tasks.Task task) (at <695d1cc93cca45069c528c15c9fdd749>:0)
System.Runtime.CompilerServices.TaskAwaiter.ValidateEnd (System.Threading.Tasks.Task task) (at <695d1cc93cca45069c528c15c9fdd749>:0)
System.Runtime.CompilerServices.TaskAwaiter`1[TResult].GetResult () (at <695d1cc93cca45069c528c15c9fdd749>:0)
Unity.Robotics.ROSTCPConnector.ROSConnection+<SendServiceMessage>d__82`1[RESPONSE].MoveNext () (at Library/PackageCache/com.unity.robotics.ros-tcp-connector@7f055108f4/Runtime/TcpConnector/ROSConnection.cs:283)
--- End of stack trace from previous location where exception was thrown ---
System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw () (at <695d1cc93cca45069c528c15c9fdd749>:0)
System.Runtime.CompilerServices.AsyncMethodBuilderCore+<>c.<ThrowAsync>b__6_0 (System.Object state) (at <695d1cc93cca45069c528c15c9fdd749>:0)
UnityEngine.UnitySynchronizationContext+WorkRequest.Invoke () (at <127e81e1cb3441cc97d26b1910daae77>:0)
UnityEngine.UnitySynchronizationContext:ExecuteTasks()
```

[image]

**Environment:**
- Unity Version: 2020.3.11f1
- ROS machine OS + version: Ubuntu 18.04, Melodic, run in Docker
- UnityRoboticsHub / ROSTCPConnector version: v0.6.0

hyounesy, 2021-12-18:
Hi @PetterVMC.  Thank you for reaching out. We have created an internal ticket (AIRO-1647) to investigate the issue. We will come back to you soon.

mrpropellers, 2022-01-12:
Hi @PetterVMC , can you confirm a few things for me so I can better reproduce and investigate this?
1. Do you restart anything between executions or are you simply stopping and starting PlayMode with the play button in editor?
2. Is the error always caused by the JointTrajectoryMessage, or do you think it could occur with any arbitrary ROS message?

PetterVMC, 2022-01-13:
Hi @mrpropellers , thanks for the follow up. See answers below.

> 1. Do you restart anything between executions or are you simply stopping and starting PlayMode with the play button in editor?

Can reproduce it by simply stopping and starting PlayMode with play button, without restarting anything else.

> 2. Is the error always caused by the JointTrajectoryMessage, or do you think it could occur with any arbitrary ROS message?

Haven't tested thoroughly with other messages yet so cannot answer this.

LaurieCheers-unity, 2022-01-20:
I was unable to reproduce this after running the trajectory solver 50 times with various conditions (leaving the cube where it was, moving the cube to random positions, restarting play mode).

Judging by the call stack, we're failing to deserialize an incoming trajectory message - but that doesn't tell us whether it's a bug in the deserializer, or we have garbage data coming in, or what.

In the file com.unity.robotics.ros-tcp-connector/Runtime/MessageGeneration/MessageDeserializer.cs at line 134, could you change the Read(out string value) function to this, try to get the error to happen again, and paste the log message here?

        public void Read(out string value)
        {
            var length = ReadLength();
#if !ROS2
            try
            {
                value = System.Text.Encoding.UTF8.GetString(data, offset, length);
            }
            catch (ArgumentOutOfRangeException e)
            {
                Debug.Log("Error: "+BitConverter.ToString(data));
                value = "";
            }
#else
            // ROS2 strings have a null byte at the end
            value = System.Text.Encoding.UTF8.GetString(data, offset, length - 1);
#endif
            offset += length;
        }

PetterVMC, 2022-01-21:
@LaurieCheers-unity,
I added the provided code, as well as logging of the length and offset variables. then reproduced the error and got the following output:

**length:** 
1970235507

**offset:** 
37

**data:**
`06-00-00-00-00-00-00-00-29-2A-E9-61-EC-80-05-1A-09-00-00-00-62-61-73-65-5F-6C-69-6E-6B-0D-00-00-00-73-68-6F-75-6C-64-65-72-5F-6C-69-6E-6B-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-91-ED-7C-3F-35-5E-BA-3F-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-F0-3F-00-00-00-00-29-2A-E9-61-EC-80-05-1A-0D-00-00-00-73-68-6F-75-6C-64-65-72-5F-6C-69-6E-6B-08-00-00-00-61-72-6D-5F-6C-69-6E-6B-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-7B-14-AE-47-E1-7A-B4-3F-62-95-39-3F-3D-4E-D4-3F-E6-E8-55-A4-C6-38-E4-BF-DD-21-56-A4-C6-38-E4-3F-71-B2-38-3F-3D-4E-D4-3F-00-00-00-00-29-2A-E9-61-EC-80-05-1A-08-00-00-00-61-72-6D-5F-6C-69-6E-6B-0A-00-00-00-65-6C-62-6F-77-5F-6C-69-6E-6B-E1-7A-14-AE-47-E1-CA-3F-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-C9-5B-77-F4-42-E1-EF-3F-4D-49-2C-EF-F5-27-B6-BF-00-00-00-00-29-2A-E9-61-EC-80-05-1A-0A-00-00-00-65-6C-62-6F-77-5F-6C-69-6E-6B-0C-00-00-00-66-6F-72-65-61-72-6D-5F-6C-69-6E-6B-73-68-91-ED-7C-3F-A5-3F-B8-1E-85-EB-51-B8-9E-3F-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-49-7B-7F-66-9E-A0-E6-3F-00-00-00-00-00-00-00-00-51-FC-7E-66-9E-A0-E6-3F-00-00-00-00-29-2A-E9-61-EC-80-05-1A-0C-00-00-00-66-6F-72-65-61-72-6D-5F-6C-69-6E-6B-0A-00-00-00-77-72-69-73-74-5F-6C-69-6E-6B-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-0A-D7-A3-70-3D-0A-C7-3F-00-00-00-00-00-00-00-00-49-7B-7F-66-9E-A0-E6-BF-00-00-00-00-00-00-00-00-51-FC-7E-66-9E-A0-E6-3F-00-00-00-00-29-2A-E9-61-EC-80-05-1A-0A-00-00-00-77-72-69-73-74-5F-6C-69-6E-6B-09-00-00-00-68-61-6E-64-5F-6C-69-6E-6B-09-1B-9E-5E-29-CB-90-3F-BA-49-0C-02-2B-87-76-BF-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-49-7B-7F-66-9E-A0-E6-3F-00-00-00-00-00-00-00-00-51-FC-7E-66-9E-A0-E6-3F`

As you can see, it seems that the length is incorrect. 

If one looks at the 4 bytes from offset=33, from where I believe the length integer is stored, one finds the bytes 73-68-6F-75, which in decimal form matches the extremely large read length of 1970235507.

In addition to the logging above, I logged "healthy" MoverServiceResponses as well. See log below. The messages are quite different. Actually, it looks like the message above that threw the exception is not an MoverServiceResponse, but is extremely similar to those tf2_msgs/TFMessage messages that I've noticed are continuously coming from ROS. Could it perhaps be that  one of those messages is mistakenly parsed as a MoverServiceResponse that causes the error? 

**MoverServiceResponse when exception is not thrown:**
`04-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-05-00-00-00-77-6F-72-6C-64-06-00-00-00-07-00-00-00-6A-6F-69-6E-74-5F-31-07-00-00-00-6A-6F-69-6E-74-5F-32-07-00-00-00-6A-6F-69-6E-74-5F-33-07-00-00-00-6A-6F-69-6E-74-5F-34-07-00-00-00-6A-6F-69-6E-74-5F-35-07-00-00-00-6A-6F-69-6E-74-5F-36-2E-00-00-00-06-00-00-00-69-96-6F-10-E6-08-76-3E-48-D7-6A-96-AB-39-07-BF-45-89-E8-82-DC-5D-E2-3E-43-AA-30-4A-26-8E-1D-BF-5E-0C-FF-A0-DD-FA-F2-BE-26-EE-5A-28-21-CE-1D-3F-06-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-06-00-00-00-35-CC-F9-B2-A7-73-C4-BF-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-06-00-00-00-0D-18-8E-36-F2-05-A9-BF-F2-9B-DA-8E-C0-AB-92-BF-5D-D0-EE-66-F8-7E-7D-BF-A9-4A-50-D3-B7-BA-25-BF-AA-75-BF-C1-49-86-83-BF-BF-71-12-54-5D-F0-A8-BF-06-00-00-00-D0-15-13-4F-16-D5-BA-BF-6B-E3-52-5E-DC-F8-A3-BF-FB-B2-1F-87-BA-AA-8F-BF-D9-44-14-27-39-D1-1D-BF-91-D1-43-EA-78-E5-94-BF-50-27-C3-EB-E9-CD-BA-BF-06-00-00-00-FE-FE-30-DA-E5-72-C3-BF-1D-44-D6-4E-E0-F3-AC-BF-55-26-29-35-F9-F3-96-BF-FD-4A-B0-9D-C3-9C-25-BF-18-74-FF-81-E1-4A-9E-BF-C4-F8-0D-CA-B2-6D-C3-BF-00-00-00-00-00-00-00-00-9B-B3-9E-2E-06-00-00-00-14-79-1C-97-F3-05-B9-BF-3C-01-F5-23-F2-A5-A2-BF-80-8A-0F-DE-8F-83-8D-BF-30-40-88-81-5C-AE-2C-BF-E7-35-57-0A-8B-81-93-BF-7A-88-5C-DC-D0-F7-B8-BF-06-00-00-00-15-CD-46-06-D6-7F-C5-BF-70-87-03-31-AA-00-B0-BF-48-F7-DC-1E-7E-5F-99-BF-E3-61-67-0D-19-E4-27-BF-5A-56-8A-C4-3F-BE-A0-BF-F8-14-C1-9F-16-7A-C5-BF-06-00-00-00-94-DC-87-0E-94-09-C2-BF-70-25-2A-B9-FE-D9-AA-BF-58-3D-69-92-8D-49-95-BF-20-D7-9B-1C-40-0B-24-BF-C0-56-FD-97-1B-18-9C-BF-A9-79-D6-96-C1-04-C2-BF-00-00-00-00-01-00-00-00-B5-28-D0-06-06-00-00-00-11-F3-78-09-77-C4-C2-BF-7E-B4-7C-00-04-F6-AB-BF-69-D6-53-C4-D1-23-96-BF-DC-1A-E0-97-00-D1-31-BF-F8-B0-CE-33-F1-3F-9D-BF-0A-EC-57-87-B9-BB-C2-BF-06-00-00-00-ED-B7-1A-F3-8A-29-CA-BF-73-A7-87-E4-2C-79-B3-BF-A2-BB-ED-B1-46-E0-9E-BF-E2-00-3E-87-98-12-2D-BF-24-05-D9-BC-E0-5F-A4-BF-B2-EE-29-6C-8C-22-CA-BF-06-00-00-00-48-99-A9-39-A0-9C-C0-BF-4F-DA-15-A2-B4-BA-A8-BF-18-41-9A-54-D8-9A-93-BF-23-D6-9F-1C-B3-75-22-BF-1E-E3-34-1A-AD-DF-99-BF-73-D5-0C-53-2F-98-C0-BF-00-00-00-00-01-00-00-00-86-6A-3E-16-06-00-00-00-98-A9-63-47-F4-05-C9-BF-E1-33-82-EE-0A-A3-B2-BF-91-E7-9F-99-DB-85-9D-BF-A0-15-FC-EE-D2-4A-35-BF-06-16-A3-AE-2B-7F-A3-BF-D8-93-81-A0-8A-FB-C8-BF-06-00-00-00-2A-5C-8F-C2-F5-28-CC-BF-B7-C9-C4-89-D6-F5-B4-BF-A6-5F-F3-4B-EB-9D-A0-BF-90-F2-2B-8D-E7-4A-2F-BF-E5-B3-45-21-28-EE-A5-BF-22-F5-98-82-6E-21-CC-BF-06-00-00-00-00-00-00-00-00-00-00-00-AA-F2-8C-AF-71-01-A2-3C-00-00-00-00-00-00-00-00-AA-F2-8C-AF-71-01-22-3C-FF-6B-53-87-2A- [... cut ...] 3-CA-BF-86-B1-3F-90-C8-02-61-BF-92-85-4A-B3-FE-BE-DE-BF-9F-FC-1C-1B-27-69-03-C0-06-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-06-00-00-00-1C-5D-5C-5F-BA-6A-C4-BF-77-46-64-71-4E-CA-62-3F-FA-C1-02-C2-ED-EE-70-BF-A5-EC-BA-87-2E-6C-17-3F-19-AB-77-8A-AE-96-5E-3F-77-86-58-F9-93-81-80-3F-00-00-00-00-15-00-00-00-AB-D7-17-2C-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00`

LaurieCheers-unity, 2022-01-21:
Thanks for this. The data here breaks down as

0-4: [array length 6]
5-29: [header with frame_id = "base_link"]
30-46: [the string "shoulder_link"]
...

This is consistent with a TFMessage (and not a MoverServiceResponse, which would have an array of strings instead of a single string there) - so I agree your guess looks correct.

When sending a service response, the Endpoint first sends a service header (to indicate what request this message is a response to). It must have sent the service header and then failed to actually send the response itself. When it fails, do you see any exceptions in the Endpoint console?

PetterVMC, 2022-01-24:
I checked the console on the ROS side but there doesn't appear to be any different output when it fails. 

Output from Play to Stop:
```
[INFO] [1642992792.336800]: Connection from 172.17.0.1
[INFO] [1642992792.342262]: RegisterSubscriber(/tf, <class 'tf2_msgs.msg._TFMessage.TFMessage'>) OK
[INFO] [1642992792.349264]: RegisterRosService(niryo_moveit, <class 'niryo_moveit.srv._MoverService.MoverService'>) OK
[INFO] [1642992792.351571]: RegisterRosService(niryo_moveit, <class 'niryo_moveit.srv._MoverService.MoverService'>) OK
[ INFO] [1642992792.393561600]: Loading robot model 'niryo_one'...
[ INFO] [1642992792.393707700]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ INFO] [1642992793.528553600]: Ready to take commands for planning group arm.
[ INFO] [1642992793.742690700]: Planning request received for MoveGroup action. Forwarding to planning pipeline.
[ INFO] [1642992793.744084900]: Planner configuration 'arm' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[ INFO] [1642992793.744888500]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1642992793.770522500]: RRTConnect: Created 5 states (2 start + 3 goal)
[ INFO] [1642992793.770803300]: Solution found in 0.026116 seconds
[ INFO] [1642992793.783790500]: SimpleSetup: Path simplification took 0.012716 seconds and changed from 4 to 2 states
[ INFO] [1642992793.802392800]: Planning request received for MoveGroup action. Forwarding to planning pipeline.
[ INFO] [1642992793.803096200]: Planner configuration 'arm' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[ INFO] [1642992793.803421000]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1642992793.814542300]: RRTConnect: Created 4 states (2 start + 2 goal)
[ INFO] [1642992793.814683300]: Solution found in 0.011320 seconds
[ INFO] [1642992793.834845600]: SimpleSetup: Path simplification took 0.020024 seconds and changed from 3 to 2 states
[ INFO] [1642992793.862456100]: Planning request received for MoveGroup action. Forwarding to planning pipeline.
[ INFO] [1642992793.863095700]: Planner configuration 'arm' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[ INFO] [1642992793.863418000]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1642992793.888868000]: RRTConnect: Created 5 states (2 start + 3 goal)
[ INFO] [1642992793.889230900]: Solution found in 0.025875 seconds
[ INFO] [1642992793.908999000]: SimpleSetup: Path simplification took 0.019474 seconds and changed from 4 to 2 states
[ INFO] [1642992793.942468300]: Planning request received for MoveGroup action. Forwarding to planning pipeline.
[ INFO] [1642992793.944304400]: Planner configuration 'arm' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[ INFO] [1642992793.944871700]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1642992793.971721300]: RRTConnect: Created 5 states (2 start + 3 goal)
[ INFO] [1642992793.971973800]: Solution found in 0.027241 seconds
[ INFO] [1642992793.989291300]: SimpleSetup: Path simplification took 0.017102 seconds and changed from 4 to 2 states
[ERROR] [1642992804.503637]: Exception: No more data available
[INFO] [1642992804.511950]: Disconnected from 172.17.0.1
[ERROR] [1642992804.563688]: Exception on Send [Errno 9] Bad file descriptor
```

(The exceptions in the end happen every time on disconnect, i.e. on stop play, regardless of the service response problem)

LaurieCheers-unity, 2022-01-24:
Aha, this must be a threading issue then. This line (https://github.com/Unity-Technologies/ROS-TCP-Endpoint/blob/715a402ec09ce8095faabed915e3371638b4246a/src/ros_tcp_endpoint/tcp_sender.py#L78) queues the service header for sending, and the next line queues the message body, but there's nothing that ensures another thread doesn't queue something inbetween.

I made a fix for this, could you switch to this branch and see if the problem stops happening? https://github.com/Unity-Technologies/ROS-TCP-Endpoint/tree/laurie/ServiceResponseIsOneQueueItem

LaurieCheers-unity, 2022-01-26:
<bump> Would be great to know whether this fix solves the problem. :-D

PetterVMC, 2022-01-27:
Sorry for the wait, just managed to get some time to try the fix.

I've played the simulation like 50~100 times now without being able to reproducing the problem, so it seems like the fix solved it! Thank you :)

LaurieCheers-unity, 2022-01-27:
Great to know, thanks! This fix is now in dev branch and we'll be doing a release to main soon.

closed 2022-01-27 by LaurieCheers-unity

cambel, 2023-06-09:
Hi,

I am getting a similar ArgumentOutOfRangeException error on the pick-and-place tutorial part 3. However, it happens every single time.

```shell
ArgumentOutOfRangeException: Index and count must refer to a location within the buffer.
Parameter name: bytes
System.Text.UTF8Encoding.GetString (System.Byte[] bytes, System.Int32 index, System.Int32 count) (at <4d56092c9a8849cab50a947b257065a3>:0)
Unity.Robotics.ROSTCPConnector.MessageGeneration.MessageDeserializer.Read (System.String& value) (at ./Library/PackageCache/com.unity.robotics.ros-tcp-connector@c27f00c6cf/Runtime/MessageGeneration/MessageDeserializer.cs:138)
Unity.Robotics.ROSTCPConnector.MessageGeneration.MessageDeserializer.Read (System.String[]& values, System.Int32 length) (at ./Library/PackageCache/com.unity.robotics.ros-tcp-connector@c27f00c6cf/Runtime/MessageGeneration/MessageDeserializer.cs:175)
RosMessageTypes.Trajectory.JointTrajectoryMsg..ctor (Unity.Robotics.ROSTCPConnector.MessageGeneration.MessageDeserializer deserializer) (at ./Library/PackageCache/com.unity.robotics.ros-tcp-connector@c27f00c6cf/Runtime/Messages/Trajectory/msg/JointTrajectoryMsg.cs:45)
RosMessageTypes.Trajectory.JointTrajectoryMsg.Deserialize (Unity.Robotics.ROSTCPConnector.MessageGeneration.MessageDeserializer deserializer) (at ./Library/PackageCache/com.unity.robotics.ros-tcp-connector@c27f00c6cf/Runtime/Messages/Trajectory/msg/JointTrajectoryMsg.cs:40)
RosMessageTypes.Moveit.RobotTrajectoryMsg..ctor (Unity.Robotics.ROSTCPConnector.MessageGeneration.MessageDeserializer deserializer) (at Assets/RosMessages/Moveit/msg/RobotTrajectoryMsg.cs:35)
RosMessageTypes.Moveit.RobotTrajectoryMsg.Deserialize (Unity.Robotics.ROSTCPConnector.MessageGeneration.MessageDeserializer deserializer) (at Assets/RosMessages/Moveit/msg/RobotTrajectoryMsg.cs:31)
Unity.Robotics.ROSTCPConnector.MessageGeneration.MessageDeserializer.Read[T] (T[]& values, System.Func`2[T,TResult] loader, System.Int32 length) (at ./Library/PackageCache/com.unity.robotics.ros-tcp-connector@c27f00c6cf/Runtime/MessageGeneration/MessageDeserializer.cs:166)
RosMessageTypes.NiryoMoveit.MoverServiceResponse..ctor (Unity.Robotics.ROSTCPConnector.MessageGeneration.MessageDeserializer deserializer) (at Assets/RosMessages/NiryoMoveit/srv/MoverServiceResponse.cs:32)
RosMessageTypes.NiryoMoveit.MoverServiceResponse.Deserialize (Unity.Robotics.ROSTCPConnector.MessageGeneration.MessageDeserializer deserializer) (at Assets/RosMessages/NiryoMoveit/srv/MoverServiceResponse.cs:28)
Unity.Robotics.ROSTCPConnector.MessageGeneration.MessageDeserializer.DeserializeMessage[T] (System.Byte[] data) (at ./Library/PackageCache/com.unity.robotics.ros-tcp-connector@c27f00c6cf/Runtime/MessageGeneration/MessageDeserializer.cs:25)
Unity.Robotics.ROSTCPConnector.ROSConnection.SendServiceMessage[RESPONSE] (System.String rosServiceName, Unity.Robotics.ROSTCPConnector.MessageGeneration.Message serviceRequest) (at ./Library/PackageCache/com.unity.robotics.ros-tcp-connector@c27f00c6cf/Runtime/TcpConnector/ROSConnection.cs:318)
Unity.Robotics.ROSTCPConnector.ROSConnection.SendServiceMessage[RESPONSE] (System.String rosServiceName, Unity.Robotics.ROSTCPConnector.MessageGeneration.Message serviceRequest, System.Action`1[T] callback) (at ./Library/PackageCache/com.unity.robotics.ros-tcp-connector@c27f00c6cf/Runtime/TcpConnector/ROSConnection.cs:286)
System.Runtime.CompilerServices.AsyncMethodBuilderCore+<>c.<ThrowAsync>b__7_0 (System.Object state) (at <4d56092c9a8849cab50a947b257065a3>:0)
UnityEngine.UnitySynchronizationContext+WorkRequest.Invoke () (at /home/bokken/build/output/unity/unity/Runtime/Export/Scripting/UnitySynchronizationContext.cs:153)
UnityEngine.UnitySynchronizationContext.Exec () (at /home/bokken/build/output/unity/unity/Runtime/Export/Scripting/UnitySynchronizationContext.cs:83)
UnityEngine.UnitySynchronizationContext.ExecuteTasks () (at /home/bokken/build/output/unity/unity/Runtime/Export/Scripting/UnitySynchronizationContext.cs:107)
```

**Environment**:

Unity Version: 2022.3.1f1
ROS machine OS + version: Ubuntu 22.04, ROS Noetic run in Docker
UnityRoboticsHub / ROSTCPConnector version: v0.7.0 / v0.5.2

---
I tried using the provided Docker with ROS Melodic and that works but I would like to use my Noetic environment

cambel, 2023-06-09:
Nevermind.

The problem was the MoveIt msgs. The RobotTrajectory.msg on Noetic must be different from the version on Melodic.

I updated the whole ROS/src/moveit_msgs with the same version as in my Noetic environment and rebuild them on Unity. 

Now it works fine.