Unity-Technologies/Unity-Robotics-Hub #338: How can I set the start joint configuration of a robot for a simulation?

marcoojer, 2021-12-22:
Hi everyone,

I wonder how I can set the joint positions for the robot before the simulation starts. Every URDF that I load, the joints are at their zeros, but I will like to start the simulation with the joints in other positions. I know you can move any joint with the articulationDrivex.target but this happens during the simulation and its taking into account the pyshics. 
Any idea about how to do that?

Thanks in advance

vidurvij-Unity, 2021-12-29:
There are two options to set the joint angles before the simulation start:
1. You can change the joint positions by changing the target positions of each joint in the Articulation Body inspector window of each joint.
[image]

2. Or use ArticulationDrive.target in the Start() or OnEnable() function of the MonoBehaviour. It sets the value of the ArticulationBody angles before the simulation starts.

marcoojer, 2022-01-04:
Thanks you for your answer. But for what I understood and try about the articulation body, what you do by setting the target, is than when the simulation starts it will go to that position as soon as the physics are enable, if there is an obstacle in the way it will get stuck there. What I want is to start the simulation directly in a particular joint configuration. For example, if you want the first joint to start at -90 but there is an obstacle in -45, setting the target will get the robot stuck in -45

mrpropellers, 2022-01-12:
Does setting the "Anchor Rotation" accomplish what you need @marcoojer? You can also set the rotation value in the encapsulating GameObject's Transform. I think either of these options are effectively changing the zero point of the joint angle, though, so you would need to also modify the drive ranges to keep the range of movement from rotating as well.

marcoojer, 2022-02-17:
I found out the solution, which was the function SetJointPositions() on the root articulated body. You also need define the Driver target to the same position.