Unity-Technologies/Unity-Robotics-Hub #339: Mention to place Pick and place robot at origin in Tutorial

husainhz7, 2021-12-23:
**Describe the bug**
The Pick and place tutorial does not specify to place the imported urdf niryo robot at origin with no rotation. If placed at any other orientation, the trajectory will be wrong. 
pick_and_place/1_urdf.md can be mention to place the robot at at origin with no change in orientation. It wasn't clear in the tutorial

**Expected behavior**
Niryo robot trajectory should be work

**Environment (please complete the following information, where applicable):**
- Unity Version:  Unity 2020.3.11f1
- Unity machine OS + version: Windows 10
- ROS–Unity communication:  Docker

vidurvij-Unity, 2021-12-29:
Hi @husainhz7 ,
This is not a bug. The tutorial is meant to followed verbatim and the import parameters written in the URDF file imports the robot in the correct position which is compatible with the IK profile setup in ROS.

closed 2021-12-29 by vidurvij-Unity

saratrajput, 2022-03-16:
@husainhz7 I think #362 solves your issue.
@vidurvij-Unity I understand it is not a bug, but being able to move the robot can help in VR/AR development where the default position of the AR/VR device is at the origin. Please have a look at #360 for more details.