Unity-Technologies/Unity-Robotics-Hub #341: ROSConnection error

Hrithik-verma, 2022-01-08:
I followed the entire tutorials, but still I'm unable to connect ros with unity. When I start play on unity nothing display and I have check my host id is also correct but I does not what is the issue.
I also tried ros-unity publisher but I'm getting a error after adding the script.
Error:
[image]

Version:
ROS TCP Connector v0.5.0
unity - 2020.3.25f1
my pc config:
ubuntu 18.04
ros melodic

Please help me. I'm very new to unity I do not understand much of it

Hrithik-verma, 2022-01-12:
Yeah I understood this error exist because I was using ROS TCP Connector of v0.5.0 and script of v0.6.0.
It took me some time to figure this out. So to help other I made a small YouTube video.
Video link: https://youtu.be/pdMxLxolQuo

But, still there is one error when I add ROS TCP Connector of v0.6.0 pakage.
error: "A meta data file (.meta) exists but its folder 'Packages/com.unity.robotics.ros-tcp-connector/Tools' can't be found, and has been created. Empty directories cannot be stored in version control, so it's assumed that the meta data file is for an empty directory in version control. When moving or deleting folders outside of Unity, please ensure that the corresponding .meta file is moved or deleted along with it."
Although, It doesn't effect ros unity connection still let me why this error is coming up.

Thanks

at669, 2022-01-12:
Thanks for making that video! Feel free to post it to the [Robotics Forum](https://forum.unity.com/forums/robotics.623/) to share it with others.

> A meta data file (.meta) exists...

This is a warning that doesn't affect the functionality of the package, not an error. The metafile in question has been removed in the latest development branches of the package, and this version will be released at a later point. 

I'll close out this issue as your question has been resolved, but feel free to open a new issue if something else comes up. Thanks!

closed 2022-01-12 by at669