Unity-Technologies/Unity-Robotics-Hub #343: Custom URDF results in rendering error (problem with inertia tensor rotation)

cedricgoubard, 2022-01-13:
**Describe the bug**
When importing my URDF file and launching the scene, the robot does not appear. I have narrowed it down to a problem with the inertia values for one of the links in the URDF file, which results in a NaN value in the object's inertia tensor rotation.

**To Reproduce**
Steps to reproduce the behavior:
1. Clone [the repository I created for this occasion](https://github.com/cedricgoubard/DemoBugURDF)
2. Load the project in Unity
3. Open the `default_with_cube` scene
3. The projects contains 2 URDF files in `Assets/URDF`: `working.urdf` and `not_working.urdf`. The only different line between the 2 is the one causing the issue
4. Import `not_working.urdf` with default settings, move the robot on top of the cube
5. Launch the scene and see error

**Console logs / stack traces**
A lot of `Invalid AABB aabb`.

**Expected behavior**
The robot appears in the scene.

**Screenshots**
Working example - notice the `URDF Inertia Tensor Rotation` values in the bottom right
[image]

Not working example - this time, with NaN values
[image]

**Environment (please complete the following information, where applicable):**
- Unity Version: Unity 2020.3.25f1
- Unity machine OS + version: Ubuntu 20.04
- ROS machine OS + version: Ubuntu 20.04, no ROS
- ROS–Unity communication: no ROS
- Branch or version: URDF Importer 0.5.0-preview

**Additional context**
I am guessing the issue is caused by the low inertia values (`1.0955E-20`), but I guess the importer should then set the value to `0` instead of `NaN`.

vidurvij-Unity, 2022-02-08:
@cedricgoubard Thanks for bringing this to our attention. Added an internal ticket and started a PR to correct the behavior.
[AIRO - 1703]

vidurvij-Unity, 2022-02-09:
Correction has been merged in dev branch.

closed 2022-02-09 by vidurvij-Unity