Unity-Technologies/Unity-Robotics-Hub #347: Find shortest path for free revolute joint on an Articulation body

agru97, 2022-01-24:
When I set the X Drive target of the free revolute joint on an Articulation body to 1 and the to a 360 it will move the whole way around.
How do I set my robot up so that it always takes the shortest part. DO I have to fix this issue in Unity or on the ROS/MoveIt Side?
Because when I get my trajectory from Moveit I think i have the same problem like the 1 to 360 rotation. This causes my Robot to do some uncontrolled movements.

vidurvij-Unity, 2022-02-02:
Hi sorry for the delayed response. We are having an internal discussion about this issue. We will get back to you as soon as possible. [AIRO-1697]

hanlonm, 2022-10-21:
Is there an update on this? I am having the same issue.
Or does anyone have a workaround?