Unity-Technologies/Unity-Robotics-Hub #349: I can't connect ROS with Unity by using tutorial of pick_and_place.

ktomohiro990103, 2022-01-25:
Hi everyone!
Please tell me how to remove the error when I connect ROS with Unity.

I execute the tutorial of pick_and_place. And I finished twelfth on the Unity Side of "Pick-and-Place Tutorial : Part2".
When I runed "roslaunch niryo_moveit part_2.launch" on ROS workspace, this error happened.

Traceback (most recent call last):
  File "/home/username/catkin_ws/src/Unity-Robotics-Hub/tutorials/pick_and_place/ROS/src/ros_tcp_endpoint/src/ros_tcp_endpoint/default_server_endpoint.py", line 5, in <module>
    from ros_tcp_endpoint import TcpServer
ImportError: No module named ros_tcp_endpoint
[server_endpoint-1] process has died [pid 3965, exit code 1, cmd /home/username/catkin_ws/src/Unity-Robotics-Hub/tutorials/pick_and_place/ROS/src/ros_tcp_endpoint/src/ros_tcp_endpoint/default_server_endpoint.py --wait __name:=server_endpoint __log:=/home/username/.ros/log/721f2210-7d99-11ec-8fda-080027322554/server_endpoint-1.log].
log file: /home/username/.ros/log/721f2210-7d99-11ec-8fda-080027322554/server_endpoint-1*.log

I'm sorry. I can't find "/.ros/log/721f2210-7d99-11ec-8fda-080027322554/server_endpoint-1.log". So, I can't show this log.

Next, I tell you about my PC.
host PC version : windows 10
Unity version : 2021.2.1f1

local PC version : Ubuntu 18.04 LTS
I installed Ubuntu by Oracle VM VirualBox 6.1.28-147628-Win.
ROS1 version : melodic
I executed ROS1 setup by using docker.

Finally, I tell you about what I did before I run "roslaunch niryo_moveit part_2.launch".
Because this error happened, I deleted ros2_packages, ros2_docker and PickAndPlaceProject on ROS1.

RuntimeError: Multiple packages found with the same name "niryo_one_urdf"
RuntimeError: Multiple packages found with the same name "unity_robotics_demo"
RuntimeError: Multiple packages found with the same name "unity_robotics_demo_msgs"

hyounesy, 2022-01-25:
Hi @ktomohiro990103, Looks like ros_tcp_endpoint is not properly added to your catkin workspace.

- Download and copy the [ROS-TCP-Endpoint](https://github.com/Unity-Technologies/ROS-TCP-Endpoint) package into the `src` folder in your Catkin workspace. 
- Then navigate to your Catkin workspace and run `catkin_make`, 
- then `source devel/setup.bash`. Ensure there are no errors.

For more details please refer to [Pick-and-Place Tutorial: Part 0](https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/main/tutorials/pick_and_place/0_ros_setup.md)

ktomohiro990103, 2022-01-26:
Thank you for telling me about how to remove the error!
When I runed "roslaunch niryo_moveit part_2.launch" on ROS workspace, two errors happened.

I tell you about what I did.
I downloaded the ROS-TCP-Endpoint package into the "/home/username/catkin_ws/src" folder.
And I succeeded "catkin_make" and "source devel/setup.bash" in my catkin workspace in this way.

username@username-VirtualBox:~/catkin_ws$ catkin_make
Base path: /home/username/catkin_ws
Source space: /home/username/catkin_ws/src
Build space: /home/username/catkin_ws/build
Devel space: /home/username/catkin_ws/devel
Install space: /home/username/catkin_ws/install
Multiple packages found with the same name "ros_tcp_endpoint":
- ROS-TCP-Endpoint
- Unity-Robotics-Hub/tutorials/pick_and_place/ROS/src/ros_tcp_endpoint
username@username-VirtualBox:~/catkin_ws$ source devel/setup.bash
username@username-VirtualBox:~/catkin_ws$ cd ~/catkin_ws/src/Unity-Robotics-Hub/tutorials/pick_and_place/ROS
...

When I runed "roslaunch niryo_moveit part_2.launch" on ROS workspace, this error happened.

Traceback (most recent call last):
  File "/opt/ros/melodic/lib/python2.7/dist-packages/roslaunch/__init__.py", line 247, in main
    args = rlutil.resolve_launch_arguments(args)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/roslaunch/rlutil.py", line 92, in resolve_launch_arguments
    resolved = roslib.packages.find_resource(args[0], args[1])
  File "/opt/ros/melodic/lib/python2.7/dist-packages/roslib/packages.py", line 528, in find_resource
    source_path_to_packages=source_path_to_packages)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/catkin/find_in_workspaces.py", line 149, in find_in_workspaces
    source_path_to_packages[source_path] = find_packages(source_path)
  File "/usr/lib/python2.7/dist-packages/catkin_pkg/packages.py", line 96, in find_packages
    raise RuntimeError('\n'.join(duplicates))
RuntimeError: Multiple packages found with the same name "ros_tcp_endpoint":
- ROS-TCP-Endpoint
- Unity-Robotics-Hub/tutorials/pick_and_place/ROS/src/ros_tcp_endpoint

So, I deleted Unity-Robotics-Hub/tutorials/pick_and_place/ROS/src/ros_tcp_endpoint.
Then when I runed "roslaunch niryo_moveit part_2.launch" on ROS workspace, two error happened.

Traceback (most recent call last):
  File "/home/username/catkin_ws/src/ROS-TCP-Endpoint/src/ros_tcp_endpoint/default_server_endpoint.py", line 5, in <module>
    from ros_tcp_endpoint import TcpServer
ImportError: No module named ros_tcp_endpoint
Traceback (most recent call last):
  File "/home/username/catkin_ws/src/Unity-Robotics-Hub/tutorials/pick_and_place/ROS/src/niryo_moveit/scripts/trajectory_subscriber.py", line 9, in <module>

    from niryo_moveit.msg import NiryoMoveitJoints, NiryoTrajectory
ImportError: No module named niryo_moveit.msg
[server_endpoint-2] process has died [pid 5439, exit code 1, cmd /home/username/catkin_ws/src/ROS-TCP-Endpoint/src/ros_tcp_endpoint/default_server_endpoint.py --wait __name:=server_endpoint __log:=/home/username/.ros/log/fc52e24e-7e4a-11ec-a5dc-080027322554/server_endpoint-2.log].
log file: /home/username/.ros/log/fc52e24e-7e4a-11ec-a5dc-080027322554/server_endpoint-2*.log
[trajectory_subscriber-3] process has died [pid 5440, exit code 1, cmd /home/username/catkin_ws/src/Unity-Robotics-Hub/tutorials/pick_and_place/ROS/src/niryo_moveit/scripts/trajectory_subscriber.py --wait __name:=trajectory_subscriber __log:=/home/username/.ros/log/fc52e24e-7e4a-11ec-a5dc-080027322554/trajectory_subscriber-3.log].
log file: /home/username/.ros/log/fc52e24e-7e4a-11ec-a5dc-080027322554/trajectory_subscriber-3*.log

StephanHasler, 2022-02-03:
Hi @ktomohiro990103, it may be that in your setup you have two ROS workspaces inside each other.

I think you placed:
Unity-Robotics-Hub/tutorials/pick_and_place/ROS
inside you own workspace:
~/catkin_ws/src
But according to:
https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/main/tutorials/pick_and_place/0_ros_setup.md
Unity-Robotics-Hub/tutorials/pick_and_place/ROS is meant to be the ROS workspace.

When you now execute some ROS commands on the level of ~/catkin_ws, it will parse all the sub-directories of Unity-Robotics-Hub/... and it may be confused of the things it finds there.
You should execute all ROS commands (build/run) inside Unity-Robotics-Hub/tutorials/pick_and_place/ROS.
Then it will only see the sub-directories of that, which should be consistent.

I think you can also copy everything from:
Unity-Robotics-Hub/tutorials/pick_and_place/ROS
into:
~/catkin_ws/src
(and then delete ~/catkin_ws/src/Unity-Robotics-Hub)
This should be consistent then again.

ktomohiro990103, 2022-02-17:
Hi @StephanHasler 
Thank you for telling me about how to remove the error. I'm sorry for the delayed response.
When I executed your solution, I understood that I couldn't use simple commands of ROS.
I did trial and error to confirm that I can use commands of ROS. So, I delayed the response.
When I can use commands of ROS, I will execute your solution again.

ktomohiro990103, 2022-02-28:
Thank you for telling me about how to remove the error!
This error happened because I didn't execute "source devel/setup.bash" before I runed "roslaunch niryo_moveit part_2.launch".
I resolved trouble of ROS. So, thank you so much!

closed 2022-02-28 by ktomohiro990103

syedjawadakhtar, 2022-06-30:
Hey! I'm getting the same problem of not being able to run the `roslaunch niryo_moveit part_2.launch` command in my ROS 2 galactic workspace on a virtual machine. 
So I was able to follow the Pick and Place tutorial till the Part 2 unity part successfully. But I've been stuck in the ROS part.

I cloned the Unity-Robotics-Hub repo in my VM and `colcon build` gave me errors of multiple files with same name. So I made a new workspace for ROS and pasted all the packages (moveit_msgs, niryo_moveit, niryo_one_ros, niryo_one_urdf, ros_tcp_endpoint) inside `src` folder. 
Now after sourcing it from the root workspace folder i.e. ROS: `source install/setup.sh` and
then - `colcon build` gives me the following error.

```
[0.242s] WARNING:colcon.colcon_core.package_selection:Some selected packages are already built in one or more underlay workspaces:

'dynamixel_sdk' is in: /home/ubicompros2/ROS/install/dynamixel_sdk, /opt/ros/galactic

If a package in a merged underlay workspace is overridden and it installs headers, then all packages in the overlay must sort their include directories by workspace order. Failure to do so may result in build failures or undefined behavior at run time.
If the overridden package is used by another package in any underlay, then the overriding package in the overlay must be API and ABI compatible or undefined behavior at run time may occur.
If you understand the risks and want to override a package anyways, add the following to the command line:
	--allow-overriding dynamixel_sdk

This may be promoted to an error in a future release of colcon-override-check.
Starting >>> moveit_msgs
Starting >>> dynamixel_sdk
Starting >>> mcp_can_rpi
Starting >>> niryo_one_description
Starting >>> ros_tcp_endpoint
--- stderr: dynamixel_sdk                                                                     
CMake Error at CMakeLists.txt:10 (find_package):
  By not providing "Findcatkin.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "catkin", but
  CMake did not find one.

  Could not find a package configuration file provided by "catkin" with any
  of the following names:

    catkinConfig.cmake
    catkin-config.cmake

  Add the installation prefix of "catkin" to CMAKE_PREFIX_PATH or set
  "catkin_DIR" to a directory containing one of the above files.  If "catkin"
  provides a separate development package or SDK, be sure it has been
  installed.

---
Failed   <<< dynamixel_sdk [0.16s, exited with code 1]
--- stderr: moveit_msgs
CMake Error at CMakeLists.txt:14 (find_package):
  By not providing "Findcatkin.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "catkin", but
  CMake did not find one.

  Could not find a package configuration file provided by "catkin" with any
  of the following names:

    catkinConfig.cmake
    catkin-config.cmake

  Add the installation prefix of "catkin" to CMAKE_PREFIX_PATH or set
  "catkin_DIR" to a directory containing one of the above files.  If "catkin"
  provides a separate development package or SDK, be sure it has been
  installed.

---
Failed   <<< moveit_msgs [0.18s, exited with code 1]
Aborted  <<< ros_tcp_endpoint [0.13s]
Aborted  <<< mcp_can_rpi [0.15s]
Aborted  <<< niryo_one_description [0.14s]

Summary: 0 packages finished [0.38s]
  2 packages failed: dynamixel_sdk moveit_msgs
  3 packages aborted: mcp_can_rpi niryo_one_description ros_tcp_endpoint
  5 packages had stderr output: dynamixel_sdk mcp_can_rpi moveit_msgs niryo_one_description ros_tcp_endpoint
  18 packages not processed
```

The stderr: dynamixel seems new to me.

Can anyone suggest what I'm doing wrong to make this launch file work?
Thank you!

BR,
Jawad