Unity-Technologies/Unity-Robotics-Hub #350: Tutorial For ROS action server and clients

marcoojer, 2022-01-25:
Hi,

Is there any tutorial/example about how to write action servers and action clients in the Unity Side?. I see there are examples for topic and services but not for actions. Are Ros actions not implemented yet?

Best Regards

hyounesy, 2022-01-25:
Hello @marcoojer You should be able to generate the Action messages but will have to publish or subscribe to the action's topic. [This part of the tutorial](https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/main/tutorials/pick_and_place/4_pick_and_place.md) uses actions and it might be a good reference. Please feel free to let us know if you have more questions.

hyounesy, 2022-01-25:
Update: we have recently fixed a bug related to the message generation for the ROS actions in ROS-TCP-Connector. The fix will be included in the next release soon, but until then you may use the [dev branch](https://github.com/Unity-Technologies/ROS-TCP-Connector/tree/dev).

marcoojer, 2022-01-26:
I see, I'll take a look at that.  Thank you very much!!!

closed 2022-01-26 by marcoojer