Unity-Technologies/Unity-Robotics-Hub #353: Using RosConnector to Oculus Quest 2

pucciland95, 2022-02-01:
Hello everybody, 
I start by saying that I am not really good both with ROS and Unity so please be patient.

I managed to finish the pick_and_place tutorial that the Git Hub page presents and everything works fine (I arrived up to part 2 tutorial).
What I would like to do now is to build everything into my Oculus Quest 2 and make it communicate with my computer where ROS nodes are running.

As far as I understood this should be possible provided that you can build the Android app with RosConnector (which I did). The building gives me no problems at all but as soon as I launch my server on my main computer (IP: 192.168.1.100) I launch the build for the Oculus. The build ends fine without errors but the server does not detect any connections even after that my up is running. 

Both the Oculus and my Pc are connected under the same network and I can reach the Oculus from my Pc pinging it.
Where can it be the problems?
Do you have a clever way to Debug in Oculus to see what could be wrong?
Is what I am describing possible?

Launching the same application in Editor mode gives absolutely no problems.

OS: Ubuntu 20.04
ROS version: Noetic

pucciland95, 2022-02-02:
Hello everybody,

I solved the issue and now everything works! 
I managed to make communicate ROS with the Robot in my Oculus Quest 2. Hereafter is a short guide on how to make part_2 and part_3 for the pick_and_place tutorial work in an Android App in general. (I write it since I have not found anything about that and I think it is really cool to see the robot perform the same actions that it does in the Unity editor).

1) Finish part_2 of the pick_and_place tutorial.
2) Follow the guide about how to start an Oculus Project from scratch from the official Unity website: https://learn.unity.com/tutorial/create-a-vr-starter-project-from-scratch
3) It will tell you to switch to the NEW action library to use hands movements inside Oculus. Say yes. Unity will restart and afterwards, it will complain that you change it since Controller.cs uses Input.somehing to get input from keyboard. Comment out everything related to that since it is not necessary.
4) Eliminate "demo" repo since it throws errors and it is not necessary
5) Now, this step is **crucial** if you want to build correctly everything in an Android App. Look Under Script folder you should have Unity.Robotics.PickAndPlace that references: URDF importer, Ros Connector TCP and RosMessage. Open all of them and look for the .asmdef (I guess it stands for assembly definition) and tick **ONLY** Android and Editor. Check it more than once for all of them to be sure that it builds only for these applications (Editor and Android). Finally, do the same for the initial .asmdef which is the one under your Script folder.

Now launch the ROS launch files and build your application. If everything works fine Oculus will connect to ROS as soon as you wear it. 

Probably I skipped some passages, feel free to reopen the issue and write for potential errors.
I hope I have been clear enough and this guide will be useful to someone.

Cheers

Niccolò

closed 2022-02-02 by pucciland95

rbriki, 2022-02-19:
Hi everyone,
sorry to reopen the issue but I am facing the same problem for a University course project and can't manage to make things work.
I followed the steps explained by Niccolò but as I press Build and Run and put my Oculus Quest 2 on the application is running but nothing happens on the ROS server as no connection is detected.

Moreover, I could not see the publish button but I think this is more a design problem since it is my first time developing VR application and I'm not familiar with the process. Anyway, if anyone has a quick solution even to this problem I would be grateful.

I'm running Unity on Windows 10 and ROS on the Docker Engine suggested at Pick and Place tutorial Part 0.
I followed the Tutorial Up to Part 3 Step by Step many times before reopening the issue.
For what regard the VR implementation I followed the guide linked below:
https://learn.unity.com/tutorial/create-a-vr-starter-project-from-scratch#620f8f56edbc2a705d1c3578
which should be the same proposed by Niccolò. 
Still, I followed it step by step the same number of times I followed the tutorial of this Repo.  Only thing that differs from the unity VR tutorial is that I could not manage to change the  _Scriptable Render Pipeline Settings_ as the Pick_and_Place Project does not include the URP as far as I understood. 
Last thing is that I also added the Oculus plug-in from the XR Plug-in Management.

OS: Windows 10, for Unity
ROS running on docker as for Part 0 instruction
Packet Manager in Unity: 1) Ros Tpc Connector 2) URDF Importer
Laptop model: Asus Rog GL752VW, that means gtx 960m so no Oculus link possibility in case that matters.

I am writing everything as I am not familiar with none of the plication involved (ROS, Unity, Oculus Quest)   and I want to be as complete as possible. 

Again, sorry for reopening the issue, if I missed something important for your understanding of the problem let me know.
I am waiting for your help,
thank you in advace
RBR

mericgeren, 2023-05-16:
Hi @pucciland95 ,

I was just following steps you have described. But, there are couple of things i couldn't get. First of all, is the 3rd step you have described specific to pick_and_place tutorial? Secondly, i see in the 5th step that you have mentioned a "initial .asmdef" which file do you refer here? Finally, i am curious about do i need to change every ".asmdef" in the project or just the ones from packages we have git cloned from Unity-Robotics-Hub? If you don't mind could you explain these points for me please?

Thank you in advance for all the help you offered.

pucciland95, 2023-05-16:
Hi, 

I must be honest I don't remember much since it passed more than one year and I am not using the device anymore. 

However, regarding your question:

- step 3 is true in general if I remember correctly. 

- regarding step 5 I don't really remember. The only thing I remember is that if you build the project for the wrong application (which is specified in each *.asmdef file) it does not throw any error but I will not run correctly (you will have weird behaviours) in your Oculus device. 
To build applications in Oculus you need to build for Android. 

Sorry for not being more helpful than this. 

Niccoló