Unity-Technologies/Unity-Robotics-Hub #354: 4 wheel differential robot wrong movement

argolomb, 2022-02-07:
**Describe the bug**
I'm trying to reproduce this robot(https://global.agilex.ai/products/scout-2-0) in unity to replace gazebo as my simulation platform. Everything is fine with URDF-Importer and ROS-TCP Connection. I can get the joint state message and convert to movement with "ArticulationDrive drive = joint.xDrive;"

The mobile robot go forward and backward but no curves or spin around his center. 

The problem occur when the wheel receive different velocities.

I've tried use cmd_vel msg and calculate the velocity in each wheel but with the same behavior.

**Console logs / stack traces**
```
void Start()
        {
            ros = ROSConnection.GetOrCreateInstance();
            ros.Subscribe<JointStateMsg>("joint_states", ReceiveROSCmd);
            rl_wheel = rear_left_wheel.GetComponent<ArticulationBody>();
            rr_wheel = rear_right_wheel.GetComponent<ArticulationBody>();
            fl_wheel = front_left_wheel.GetComponent<ArticulationBody>();
            fr_wheel = front_right_wheel.GetComponent<ArticulationBody>();
            
            
            SetParameters(rl_wheel);
            SetParameters(rr_wheel);
            SetParameters(fl_wheel);
            SetParameters(fr_wheel);
            Debug.Log("Start");
        }

        void ReceiveROSCmd(JointStateMsg joint_state_msg){
            Debug.Log("ReceiveROSCmd");
            rl_wheel_vel = joint_state_msg.velocity[2];
            rr_wheel_vel = joint_state_msg.velocity[3];
            fl_wheel_vel = joint_state_msg.velocity[0];
            fr_wheel_vel = joint_state_msg.velocity[1];

            SetSpeed(rl_wheel, (float) rl_wheel_vel);
            SetSpeed(rr_wheel, (float) rr_wheel_vel);
            SetSpeed(fl_wheel, (float) fl_wheel_vel);
            SetSpeed(fr_wheel, (float) fr_wheel_vel);
                          
        }

        private void SetSpeed(ArticulationBody joint, float wheelSpeed = float.NaN)
        {
            ArticulationDrive drive = joint.xDrive;
            if (float.IsNaN(wheelSpeed))
            {
                drive.targetVelocity = ((2 * maxLinearSpeed) / wheelRadius) * Mathf.Rad2Deg;
            }
            else
            {
                drive.targetVelocity = wheelSpeed*Mathf.Rad2Deg;
            }
            joint.xDrive = drive;
        }
```

**Expected behavior**
The same behavior in all software

**Screenshots**
<a href="http://www.youtube.com/watch?feature=player_embedded&v=1GrUhLVfuIg
" target="_blank"><img src="http://img.youtube.com/vi/1GrUhLVfuIg/0.jpg" 
alt="IMAGE ALT TEXT HERE" width="240" height="180" border="10" /></a>

**Environment (please complete the following information, where applicable):**
- Unity Version:2020.3.26f1
- Unity machine OS + version:Ubuntu 20.04
- ROS machine OS + version: Ubuntu 20.04, ROS Noetic
- ROS–Unity communication: NA
- Branch or version: v0.6.0

vidurvij-Unity, 2022-02-08:
Can you share the repository for the project. It would help me understand the Physics settings you are using.

argolomb, 2022-02-08:
> Can you share the repository for the project. It would help me understand the Physics settings you are using.
[AIRO-1702]

Yes, sure. 

[scout_beta](https://github.com/argolomb/scout_beta)

DLu, 2022-02-23:
I am also hoping for a solution here. @vidurvij-Unity any luck?

argolomb, 2022-02-24:
> I am also hoping for a solution here. @vidurvij-Unity any luck?

My solution was: 
- Not use URDF importer
- Create the robot in unity 
- Add wheel coliders on wheels 
- Create 4 wheel drive controller(using PID)

argolomb, 2022-07-13:
The solution is set a rigidbody.inertiaTensor value.
```               
        public Vector3 tensor;
        public Rigidbody rb;
        void Start()
        {
            ros = ROSConnection.GetOrCreateInstance();
            ros.Subscribe<TwistMsg>("cmd_vel", ReceiveROSCmd);
            rb.inertiaTensor = tensor;
            rb.centerOfMass = centerOfMass.transform.localPosition;
        }
```

closed 2022-07-13 by argolomb

ptiwari0664, 2023-12-01:
@argolomb I am trying to recreate another vehicle using similar concept. Could you please illustrate how have you calculate the inertia tensor ? My vehicle is not able to climb on the hills.